solectrus/solectrus #2152: Zeitanzeige im '

Zeppelin-PV, 2023-07-06:


Zeppelin-PV, 2023-07-06:
wäre es möglich und denkbar dass die Zeitanzeige im 'Jetzt' (die ja immer die aktuelle Stunde abbildet) bei gedrückter Maustaste
nach links und rechts verschiebbar zu machen ? (maximal bis zur Tagesgrenze)
Das hätte einen großen Vorteill wenn man mal kurz zwei Stunden zurückschauen möchte und die Anzeige im 'Tag' zu grob dafür ist.
Für den Desktop sicherlich sinnvoll.

ledermann, 2023-07-07:
Ja, das ist denkbar, dürfte aber etwas knifflig zu realisieren sein. Ich lasse es erst mal so stehen.

MartinKBroll, 2024-09-24:
> wäre es möglich und denkbar dass die Zeitanzeige im 'Jetzt' (die ja immer die aktuelle Stunde abbildet) bei gedrückter Maustaste nach links und rechts verschiebbar zu machen ? (maximal bis zur [...]

Aber gerade auf dem Desktop ist es doch mit der Maus problemlos möglich, einen beliebigen Zeitabschnitt größer zu ziehen?

ledermann, 2024-09-24:
@MartinKBroll Das Knifflige beim oben beschrieben "Schieben" ist, dass ja Daten nachgeladen werden müssen. Das ist natürlich machbar, aber eben nicht so einfach.

Übrigens: Das derzeitige Zoomen sorgt nicht für eine höhere Auflösung. Es wird nur das vorhandene Diagramm hochskaliert, es erscheinen keine zusätzlichen Datenpunkte.