solectrus/solectrus #2174: "Einspeisung" ohne Quelle

spyro2000, 2023-07-13:
Insbesondere nach höheren Lasten, die direkt aus dem Akku bedient werden, erhalte ich oft Werte wie folgt:

[image]

Obwohl es längst dunkel ist, wird hier eine Einspeisung angezeigt, die auch auf der linken Seite nicht auftaucht.

Datenquelle ist ein Senec Home v2.

ledermann, 2023-07-14:
Ich denke, das liegt in der Trägheit des Speichers begründet: 

Dein Hausverbrauch hat, wie man im Diagramm sehen kann, gerade eine höhere Last verursacht, die dann aus dem Speicher gedeckt wurde. Dann hat der Verbraucher **abgeschaltet**, der Speicher liefert aber noch ein paar Sekunden die höhere Leistung, bevor er das "merkt". Und da der Strom irgendwo hin muss, wird er eben eingespeist. Insofern: Ja, es erfolgt eine (kurze) Einspeisung aus dem Akku ins Netz. Analog erfolgt beim **Einschalten** eines starken Verbrauchers auch ein kurzer Netzbezug, bevor der Speicher reagiert und liefert.

Ob andere Hersteller dies anders handhaben und ob das überhaupt möglich ist, weiß ich nicht. Meinen Beobachtungen zufolge hat das aber nur minimale Auswirkungen.

closed 2023-07-14 by ledermann

spyro2000, 2023-07-14:
Ahh, verstehe, danke für die Aufklärung! :)
Hätte nicht gedacht, dass das gleich mehrere Minuten dauert.

ledermann, 2023-07-14:
Soweit ich das bei meinem V3-Speicher beobachten kann, sind das immer nur wenige Sekunden. Nicht Minuten.

Zeppelin-PV, 2023-07-14:
ja, das habe ich auch schon beobachtet und festgestellt dass es sich wirklich um ein paar Sekunden handelt
steht man bei diesem Regelverhalten direkt am Speicher laufen die LEDs von oben nach unten und queerbeet (wie in der Disco)
die Anzeige am Speicher ist 500ms schnell