solectrus/solectrus #2213: Balkendiagramme mit abgerundeten Ecken

ledermann, 2023-07-29:
> Eine Kleinigkeit zur Optik: Bei der Anzeige Verwendung und Herkunft sind die Ecken leicht abgerundet dargestellt. Vielleicht wäre das bei den Balkendiagrammen auch möglich und würde diese etwas "weicher" aussehen lassen. Als Beispiel hier die Small radius Balken bei Charts.js. https://www.chartjs.org/docs/latest/samples/bar/border-radius.html

_Originally posted by @StevieC121176 in https://github.com/solectrus/solectrus/issues/1881#issuecomment-1656038026_

closed 2023-07-29 by ledermann

ledermann, 2023-07-29:
@StevieC121176 Ja, abgerundete Ecken gibt es in SOLECTRUS überall, daher sollte das auch für die Balken gelten. 

Ich habe das gerade erledigt, es wird ins nächste Update einfließen. In der Live-Demo ist es bereits jetzt zu sehen: https://solectrus.ledermann.dev/grid_power/2023-W30

StevieC121176, 2023-07-29:
Sieht sehr gut aus, vielen Dank.

commit ed84548d (2023-07-29): Charts: Use rounded corners for bars

Resolves #2213