solectrus/solectrus #2214: Strich statt Zeiger in Diagrammen

Zeppelin-PV, 2023-07-29:
viele Geräte, viele Darstellungsformen
eine ansprechende Version habe ich in einer App gefunden
Der Messwert wird mit einem vertikalen Strich markiert anstatt mit dem Pfeil an der Infobox
Der Messwertpunkt ist weiß mit Rahmen in der Diagrammfarbe
Die Infobox transparent
auch für Solectrus eine Aufwertung, oder ???

[image]

StevieC121176, 2023-07-29:
Ja die Markierung mit dem senkrechten Strich finde ich gut. Ich habe mal in Charts.js gestöbert, aber leider nix derartiges gefunden.

ledermann, 2023-07-29:
Es gibt ein Plugin für Chart.js, das genau so etwas leistet:
https://chartjs-plugin-crosshair.netlify.app/samples/

Allerdings wird das Plugin [nicht mehr gepflegt](https://github.com/AbelHeinsbroek/chartjs-plugin-crosshair/issues/120), was auch der Grund ist, warum ich Plugins eher vermeiden möchte.

Ansonsten könnte man das Feature natürlich auch selbst bauen, aber ich denke momentan nicht, dass sich der Aufwand dafür lohnt.

StevieC121176, 2023-07-29:
Ah schade, in dem Beispiel macht das einen guten Eindruck und würde meiner Meinung nach gerade in der mobilen Ansicht die Übersicht verbessern.

PR #3895 (2025-01-22): Liniendiagramme mit vertikalem Fadenkreuz
Closes #2214
commit cc681cef (2025-01-22): Liniendiagramme mit vertikalem Fadenkreuz (#3895)

* Add vertical crosshair to line charts

Closes #2214
closed 2025-01-22 by ledermann