solectrus/solectrus #2222: Minor Nitpick: Zahlenformatierung im Chart-Mouseover

gereons, 2023-08-01:
In den Chart-Mouseovers werden Zahlen "englisch" mit Dezimalpunkt und Tausenderkomma statt mit Dezimalkomma und Tausenderpunkt formatiert.

[image]

ledermann, 2023-08-02:
Interessant, bei mir (Firefox/Chrome/Safari, Sprache=DE) sieht es korrekt aus:

[image]

Für die Formatierung verwende ich [Intl.NumberFormat().format](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Intl/NumberFormat), wodurch es dem Browser überlassen wird, wie genau das Ergebnis aussieht.

Welchen Browser verwendest Du denn? Hast Du besondere Spracheinstellungen im Browser vorgenommen? Tritt das gleiche Phänomen bei Dir auch in anderen Browsern auf?

gereons, 2023-08-02:
Tatsächlich passiert das nur in Firefox; Safari und Chrome sind OK, die benutze ich nur so gut wie nie. Ich wüsste im Moment nicht was oder wie ich an den Locale-Einstellungen im FF gedreht haben sollte. Das UI von Firefox ist jedenfalls deutsch... Danke für den Hinweis schonmal.

gereons, 2023-08-02:
Ah, habs gefunden. Aus mir unbekannten Gründen war in meinem FF-Profil `javascript.use_us_english_locale = true` gesetzt, vermutlich durch eine Extension. Löschen der Einstellung hat Abhilfe geschaffen. Danke!

closed 2023-08-02 by gereons

ledermann, 2023-08-03:
Wow, eine solche Ursache muss man erst mal finden ;-)