solectrus/solectrus #2226: Beim Wechsel der Charts wird mit den Daten der vorherigen Quelle weitergezeichnet

gereons, 2023-08-02:
Folgendes merkwürdige Phänomen ist mir schon vor einigen Tagen aufgefallen, jetzt hab ich einen Weg gefunden es zu reproduzieren:

Angenommen ich schaue mir gerade den "Jetzt"-Chart für die Erzeugung an. Das sah vorhin bei mir so aus:

[image]

Also relativ konstant etwas mehr als 3 kWh.

Dann wechsle ich über das Dropdown auf "Hausverbrauch":

[image]

... und der Chart wird mit den Werten die eigentlich eher zur "Erzeugung" passen weitergemalt.

Klicke ich dann oben in "Verwendung" auf den grauen Abschnitt normalisiert sich der Chart wieder und es geht mit realistischen Verbrauchswerten weiter.

[image]

Das Problem tritt in Firefox, Safari und Chrome auf.

ledermann, 2023-08-02:
Danke für die ausführliche Beschreibung, da scheint etwas mit dem Dropdown nicht korrekt zu laufen. Ich kümmere mich darum!

ledermann, 2023-08-03:
Der Fehler ist nun korrigiert. Beim Wechsel des Diagramms über das Dropdown werden die Messungen korrekt fortgeführt.

Die Korrektur wird im nächsten Update enthalten sein. In der Live-Demo ist es bereits jetzt zu sehen:
https://solectrus.ledermann.dev/inverter_power/now

closed 2023-08-03 by ledermann