solectrus/solectrus #2238: When using MQTT, the battery is sometimes 0%

ledermann, 2023-08-04:
I am facing the behaviour that the SOC of the battery show 0% from time to time and then a few mins later jumps back to the actual value:

[image]

As far as I remember, this was an issue when the MQTT collector does not get an updated value for some time for this data point. As the ioBroker plugin reads some of the values (e.g. SOC) only every 5 or 10 mins, this might lead to this behaviour.

_Originally posted by @loebse in https://github.com/solectrus/solectrus/discussions/2183#discussioncomment-6635770_

closed 2023-08-04 by ledermann

ledermann, 2023-08-04:
This is now fixed by using a less restrictive InfluxDB query: Instead of getting the last value within 5 minutes only, a much wider range is searched. This achieves that older values are also considered (if no more recent ones are available).

The fix will be available in the upcoming update (probably named `v0.12.1`).

@loebse: Because I'm not using the MQTT collector by myself, maybe you can try this out in your setup and confirm that my fix works. Simply use the `development` image by referencing `ghcr.io/solectrus/solectrus:develop` (instead of `:latest`) in your `docker-compose.yml`.

ledermann, 2023-08-11:
@loebse: Can you confirm that the bug is fixed (by using the `develop` image)? I would like to release an update in the next few days.

loebse, 2023-08-14:
I can confirm, it is working now @ledermann!

commit 55619db2 (2023-08-04): Use less restrictive query to determine the latest measured values

Resolves #2238