solectrus/solectrus #2239: Wärmepumpe als separaten Verbraucher darstellen

spirit0607, 2023-08-04:
Es wäre toll, wenn man eine Wärmepumpe als einzelnen Verbraucher in Solectrus darstellen könnte. Ähnlich die der Anzeige des E-Autos, bzw. einer Wallbox. 

Ich nutze zum Beispiel Solectrus mit dem MQTT Collector in Verbindung mit evcc und stelle mir vor, dass ich zukünftig zusätzlich den Stromverbrauch meiner Wärmepumpe mit einem separaten Stromzähler inkl. Modbus Adapter auslese und per MQTT an den MQTT Collector übermittle. Nun müsste noch der Stromverbrauch der Wärmepumpe vom Hausverbrauch abgezogen werden, damit dieser nicht doppelt ausgewiesen wird, da evcc aktuell keine Wärmepumpen o.ä. unterstützt.

Mit diesen Werten könnte man dann eine einzelne Darstellung einer Wärmepumpe oder eines Heizstabes realisieren.

ledermann, 2023-08-04:
Das ist ein spannendes Thema, vor dem auch ich bald stehen werde, da im Herbst eine Wärmepumpe bei uns einziehen wird - dann werde ich mich genauer damit beschäftigen.

Hier sind drei Dinge zu tun - wie Du schon erwähnt hast, ich fasse es nur noch mal kurz für mich zusammen:

1. Verbrauchswerte der Wärmepumpe in InfluxDB speichern (mit externen Tools, sehr individuell)
2. SOLECTRUS: Konfiguration des Feldes für die Wärmepumpe in InfluxDB (mittels `.env`)
3. SOLECTRUS: Wenn Wärmepumpe konfiguriert, dann dessen Verbrauch anzeigen und vom Hausverbrauch abziehen

Auf ähnliche Weise könnten weitere große Verbraucher konfiguriert werden. Es wäre daher sinnvoll, eine möglichst allgemeingültige Lösung zu bauen.

spirit0607, 2023-08-05:
Ja genau, das hört sich gut an. So könnte man evtl. auch eine zweite Wallbox als separaten Verbraucher darstellen und den Zweitwagen getrennt auswerten usw. 
Die Datenquellen werden in der Tat dann sehr individuell sein, daher dachte ich, dass eine MQTT Schnittstelle wahrscheinlich eine sinnvolle Möglichkeit wäre.

ledermann, 2023-08-05:
Bezüglich "zweiter Wallbox": Der SENEC-Speicher liefert in den Messwerten den Verbrauch der maximal vier Wallboxen separat. Der SENEC-Collector schreibt derzeitig jedoch nur die Summe nach InfluxDB. Das habe ich gerade geändert (https://github.com/solectrus/senec-collector/issues/219), sodass ab dem nächsten Update des SENEC-Collectors grundsätzlich die Werte für alle vier Wallboxen zur Verfügung stehen. Das UI muss dafür aber noch entsprechend erweitert werden (https://github.com/solectrus/solectrus/issues/2241)

Falls man also mehrere Original-SENEC-Wallboxen einsetzt, muss man dafür nicht auf MQTT wechseln. Für die Wärmepumpe bleibt es aber bei der oben skizzierten Lösung.

jbaesner, 2023-08-05:
Eine "generische" Lösung wäre hier in der Tat sehr wünschenswert, denn so könnte bspw. eine (zusätzliche?) über evcc+MQTT angebundene Wallbox als Verbraucher konfiguriert werden, der dann vom Hausverbrauch abgezogen werden könnte.

Auch andere Großverbraucher wie eine Wärmepumpe könnten dann ja entsprechend konfiguriert werden. Es wäre schön, wenn man dafür dann ggf. einen eigenen Namen + Symbol verwenden könnte?

StevieC121176, 2023-08-06:
Eine allgemeingültig Lösung würde ich auch bevorzugen. Ich denke aber bei mehreren Wallboxen und Verbrauchern wird es im Feld "Verwendung" schnell unübersichtlich. Momentan wird dort Wallbox, Hausverbrauch, Akkubeladung und Einspeisung angezeigt.
Könnte man "Verbrauch" nicht als Gesamtverbrauch anzeigen und z.Bsp. durch ein + Symbol eine detailliertere Ansicht öffnen, wo alle Verbraucher aufgeschlüsselt sind mit den jeweiligen Werten?

ledermann, 2023-08-06:
Danke für das allseitige Feedback :)

Ja, Konfiguration von Name und Symbol wäre sinnvoll. Das Problem der Übersichtlichkeit bei vielen Verbrauchern sehe ich auch. Mal sehen. Ich warte aber mit der Umsetzung, bis ich selbst den zusätzlichen Verbraucher (in Form einer Wärmepumpe) zur Verfügung habe - dann habe ich für mich einen praktischen Anwendungsfall.

spirit0607, 2023-08-06:
Alles klar, danke auch für die schnelle Rückmeldung. Würde mich über eine Umsetzung sehr freuen. Hier nur schonmal die Hardware, welche ich mir für das Projekt vorgestellt hatte zur Info:

Stromzähler:
https://www.amazon.de/Drei-Phasen-Elektro-Energiezähler-DIN-Schienen-Leistungsmesser-LCD-Display-Hintergrundbeleuchtung-MID-zertifiziert/dp/B08X96T1X2/ref=sr_1_1_sspa?crid=2S7UQX9G80QIF&keywords=sdm72d-m-2&qid=1691333429&sprefix=sdm72%2Caps%2C109&sr=8-1-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&psc=1

RS485-zu-LAN-Server:
https://www.amazon.de/RS485-zu-Ethernet-Seriell-Server-Router-zur-Schienenmontage-100-240VAC-PE11-H/dp/B09JC8G1P6/ref=sr_1_1?__mk_de_DE=ÅMÅŽÕÑ&crid=1PJUKGZE87OB3&keywords=pe11-h&qid=1691333480&sprefix=pe11-h%2Caps%2C102&sr=8-1

spirit0607, 2024-01-21:
Hallo,
hier nur mal zur Info: Ich habe seit kurzem die von mir gepostete Hardware im Einsatz und kann damit per Mqtt oder Modbus die aktuelle Leistung und den Verbrauch der Wärmepumpe abfragen, bzw. einsehen. Das würde also mit dieser Konfiguration ganz unabhängig vom angeschlossenen Verbraucher funktionieren. 

Gibt es denn schon Neuigkeiten bzgl. der Planung der Umsetzung?
Vielen Dank!

ledermann, 2024-01-21:
@spirit0607 Hier bei mir ist zwar mittlerweile die Wärmepumpe (Daikin) im Einsatz, die Abfrage von Verbrauchswerten direkt aus der Wärmepumpe funktioniert aber nicht bzw. nicht zuverlässig (großes Thema, das hier jetzt zu weit führen würde). Insofern läuft es auf einen separaten Stromzähler hinaus (geplant ist ein Shelly), der wurde aber noch nicht installiert. Ärgerlich, aber es soll bald passieren, sagt der Installateur. 

Momentan habe ich hier (bei knackiger Kälte in den letzten Wochen) einen erwartungsgemäß hohen Hausverbrauch, aus dem ich gerne die Wärmepumpe herausrechnen möchte. Theoretisch sehe ich da keine Probleme. Sobald der Stromzähler da ist, lege ich los. Vorher wäre das nur eine Trockenübung.

spirit0607, 2024-01-21:
Ja super, danke für die Rückmeldung. Ich habe eine Vaillant Wärmepumpe. Konnte zwar auch hier ein paar Daten mit iobroker oder Home Assistant auslesen, aber auch nicht wirklich zuverlässig. Ich denke über einen separaten Zähler ist das deutlich stabiler und genauer. 

Den höheren Verbrauch kenne ich gut 😉, aber dafür kein Gasverbrauch mehr. 

Freu mich auf die Umsetzung. 👍🏼

ledermann, 2024-02-20:
Lange hat es gedauert, bis ich mich diesem Thema zuwenden konnte, aber gestern wurde endlich vom Elektriker ein "Shelly Pro 3EM" in den Zählerkasten montiert 🎉 

Damit kann jetzt live der Stromverbrauch der Wärmepumpe ermittelt werden. Ich habe dann schnell einen [Shelly-Collector](https://github.com/solectrus/shelly-collector) gebaut, der alle 5s über die http-API den Verbrauch abruft und in die InfluxDB schreibt.

Nun habe ich das Dashboard so erweitert, dass der zusätzliche Verbrauchswert vom Hausverbrauch **abgezogen** und separat dargestellt wird. Die historischen Werte (also aus den zwei Monaten zwischen Installation der Wärmepumpe und Installation des Shelly) habe ich manuell importiert (nur ein Wert, kWh pro Tag). Das alles funktioniert auf den ersten Blick ganz gut und ist in der [Live-Demo](https://demo.solectrus.de/power/day) bereits zu sehen:

[image]

Das ist aber alles noch _Work-in-Progress_ - bevor ich ein Update veröffentliche, muss intern noch aufgeräumt werden. 

Wer das zukünftig nutzen möchte, kann sich aber schon mal überlegen, wie er/sie die Verbrauchswerte der Wärmepumpe (oder eines beliebigen anderen Großverbrauchers) nach InfluxDB bekommt. In meinem Fall ist es ein Shelly, aber je nach Gerät kann es da auch andere Möglichkeiten geben.

StevieC121176, 2024-02-20:
Gefällt mir sehr gut. Wäre es möglich die Werte der Wärmelampe auch per mqtt bereitzustellen, analog der „nicht Senec PV Anlagen“
um das somit in SOLECTRUS darzustellen?

ledermann, 2024-02-20:
@StevieC121176 Ja, absolut. Eine Erweiterung des [MQTT-Collectors](https://github.com/solectrus/mqtt-collector) erscheint mir sehr sinnvoll.

spirit0607, 2024-02-20:
Das hört sich super an und sieht auch richtig gut aus! Klasse! Freue mich schon auf das Update. Über die MQTT Unterstützung würde ich mich auch freuen. Mein Protoss PE-11 kann die Daten zwar auch per http zur Verfügung stellen, ich habe aber keine Idee, wie ich sowas wie deinen Shelly Importer dafür nutzen, bzw. umbauen könnte. Aktuell  ziehe ich die Daten per Modbus an HomeAssistant und würde diese dann per MQTT weiter an den MQTT Collector übermitteln. 

Bzgl. historischer Daten: 
Über die Vaillant App kann ich mir auch den Stromverbrauch der bereits vergangenen Tage als Excel exportieren. Müsst dann mal sehen, wie ich die dann in die Influx DB bekomme. Vielleicht bräuchte ich da mal Hilfe, aber wirklich tolle Arbeit! Vielen Dank! 😊

ledermann, 2024-02-20:
@spirit0607 Ja, wenn man sowieso schon MQTT verwendet, macht es natürlich Sinn, das auch für neue Dinge zu verwenden. Bei mir selbst sieht es genau andersherum aus: Ich benutze MQTT überhaupt nicht, da ist ein Shelly-Collector für mich einfacher (der letztendlich nur eine Variante des SENEC-Collectors ist).

Da es hier sowieso immer mehr Nicht-SENEC-User gibt, ist ein weiterer Ausbau des MQTT-Collectors für SOLECTRUS auf jeden Fall eine gute Sache.

Zeppelin-PV, 2024-02-20:
wenn ich das jetzt alles so lese, kommt mir folgende Idee
Wenn man den Shelly pro E3m auslesen kann, dann doch sicherlich auch nur eine Phase des Shelly, warum ?
Ich habe einen solchen Shelly in der Unterverteilung, und dabei misst der Shelly L3 was die Einspeisung meines BKW ist.
Nun wäre es ja nicht mehr weit den Anteil des BKWs links bei Erzeugung separat anzuzeigen, eben so wie rechts die Wärmepumpe.
was meint Ihr dazu ?
Georg, wäre das denkbar ?

ledermann, 2024-02-21:
@Zeppelin-PV Ja, das klingt machbar. Es muss nur geklärt werden, was auf der rechten Seiten passieren soll, wenn ich links etwas drauf addiere.

spirit0607, 2024-02-26:
Gerne bin ich beim Testen behilflich:

> - Wärmepumpe vorhanden und im Betrieb ;-)

Check

> - Stromverbrauch der Wärmepumpe kann in Eigeninitiative in die InfluxDB gebracht werden. Wer den Verbrauch der Wärmepumpe mit einem Shelly überwacht, kann hierfür den neuen [Shelly-collector](https://github.com/solectrus/shelly-collector) benutzen. Wer die Werte anders abgreift, muss derzeitig selber dafür sorgen, wie die Werte in die InfluxDB kommen. Den MQTT-Collector habe ich noch nicht angepasst.

Tja, davon habe ich bislang keine Ahnung. :-/
> - Verständnis von InfluxDB (und Docker und Linux natürlich)

Naja, ein ganz klein wenig

Das Problem ist nur, dass ich noch nicht so viel Ahnung habe, wie ich die Daten in die Influx bekomme (ist Telegraf vll. eine Möglichkeit von MQTT in influx zu schreiben?). Ich müsste mir auch erst noch einen Raspberry zum testen organisieren (würde ungern an meiner laufenden Solectrus Installation rumfummeln). Da komme ich aber kurzfristig ran.

Das andere Problem ist grade etwas die Zeit. Aber ich probiere da gerne was ich kann. Ich kann nur nicht versprechen, wie schnell das dann alles klappt und ob es überhaupt klappt,... :-/

**An dieser Stelle aber auch nochmal ein herzliches Dankeschön für die ganze Mühe, die ganze Zeit und den Ehrgeiz den du in dieses Projekt steckst. Ich weiß das wirklich sehr zu schätzen!**

ledermann, 2024-02-26:
@spirit0607 Danke für die Blumen und dein Angebot zu Testen :) Ich suche momentan nach einem einfachen Weg, das neue Feature möglichst bald veröffentlichten zu können, und somit einen Beta-Test kurz und schmerzlos abzuwickeln. Eine separate Testumgebung aufsetzen zu müssen, möchte ich niemandem zumuten.

Was jeder Interessierte eigentlich schnell und risikolos probieren kann, ist eine Umstellung des Docker-Images von `latest` auf `pr-2836`. Auch wenn ich intern einiges geändert habe, sollte das vollständig kompatibel zur aktuellen v0.14.2 sein. Und falls doch etwas nicht stimmt, stellt man halt wieder zurück auf `latest`. Die App greift ja nur lesend auf InfluxDB zu, das ist also ungefährlich. Damit die Wärmepumpe angezeigt wird, muss natürlich noch die oben erwähnte Konfiguration (`INFLUX_SENSOR_HEATPUMP_POWER`) ergänzt werden.

Um die Werte flott nach InfluxDB zu bekommen, ist es wohl doch besser, wenn ich den MQTT-Collector kurzfristig anpasse. Ich frage daher mal anders: Wer von den Nutzern des MQTT-Collectors kann denn bereits jetzt den Stromverbrauch seiner Wärmepumpe (in Watt) ohne Verrenkungen per MQTT bereitstellen? Dann würde ich ein Update des MQTT-Collector erstellen und die Sache wäre vollständig und einfach auszuprobieren.

spirit0607, 2024-02-26:
Okay alles klar. Das stimmt. Ich probiere morgen Abend mal ein wenig rum.

GrimmiMeloni, 2024-02-26:
> Ich frage daher mal anders: Wer von den Nutzern des MQTT-Collectors kann denn bereits jetzt den Stromverbrauch seiner Wärmepumpe (in Watt) ohne Verrenkungen per MQTT bereitstellen? Dann würde ich ein Update des MQTT-Collector erstellen und die Sache wäre vollständig und einfach auszuprobieren.

@ledermann wo siehst Du den Unterschied zur vorhandenen Einbindung, bei der evcc Werte als dedizierter Wallbox Verbraucher geschrieben werden? Das müßte doch komplett das gleiche in Grün sein, mal abgesehen von Source Wert aus MQTT und dem Zielwert in den Influx. Sofern ich da jetzt nichts übersehe sollte das unter "Fingerübung" laufen... 🤔

ledermann, 2024-02-26:
@GrimmiMeloni SOLECTRUS habe ich für die Erweiterung auf Wärmepumpe (stellvertretend für weitere zukünftige Verbraucher) etwas aufgebohrt und eine zusätzliche Abstraktionsschicht eingezogen, um beim Datenzugriff auf InfluxDB flexibler zu sein. Die von SENEC stammenden Begrifflichkeiten passten intern nicht mehr gut. Es gibt daher jetzt "Sensoren", die auf Measurements/Fields in InfluxDB abgebildet werden (voll abwärtskompatibel). 

Sinnvollerweise sollte sich das im MQTT-Collector widerspiegeln. Das muss aber nicht direkt gemacht werden, ich beschränke mich daher besser erst mal auf die Fingerübung und füge morgen einfach nur ein neues Feld hinzu, dann kann das jeder ausprobieren.

StevieC121176, 2024-02-27:
Ein super Feature, vielen Dank.
Über MQTT Collector könnte ich mit testen.

ledermann, 2024-02-27:
Der MQTT-Collector steht parat, kann also nun das neue Topic verarbeiten. Somit kann das jetzt von allen per MQTT getestet werden. Anleitung hier:
https://github.com/orgs/solectrus/discussions/2861

Bei Problemen mit der Installation bitte dort schreiben, nicht hier im Issue.

ledermann, 2024-02-28:
Nachdem das mit der Wärmepumpe gut funktioniert, habe ich heute nachgelegt. Interessenten für die Anbindung externer Wallboxen bitte hier entlang:
https://github.com/orgs/solectrus/discussions/2865

closed 2024-05-30 by ledermann