solectrus/solectrus #2241: Darstellung mehrerer Wallboxen

ledermann, 2023-08-05:
Mit dem nächsten Update des SENEC-Collectors werden Messwerte für jede Wallbox separat übermittelt:
https://github.com/solectrus/senec-collector/issues/219

Diese sollten angezeigt werden, sofern vorhanden.

q2doc, 2023-10-05:
Kann man das:

wallbox_charge_power0:

über MQTT auch schon "zweckentfremden" und so eine zweite Wallbox sichtbar machen oder geht es aktuell nur bei dem SENEC-Collector? Also analog zu dem forecast:
MQTT_TOPIC_WALLBOX_1_CHARGE_POWER=
MQTT_TOPIC_WALLBOX_2_CHARGE_POWER=

Robbe64, 2024-04-09:
Wenn man in EVCC zwei Wallboxen hat wird aktuell auch nur die erste berücksichtigt. Aktuell „verschwindet“ der Strom einfach.

ledermann, 2024-04-09:
@Robbe64 Ja, derzeitig wird nur eine Wallbox von SOLECTRUS berücksichtigt und angezeigt. Der SENEC-Collector übermittelt zwar die Werte der weiteren Wallboxen, es erfolgt aber noch keine Anzeige. Aufgrund der Arbeiten in #2865 ändert sich da sowieso noch etwas.

In deinem Screenshot sehe ich aber gar keine Wallbox, also auch nicht die Erste. Außerdem sehe ich da die Anzeige "Keine Verbindung". Das sollte nicht sein. Falls es da Probleme gibt, lege bitte ein Issue hier an, dann klären wir das: https://github.com/solectrus/hosting/issues

deadrabbit87, 2024-10-06:
@ledermann Ich weiß, die Fragen sind müselig, aber gibt es eine Roadmap wann man mit mehreren Ladepunkten rechnen _könnte_?

ledermann, 2024-10-06:
@deadrabbit87 Die Frage ist völlig verständlich. Da aber SOLECTRUS größtenteils eine One-Man-Show ist, versuche ich meine Ressourcen auf die Dinge zu konzentrieren, die den größten Benefit für alle User bringen. Ich habe dazu kürzlich eine Umfrage gestartet:
https://github.com/orgs/solectrus/discussions/3523

Momentan ist dieses Issue (Support für mehrere Wallboxen) kaum gefragt, daher hat es momentan keine hohe Priorität. Das kann sich aber ändern, je nachdem, wie sich das weitere Feedback so entwickeln. Du kannst das mit beeinflussen ;-)

deadrabbit87, 2024-10-13:
Sorry, ich nochmal. Gibt es denn eine Möglichkeit in der `.env` die Werte von zwei Wallboxen zu addieren? 

Aktuell funktioniert weder PowerSplitter noch die dazugehörigen Berechnungen, weil die Leistung der zweiten Wallbox einfach "verschwindet". 

Als Workaround wäre es hilfreich, wenn man einfach beide Wallbox addiert und diese dann zusammen in Solectrus dargestellt werden. 

In meinem Fall kommen die Werte von evcc. Leider gibt es dort keinen Datenpunkt, der beide Wallboxen summiert ausgibt.

q2doc, 2024-10-13:
Ich mache das schon länger indirekt über ioBroker (JavaScript Summe) und pushe es dann per MQTT. Fehlt halt leider die Aufschlüsselung nach Wallbox und die Differenzierung nach Auto. Aber sonst funktioniert es stabil.

deadrabbit87, 2024-10-13:
Danke - hätt ich auch wieder drauf kommen können. Habs jetzt in HomeAssistant umgesetzt.