solectrus/solectrus #2251: Top10 Anzeige niedrigster Tageswert

StevieC121176, 2023-08-08:
Mir ist eine Kleinigkeit aufgefallen, bei mir wird als niedrigster Tageswert der Tag der Solectrus Installation angezeigt (das war aber gegen Abend und somit sind nur ein paar Stunden Daten vorhanden.
Die Info unten interpretiere ich so, das nur ganze Perioden (sprich in diesem Fall Tage) zur Anzeige herangezogen werden.

[image]
[image]

ledermann, 2023-08-08:
Der Hinweis "Es werden nur vollständige Perioden berücksichtigt" ist vielleicht etwas missverständlich. Gemeint ist, dass die erste Periode (also hier: der erste Tag) seit Installation der PV-Anlage (siehe `INSTALLATION_DATE` in `.env`) ignoriert wird, also **nicht** die erste Periode seit Installation von SOLECTRUS.

Das macht dann einen Unterschied, wenn man **keinen** Datenimport durchgeführt, also eine Datenlücke besteht. 

In Deinem Fall besteht die Lücke wohl deshalb, weil der csv-Importer noch nicht mit KOSTAL-Daten umgehen kann (siehe https://github.com/solectrus/csv-importer/issues/69). Dazu bin ich leider noch nicht gekommen ...

StevieC121176, 2023-08-08:
Ah ok, danke für die Erklärung. Ich könnte mir erstmal das PV Installationsdatum in der `*.env` mit dem Installationsdatum von Solectrus gleichsetzen. Ich warte aber lieber auf den Kostal Importer.