solectrus/solectrus #2259: Status Meldung Batteriediagnose speichern

Zeppelin-PV, 2023-08-10:
In den Diskussionen unter Ideen habe ich schon mal erwähnt dass es gut wäre
den Status Batteriediagnose abzuspeichern mit Datum und somit der Dauer wie lange dieser Zustand dauert.
Leider bin ich seit gestern schon zum zweiten mal in diesem Jahr von einem Konditionierungsbetrieb betroffen.
Die Statusmeldung steht jetzt schon seit gestern Abend an.
Eine Auflistung wäre sinnvoll wenn man z.B. Belege für eventuelle Ansprüche braucht. 

- save status "battery diagnostic" with day and duration to get an overall list on this matter

ledermann, 2023-08-10:
Grundsätzlich wäre eine Auswertung nach Status denkbar, aber das ist ein sehr großes Thema. Auf die Schnelle hier mal eine Flux-Abfrage, die man direkt im InfluxDB-Fronten (im Data Explorer) ausführen kann und mit der die Statusmeldungen der letzten 12 Monate (gruppiert nach Anzahl) ausgegeben werden:

```
from(bucket: "SENEC")
  |> range(start: -1y)
  |> filter(fn: (r) => r._measurement == "SENEC" and r._field == "current_state")
  |> duplicate(column: "_value", as: "state")
  |> group(columns: ["state"])
  |> count()
  |> group()
  |> sort(columns: ["_value"], desc: false)
```

Bei mir ergibt das:

| Count  | State                  |
|-------:|:-----------------------|
| 9      | FEHLER LADEGERAET      |
| 94     | MAN. SICHERHEITSLADUNG |
| 119    | NEUSTART               |
| 122    | PASSIV                 |
| 123    | ISOLATIONSTEST         |
| 292    | SOFTWAREAKTUALISIERUNG |
| 386    | SELBSTTEST             |
| 1547   | BATTERIEDIAGNOSE       |
| 7422   | EIGENVERBRAUCH         |
| 75362  | AKKU LEER              |
| 174078 | LADESCHLUSSPHASE       |
| 209913 | NETZ + ENTLADEN        |
| 249989 | PV + ENTLADEN          |
| 334871 | AKKU VOLL              |
| 371507 | LADEN                  |

Ich fürchte, dies wird nun Begehrlichkeiten wecken :-) 

Solche Abfragen lassen sich mit Flux selber erstellen, wobei das aber nicht trivial ist. Ich selbst habe es bislang nicht geschafft, statt der Anzahl die Dauer des jeweiligen Zustands auszugeben.
Bitte möglichst selber damit experimentieren.

ledermann, 2023-08-10:
Endlose Möglichkeiten - hier noch eine Abfrage, um eine vollständige Liste zu erhalten, an welchen Tagen ein bestimmter Status aufgetreten ist und wie oft, hier am Beispiel `"BATTERIEDIAGNOSE"`:

```
from(bucket: "SENEC")
  |> range(start: -1y)
  |> filter(fn: (r) => r._measurement == "SENEC" 
                   and r._field == "current_state" 
                   and r._value == "BATTERIEDIAGNOSE")
  |> aggregateWindow(every: 1d, fn: count)
  |> filter(fn: (r) => r._value > 0)
  |> keep(columns: ["_value", "_time"])
  |> sort(columns: ["_time"], desc: true)
```

Zeppelin-PV, 2023-08-10:
Danke, Georg für die schnellen Vorschläge
,,, aber das traue ich mir dann doch nicht zu.