solectrus/solectrus #2280: Postgres container startet permanent neu

spyro2000, 2023-08-16:
Hi zusammen,

Solectrus hat einige Wochen hervorragend funktioniert aber dann war das Webinterface auf einmal nicht mehr erreichbar. Ich hatte erst meinen Raspberr Pi3 im Verdacht, CPU/RAM und SD Speed scheinen aber soweit ok zu sein, wenn auch nicht super-schnell (ca. 4 MB/s). 

Was mir auffällt, ist das der DB-Container für Postgres (solectrus-db-1) wohl ständig neu startet. Im Log des Containers stehen hunderte solche Einträge (sonst nichts):

```
PostgreSQL Database directory appears to contain a database; Skipping initialization
```

ledermann, 2023-08-17:
Die Meldung von PostgreSQL ("PostgreSQL Database directory appears to contain a database; Skipping initialization") ist ganz normal. Sie besagt beim Start des Containers, dass keine neue Datenbank angelegt wird, sondern eine bestehende weiter genutzt wird. Das ist kein Hinweis auf einen Fehler.

Merkwürdig finde ich jedoch die Ausgabe des Redis-Containers ("find: unknown user redis"). Hier scheint etwas nicht zu stimmen.

Der Einfachheit halber schlage ich vor, die Redis-Datenbank komplett zu löschen. Dabei geht nichts verloren, das ist nur der Cache, der danach wieder neu aufgebaut wird. Das geht wie folgt:

```
docker compose down
rm -rf /home/<user>/solectrus/redis
docker compose up
```

spyro2000, 2023-08-17:
Hallo @ledermann und danke für deine schnelle Antwort.

Ich habe das redis Verzeichnis jetzt neu angelegt (sicherheitshalber mit vollen Rechten) und nochmal ein `docker compose up` gemacht.

Das Ergebnis war leider dasselbe:

```
[+] Building 0.0s (0/0)
[+] Running 5/0
 ✔ Container solectrus-influxdb-1            Created                                                                                                    0.0s
 ✔ Container solectrus-db-1                  Created                                                                                                    0.0s
 ✔ Container solectrus-redis-1               Created                                                                                                    0.0s
 ✔ Container solectrus-senec-collector-1     Created                                                                                                    0.0s
 ✔ Container solectrus-forecast-collector-1  Created                                                                                                    0.0s
Attaching to solectrus-app-1, solectrus-db-1, solectrus-forecast-collector-1, solectrus-influxdb-1, solectrus-redis-1, solectrus-senec-collector-1
solectrus-db-1                  |
solectrus-db-1                  | PostgreSQL Database directory appears to contain a database; Skipping initialization
solectrus-db-1                  |
solectrus-redis-1               | find: unknown user redis

(haufenweise logs von solectrus-influxdb-1)

solectrus-db-1 exited with code 0
solectrus-redis-1 exited with code 0
dependency failed to start: container solectrus-redis-1 is unhealthy
```

ledermann, 2023-08-17:
Seltsam. Mir sind zwei Dinge nicht klar:

- Wie kommt es zur Fehlermeldung `find: unknown user redis` bei Redis? Ich konnte dazu nichts Erhellendes herausfinden.
- Warum beendet sich der PostgreSQL-Container, obgleich da gar kein Problem auftritt? InfluxDB läuft ja auch weiter.

Dennoch denke ich, dass wir uns zuerst Redis ansehen sollten. Mach doch mal bitte Folgendes, um nur Redis allein zu starten:

```
de /home/spyro/solectrus
docker compose down
rm -rf redis
mkdir redis
docker compose up redis
```

Welchen Output liefert das? Bei mir ergibt das:

```
pi@raspberrypi:~/solectrus $ docker compose up redis
[+] Running 1/0
 ✔ Container solectrus-redis-1  Created                                                                                                                                                                                                                                                                                                                                0.0s
Attaching to solectrus-redis-1
solectrus-redis-1  | 1:C 17 Aug 2023 12:11:40.358 * oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
solectrus-redis-1  | 1:C 17 Aug 2023 12:11:40.358 * Redis version=7.2.0, bits=64, commit=00000000, modified=0, pid=1, just started
solectrus-redis-1  | 1:C 17 Aug 2023 12:11:40.358 # Warning: no config file specified, using the default config. In order to specify a config file use redis-server /path/to/redis.conf
solectrus-redis-1  | 1:M 17 Aug 2023 12:11:40.365 * monotonic clock: POSIX clock_gettime
solectrus-redis-1  | 1:M 17 Aug 2023 12:11:40.372 * Running mode=standalone, port=6379.
solectrus-redis-1  | 1:M 17 Aug 2023 12:11:40.377 * Server initialized
solectrus-redis-1  | 1:M 17 Aug 2023 12:11:40.377 * Ready to accept connections tcp
```

Analog für PostgreSQL (compose-service `db`), aber ohne die Datenbank vorher zu löschen:

```
pi@raspberrypi:~/solectrus $ docker compose up db
[+] Running 1/1
 ✔ Container solectrus-db-1  Created                                                                                                                                                                                                                                                                                                                                   0.3s
Attaching to solectrus-db-1
solectrus-db-1  |
solectrus-db-1  | PostgreSQL Database directory appears to contain a database; Skipping initialization
solectrus-db-1  |
solectrus-db-1  |
solectrus-db-1  | 2023-08-17 12:15:25.612 UTC [1] LOG:  starting PostgreSQL 15.4 on aarch64-unknown-linux-musl, compiled by gcc (Alpine 12.2.1_git20220924-r10) 12.2.1 20220924, 64-bit
solectrus-db-1  | 2023-08-17 12:15:25.614 UTC [1] LOG:  listening on IPv4 address "0.0.0.0", port 5432
solectrus-db-1  | 2023-08-17 12:15:25.614 UTC [1] LOG:  listening on IPv6 address "::", port 5432
solectrus-db-1  | 2023-08-17 12:15:25.628 UTC [1] LOG:  listening on Unix socket "/var/run/postgresql/.s.PGSQL.5432"
solectrus-db-1  | 2023-08-17 12:15:25.656 UTC [24] LOG:  database system was shut down at 2023-08-17 12:04:44 UTC
solectrus-db-1  | 2023-08-17 12:15:25.701 UTC [1] LOG:  database system is ready to accept connections
```

Noch mal so als Referenz, wie es aussehen **sollte**.

spyro2000, 2023-08-17:
Hallo nochmal,

ok, hier der Output für Redis:

```
spyro@raspberrypi:~/solectrus $ rm -rf redis
spyro@raspberrypi:~/solectrus $ mkdir redis
spyro@raspberrypi:~/solectrus $ docker compose up redis
[+] Building 0.0s (0/0)                                                      [+] Running 2/2
 ✔ Network solectrus_default    Created                                 0.3s  ✔ Container solectrus-redis-1  Create...                              17.6s Attaching to solectrus-redis-1
solectrus-redis-1  | find: unknown user redis
solectrus-redis-1 exited with code 0
```

... und hier für Postgres:

```
spyro@raspberrypi:~/solectrus $ docker compose up db
[+] Building 0.0s (0/0)                                                      [+] Running 1/1
 ✔ Container solectrus-db-1  Created                                   14.6s Attaching to solectrus-db-1
solectrus-db-1  |
solectrus-db-1  | PostgreSQL Database directory appears to contain a database; Skipping initialization
solectrus-db-1  |
solectrus-db-1 exited with code 0
```

ledermann, 2023-08-17:
Kurios. Beide Container geben nicht mal die Versionsnummer aus. Lösche doch mal die beiden Images und versuche es dann nochmal, wodurch die aktuelle Version neu geladen wird. Also:

``` 
docker compose down
docker rmi redis:7-alpine
docker rmi postgres:15-alpine
docker compose up redis
docker compose up db

spyro2000, 2023-08-17:
Ja, Neustart und Dateisystem-Check habe ich schon hinter mir.

Ok, schade dass wir hier nicht weiter kommen. Ich danke dir aber trotzdem für die Unterstützung bis hierher :)