solectrus/solectrus #2283: Battery status 'FAIL'

Timokirch, 2023-08-17:
Hi Community,

since today my dashboard is frozen wit the status 'FAIL'.
[image]

I already restarted the docker host, recreated the images & restarted the battery unit. Any help on what I should do next for investigation would be awesome!

Thank you all in advance & kind regards
Timo

hambue, 2023-08-17:
Did the IP-address of your unit change somehow?
If not, what happens if you open your battery's IP-address in your browser. Do you see current usage and status?

ledermann, 2023-08-18:
Yes, as @hambue mentioned, SENEC seems to have changed something so that the collector is currently broken. I am working on it in solectrus/senec-collector#224.

@Timokirch Please take a look at the logs of senec-collector. I assume there are errors like "VARIABLE_NOT_FOUND has unknown prefix".

Timokirch, 2023-08-18:
Hi all, 

thanks for getting back on this so quick and sorry for not seeing the other issue before. My logs confirm that it's exactly your mentioned behaviour.
```shell
Getting state names from SENEC by parsing source code...
Error getting data from SENEC at 192.168.2.101: Failed to open TCP connection to 192.168.2.101:80 (Connection refused - connect(2) for "192.168.2.101" port 80)
``` 

I also run the requested CURL and the output was this:
```shell
{"STATISTIC":{"CURRENT_STATE":"VARIABLE_NOT_FOUND","MEASURE_TIME":"VARIABLE_NOT_FOUND"},"ENERGY":{"GUI_BAT_DATA_CURRENT":"fl_BE570A3D","GUI_BAT_DATA_FUEL_CHARGE":"fl_40814AFD","GUI_BAT_DATA_POWER":"fl_C11CBD1D","GUI_BAT_DATA_VOLTAGE":"fl_423A978D","GUI_GRID_POW":"fl_451CD9C2","GUI_HOUSE_POW":"fl_4546A706","GUI_INVERTER_POWER":"fl_4424C21D","STAT_HOURS_OF_OPERATION":"u3_00000E67"},"PV1":{"MPP_POWER":["fl_4424C21D","fl_00000000","fl_00000000"]},"TEMPMEASURE":{"CASE_TEMP":"fl_420EC006"},"WALLBOX":{"APPARENT_CHARGING_POWER":["fl_00000000","fl_00000000","fl_00000000","fl_00000000"]}}
```

ledermann, 2023-08-18:
Thanks a lot, @Timokirch! For further news, please see solectrus/senec-collector#224. I close this issue because it's a duplicate.

closed 2023-08-18 by ledermann