solectrus/solectrus #2285: Fehler 500

ledermann, 2023-08-18:
>Hallo, ich habe nach dem Update das Problem das mir nur der aktuelle Verlauf angezeigt wird. wenn ich etwas anderes auswähle kommt "Serverfehler 500", liegt das an meinem Senechspeicher (Version 1824) oder mache ich etwas falsch?
Der Fehler kommt auch wenn ich mich registrieren möchte.
>
>Danke für die Hilfe
>
_Originally posted by @struppie87 in https://github.com/solectrus/solectrus/discussions/2267#discussioncomment-6755054_

ledermann, 2023-08-18:
@struppie87: Nein, am Speicher dürfte das nicht liegen. Bei Fehler 500 vermute ich ein Problem mit der PostgreSQL-Datenbank. Bitte schaue mal in die Docker-Logs der App, dort steht genauer, was da schiefläuft.

awi69, 2023-08-18:
ledermann hat Recht, ich hatte das Problem auch nach meiner Erstinstallation gestern … nach löschen des postgresql Ordners und Neustart funktionierte alles Bestens und ich konnte die Installation registrieren. Strompreis kannst Du dann auchnin der GUI eintragen …

struppie87, 2023-08-18:
Hallo, ich hab jetzt die Logs geprüft.
Den Fehler habe ich selbst verursacht, hatte in der .env das Passwort geändert, nach dem es wieder das Original ist geht alles.

Vielen Dank

ledermann, 2023-08-19:
Genau so ist es: Beim ersten Start des PostgreSQL-Containers wird eine Datenbank angelegt und darin auch das in `POSTGRES_PASSWORD` festgelegte Passwort vermerkt. Ändert man dann später `POSTGRES_PASSWORD`, akzeptiert das die Datenbank nicht. 

Um ein anderes PostgreSQL-Passwort zu verwenden, muss man es also **vor dem ersten Start** setzen.

closed 2023-08-19 by ledermann