solectrus/solectrus #2288: Fehler 500

bgriese0, 2023-08-19:
Zuerstmal auf diesem Weg mein Dickes Lob. Ich bin gestern Abend über solectrus gesolpert und habe in diesem Projekt genau das gefunden, was ich mir selber mit ioBroker, influxdb, grafana und einigen scripten über Wochen selber gebaut habe. SUPER!

Nun zu meinem Problem. Ich zapfe einen Huawei SUN2000 Inverter über Modbus TCP an und speichere per Node-Red diese Daten in der Influx DB von ioBroker. Der MQTT Broker ist eingerichtet, läuft undliefert die Werte korrekt aus. Leider erhalte ich im Solectrus nun einen Fehler 500. Der Log der Container spuckt mir dabei diese Fehler aus.

solectrus-app-1                 | F, [2023-08-19T06:18:35.737516 #10] FATAL -- : [192.168.X.X]   
solectrus-app-1                 | [192.168.X.X] ActionView::Template::Error (percent must be between 0 and 100):
solectrus-app-1                 | [192.168.X.X]      5: turbo-stream action="morph" target="stats" children-only= true
solectrus-app-1                 | [192.168.X.X]      6:   template
solectrus-app-1                 | [192.168.X.X]      7:     - if timeframe.now?
solectrus-app-1                 | [192.168.X.X]      8:       = render StatsNow::Component.new calculator:, field:
solectrus-app-1                 | [192.168.X.X]      9:     - else
solectrus-app-1                 | [192.168.X.X]     10:       = render StatsRange::Component.new calculator:, timeframe:, field:

etwas weiter unten kommt dazu noch dieser Fehler:

solectrus-forecast-collector-1  | Error 500 "Internal Server Error"
solectrus-forecast-collector-1  | Sleeping for 900 seconds ...

Dieser dürfte zwar wenig mit dem Fehler weiter oben zu tun haben, das Ergebnis ist aber, dasss ich nichts angezeigt bekomme. Ich habe die Datenverzeichnisse bereits alle gelöscht und auch zur Sicherheit nochmal alles neu installiert. Das Ergebnis bleibt das gleiche.

Viele Grüße
Björn

ledermann, 2023-08-19:
Freut mich, dass du auf SOLECTRUS gestoßen bist :) 

Dein Setup ist ja aufgrund der beteiligten Komponenten durchaus komplex. Dennoch sollte der genannte Fehler natürlich nicht auftreten. Mir fehlen zur Behebung aber noch Infos:

- Beim Aufruf welcher Seite erscheint die Meldung? Bei JETZT oder bei einem der Zeiträume (Tag, Woche, Monat, Jahr, Gesamt)?
- Der Stacktrace müsste etwas größer sein. Dem Ausschnitt kann ich nicht genau entnehmen, wo der Fehler auftritt.

Was aber erkennbar ist, dass es sich wohl um einen der Badges handelt, die eine prozentuale Darstellung vornehmen. Das kann also nur Autarkie, Verbrauch oder Ladestand sein. Im Verdacht habe ich den Ladestand. Möglicherweise ist der bei dir größer als 100. Du könntest manuell in der InfluxDB prüfen, ob im Feld `bat_fuel_charge` sinnvolle Werte gelandet sind.

Die Fehlermeldung beim forecast-collector ist unabhängig davon. Kürzlich gab es bei https://forecast.solar einen Ausfall und der Server antwortete nicht korrekt. Das löst sich üblicherweise nach ein paar Stunden wieder von selbst.

bgriese0, 2023-08-20:
Ich habe mal ein bisschen getestet. Wenn ich den Wert in der InfluxDB manuell auf 0 setze, dann funktioniert es. Es ist in der Tat der Wert BAT_FUEL_CHARGE.

Sobald der Wert auf 1, 2 oder andere Werte geht, kommt der Fehler. Dar der wert denn eine Kommastelle haben?

Hier mal ein Set meiner Werte:

solectrus-mqtt-collector-1      | 2023-08-20 06:32:03 +0000 {"house_power"=>-547}
solectrus-mqtt-collector-1      | 2023-08-20 06:32:03 +0000 {"inverter_power"=>187}
solectrus-mqtt-collector-1      | 2023-08-20 06:32:03 +0000 {"case_temp"=>43.4}
solectrus-mqtt-collector-1      | 2023-08-20 06:32:03 +0000 {"bat_power_plus"=>340, "bat_power_minus"=>0}
solectrus-mqtt-collector-1      | 2023-08-20 06:32:03 +0000 {"bat_fuel_charge"=>10.0}
solectrus-mqtt-collector-1      | 2023-08-20 06:32:03 +0000 {"grid_power_plus"=>556, "grid_power_minus"=>0}

Wie gesagt, setze ich den Ladestand auf 0 funktioniert das Dashboard, aber halt ohne Akku. Hast Du eigentlich eine genaue Definition der Werte, also welche gültigkeitsrange die haben dürfen und was die Werte genau bedeuten? Ein Mapping zu den Huawei Registern könnte ich anbieten. Damit kenne ich mich recht gut aus. Der MQTT Collector funktioniert ansonsten super!

Viele Grüße
Björn

ledermann, 2023-08-20:
Danke für deine ausführliche Analyse!

Ich konzentriere mich nun auf `BAT_FUEL_CHARGE`. Dieser muss ein Float sein (weil InfluxDB einen Wechsel des Datentyps nicht erlaubt) und zwischen 0 und 100 liegen. Das scheint bei dir ja auch der Fall zu sein, insofern ist mir bislang nicht klar, was da schiefläuft. Ich habe gerade [die Fehlermeldung verbessert](https://github.com/solectrus/solectrus/commit/8e7a1eeaca9c572e0df001bc2bc103c674b7dcb1) und würde dich bitten, das `develop`-Image zu testen (in der `docker-compose.yml` einfach von `ghcr.io/solectrus/solectrus:latest` auf `ghcr.io/solectrus/solectrus:develop` wechseln und dann neu starten). Es wird genauso mit einem Fehler 500 abbrechen wie zuvor, aber es wird im Log ausgegeben, welcher Wert der Auslöser war. Diesen Ausschnitt aus dem Log hätte ich gerne - bitte mit vollständigem Stacktrace.

Eine genaue Beschreibung der Datenfelder in InfluxDB gibt es bisher nicht, die werde ich noch erstellen, danke für die Anregung!

Deine `.env`-Werte für den MQTT-Collector sehen auf den ersten Blick ok aus. Woher genau die Temperatur kommt, ist auch unerheblich.

Dass https://forecast.solar wegen zu vieler Requests meckert, dürfte an deinen Tests liegen. Durch häufiges Neustarten des Forecast-Containers wird immer wieder ein Request abgesetzt und dadurch das dortige Rate-Limit überschritten. Das würde ich erst mal nicht beachten - wenn die Container normal durchlaufen, verschwindet das Problem.

bgriese0, 2023-08-20:
Ich habe nochmal alle Verzeichnisse gelert und den Wert im Node Red nochmal mit Minimum und Maximum definiert. Aus irgendeinem Grund läuft es nun, auch mit korrekter Ladestandsanzeige. Wenn Du noch irgendwelche Logs oder Werte benötigst, dann schicke ich die gerne zu. Spiele gerne auch Testobjekt für den MQTT Client. 

Vielen Dank nochmal!

closed 2023-08-21 by ledermann