solectrus/solectrus #2289: Anzeige Konditionierungs-Modus

Zeppelin-PV, 2023-08-20:
gibte es die Möglichkeit den Konditionierungsmodus von SENEC auch in Solectrus anzuzeigen ?
Der Speicher selbst zeigt es nicht an; nur dass bei 70% der Akku voll ist.
Ist der Status Konditionierung nur in der SENEC App sichtbar ?
lg Ernst

[image]

ledermann, 2023-08-20:
Denkbar wäre das, der Speicher müsste das ja wissen. Allerdings ist mir nicht bekannt, wo genau sich diese Information versteckt und ob sie überhaupt von der API geliefert wird. Die API ist von SENEC ja nicht dokumentiert. 

Da mein Speicher sich derzeitig **nicht** im Konditionierungsbetrieb befindet, kann ich das nicht selber herausfinden. Du könntest mir aber helfen, indem du eine volle Datenabfrage bei deinem Speicher durchführst und mir das Ergebnis zukommen lässt.

Dies geht wie folgt:

- Logge dich auf deinem Raspi ein
- Führe nachfolgenden Befehl auf der Kommandozeile aus
- Poste das Ergebnis (etwa 30 KB JSON-Daten) hier oder schicke es mir per E-Mail. Letzteres ist vielleicht zu empfehlen, weil in dem Datenpaket auch Seriennummern u.a. stehen.
- Ich prüfe dann, ob ich den Konditionierungsstatus in dem Datenpaket finde und ergänze das dann im SENEC-Collector.

Hier nun der Befehl für eine vollständige Datenabfrage - `ipadresse` mit der IP-Adresse des SENEC-Speichers ersetzen:

```
curl \
  -X POST \
  -H "Content-Type: application/json" \
  -d '{
    "BAT1": {},
    "BAT1OBJ1": {},
    "BMS": {},
    "BMS_PARA": {},
    "CASC": {},
    "DEBUG": {},
    "DISPLAY": {},
    "ENERGY": {},
    "FACTORY": {},
    "FEATURES": {},
    "FILE": {},
    "GRIDCONFIG": {},
    "LOG": {},
    "PM1": {},
    "PM1OBJ1": {},
    "PM1OBJ2": {},
    "PV1": {},
    "PWR_UNIT": {},
    "RTC": {},
    "SELFTEST_RESULTS": {},
    "SOCKETS": {},
    "STATISTIC": {},
    "STECA": {},
    "SYS_UPDATE": {},
    "TEMPMEASURE": {},
    "TEST": {},
    "UPDATE": {},
    "WALLBOX": {},
    "WIZARD": {}
  }' \
  http://ipadresse/lala.cgi
```

Wenn sich die Information nicht finden lässt, wäre es theoretisch denkbar, aus Status "Akku voll" und "Ladezustand 70%" rückwärts zu schließen, dass wohl der Konditionsbetrieb aktiv ist. Das gefällt mir aber nicht, weil der Ladestand zwischendurch natürlich immer wieder unter die 70% fällt - man müsste also nach letztem Maximalwert suchen, der aber auch aus anderen Gründen 70% sein könnte. Außerdem könnte SENEC je nach Situation auch einen anderen Prozentwert wählen.

ledermann, 2023-08-27:
Alternativvorschlag zum Herausfinden, ob die API die benötigte Information liefert: 

Gehe im Browser auf `http://[ipadresse]/Vars.html` und versuche selbst, im Dickicht der vielen Messwerte diejenige herauszufinden, die auf Konditionierungsbetrieb hindeutet.

Zeppelin-PV, 2023-08-27:
... da ist man ja ganz nah am Herzen
aber ein Parameter der auf die Konditionierung hinweist, Fehlanzeige.
Interessant ist das hier; Parameter in denen der Akku-Typ festgelegt werden kann; BMZ-Samsung
[image]
Ich werde meinem Solateur und SENEC so lange auf den Wecker fallen bis hin zum Tausch auf Samsung Akkus,
die, so wie man hört, ohne ständige Konditionierung ihren Dienst tun.

Oggy512, 2023-08-31:
Es gibt im Block 
'BMS_PARA'

die Variable
'OPERATIONAL_MODE'

Die hat bei mir aktuell den Wert 03 (und ich bin im 70% Betrieb). Wenn man annimmt, dass diese Werte für entsprechende Modis stehen, könnte man den Konditionierungsbetrieb daraus ableiten.
Analog zu 'STATISTC' und dem dortigen 'CURRENT_STATE', dessen int Werte ja auch entsprechend übersetzt werden...

In dem gleichen Block gibt's auch noch 'FULL_CELL_VOLTAGE_MV'. Diese setzt mit ziemlicher Sicherheit die maximale Spannung, die ein Akkupack laden, darf fest. Bei mir aktuell mit Wert 3829.
Und im Log sehe ich dann z.B.:
Module #C considered as full with [3830 mV,64% SOC]. Charge is now prohibited.

Man könnte das gegenprüfen, wenn man jmd mit Speicher im Normalbetrieb hätte und dessen Variablen entsprechend gegenüberstellt..

ledermann, 2023-08-31:
@Oggy512 Ja, sehr guter Hinweis! Fragt sich nur, wie der Wertebereich ist. Mein Speicher befindet sich im Normalbetrieb (zumindest weiß ich nichts Gegenteiliges) - die Variable hat den Wert `02`. Hat noch jemand andere Werte?

Zeppelin-PV, 2023-08-31:
mein System fährt derzeit Konditionierung mit 70% Ladekapazität
hier mein Wert
[image]

Zeppelin-PV, 2023-08-31:
[image]

Oggy512, 2023-08-31:
> @Oggy512 Ja, sehr guter Hinweis! Fragt sich nur, wie der Wertebereich ist. Mein Speicher befindet sich im Normalbetrieb (zumindest weiß ich nichts Gegenteiliges) - die Variable hat den Wert `02`. Hat noch jemand andere Werte?

Das passt zu meiner Vermutung.
03 ist die 70% Abriegelung
02 dann scheinbar Normalbetrieb 

Ich hab im übrigen einen 7,5 kWh Speicher (somit 3 Batterie Module) und einen 3.0 Duo Inverter.

@Zeppelin-PV welche Speichergröße hast du? Wobei sich die Cell Voltage vermutlich auf das Bat Modul bezieht, insofern dann linear zu der Anzahl der Module ist, sprich jedes Modul wird darauf limitiert, daraus ergibt sich dann die 70%< unabhängig wie viele Module verbaut sind.

niceguy0815, 2023-09-08:
SENEC.Home V3 hybrid duo Mit 7,5KW momentan im 70% Konditionierungsmodus

[image]
[image]

WildesFried, 2023-10-01:
Hallo , ich habe die Werte auch geprüft 
[image]
[image]
SENEC.Home V3 hybrid duo Mit 7,5KW momentan im 70% Konditionierungsmodus

Beste Grüße

noffycws, 2024-05-08:
Nabend zusammen,
ich kann vielleicht auch noch Werte beisteuern. Meine Anlage war auch im Konditionierungsmodus COM70 und hatte den Wert 03. Seit 2 Tagen ist das System auf 50% gedrosselt (ROA50) und nun habe ich den Wert 01. Das kommt ja, wenn Safeguard etwas gefunden hat.

Dazu ist noch ein anderer Wert aufgetaucht:
[image]

Der Wert bei FULL_CELL_VOLTAGE_MV ist bei 3689

Vielleicht hilft das schon mal für die Betriebsanzeige. Andere Betriebe wie Full mode, 70% Drosselung und 50% Drosselung kenne.

Viele Grüße 
Michael