solectrus/solectrus #2305: Anzeige Firmware Version

Oggy512, 2023-08-27:
Hallo,
erstmal Kompliment zu dem Projekt, echt tolle Sache!

Wäre es denkbar, die aktuelle FW Version auszulesen und im Dashboard als Info anzuzeigen? Könnte ggf. interessant sein, da gerade bei den aktuellen Abschaltungen ja doch öfter Updates eingespielt werden, sodass man auf den ersten Blick sieht, welche Version installiert ist.
Ggf sogar mit Historie, sodass man sieht wann/wie lange welche Version aktiv war?

VG

ledermann, 2023-08-27:
Danke für die netten Worte :)

Und ja, guter Gedanke! Auf dem Dashboard könnte man dies dann auf einer separaten Seite als Liste darstellen, in der das jeweilige Installationsdatum aufgeführt wird, also z.B. so:

|Installationsdatum|Revision MCU|
|-----|-------|
|...|...|
|02.03.2023 09:42|3823|
|01.05.2023 17:49|3824|
|31.08.2023 10:42|3825|

Das Datum wäre der Zeitpunkt, an dem eine neue Version erstmalig gemeldet wird.

Ich frage mich aber, welches die wirklich relevante Versionsnummer ist. Bei mir sieht es momentan so aus:

[image]

Ich gehe davon aus, dass `Revision MCU` als relevante Information genügt. Ich möchte mich im UI auf das Wesentliche beschränken.

Oggy512, 2023-08-27:
Das klingt prima!
Ich denke auch, dass der Wert MCU mitunter der relevanteste sein dürfte. Ich meine mich zu erinnern, dass bevor der ganze Abschaltungsmist los ging im Senec WebIF auch nur dieser angezeigt wurde, Die anderen FW Versionen kamen nach und nach erst dazu.
Bei mir schaut das aktuell so aus:

[image]

ledermann, 2023-08-27:
Ok, prima! Ich habe zwischenzeitlich den SENEC-Collector so erweitert, dass der die `Revision MCU` abgreift und nach InfluxDB schreibt. Somit sind nun die Voraussetzungen gegeben, eine Liste mit den jeweiligen Installationszeitpunkten zu erstellen. 

Das wäre dann von mir im Dashboard noch zu implementieren.

hambue, 2023-08-29:
Rückblickend kann man übrigens die Logfiles mit folgendem Befehl nach MCU und GUI Updates durchsuchen:
`grep -rh "\[I|Settings\] MCU firmware" Logfiles | sort | uniq --skip-fields=2`

Das liefert diese Ausgabe:
`2023-03-30 06:40:59 [I|Settings] MCU firmware version: 1822, GUI version 1964
2023-04-24 13:30:12 [I|Settings] MCU firmware version: 1824, GUI version 1964
2023-08-15 10:56:42 [I|Settings] MCU firmware version: 1824, GUI version 8964`

`Logiles` ist in diesem Fall der Ordner mit allen Log-Dateien des Heimspeichers. 

Für die Integration in Solectrus natürlich nicht geeignet, aber vielleicht für den ein oder anderen hier trotzdem interessant.

ledermann, 2023-08-29:
Danke für die Ergänzung, @hambue! Ja, so kann man es natürlich machen, wobei man aber vorher noch sämtliche Logfiles aus dem Speicher herunterladen muss. Um nicht off-topic zu werden, habe ich das gerade mal hier beschrieben:
https://github.com/orgs/solectrus/discussions/2311

Zeppelin-PV, 2023-08-31:
[image]
SENEC selbst zeigt diese beiden Firmware-Werte an

ledermann, 2025-09-14:
Diese vor über zwei Jahren aufgekommende Idee möchte ich aus heutiger Sicht nicht mehr verfolgen.

Einerseits unterstützt SOLECTRUS via MQTT auch viele andere Stromspeicher und Wechselrichter, daher können die verschiedensten Geräte beteiligt sind, die alle ihre eigene Firmware-Versionen haben. Das zu vereinheitlichen sehe ich als zu problematisch (und gleichzeitig wenig gewinnbringend) an.

Auch möchte ich den Fokus von SOLECTRUS auf die Darstellung von Leistung, Energie, Kosten und Erträgen belassen. Ja, das ist eine Gratwanderung, aber SOLECTRUS soll eher nicht zu einem Debugging-Tool für Techniker werden.

Ich schließe daher das Issue. Für die genauere Analyse von SENEC-spezifischen Probleme möchte ich auf den [SENEC Log Downloader](https://github.com/solectrus/senec-log-downloader) verweisen.