solectrus/solectrus #2327: FAIL + Keine Daten vorhanden!

Collins1975, 2023-09-04:
Hallo Georg,

Seit 10:15 Uhr heute Vormittag erscheint "FAIL" bei mir und die Anzeige "leuchtet" rot. Die aktuellen Daten werden weiterhin richtig dargestellt nur drunter sehe ich nur "Keine Daten vorhanden!". Ich habe nichts an der Konfiguration geändert?

[image]

Unter Tag ist zu sehen das bis 10:20 Uhr alles normal lief:

[image]

Hast Du eine Idee woran das liegen kann? Solectrus läuft bei mir auf meine Synology v 7.1.1-42962

Vielen Dank im Voraus!

Collins

ledermann, 2023-09-04:
Sieht so als, als wäre das Update der SENEC-Firmware bei dir angekommen. Hier habe ich dazu etwas aufgeschrieben:

- https://solectrus.de/blog/2023-08-18-unterstuetzung-fuer-neue-senec-firmware/
- https://github.com/orgs/solectrus/discussions/2310

Collins1975, 2023-09-04:
Hi Georg,

Tja, was soll ich sagen... Wer lesen kann ist klar im Vorteil! Danke für die Info. Ich habe was zu tun!

VG

hendrik20062001, 2023-09-04:
Moin, ich habe das gleiche Problem, nur das bei mir keine Daten bei Jetzt angezeigt werden. Die Senec Anlage ist noch über HTTP erreichbar.

ledermann, 2023-09-04:
Erste Anlaufstelle bei derartigen Problemen sind die Logs, hier also die Logs des SENEC-Collectors. Die bekommt man wie folgt:

```
docker logs solectrus-senec-collector-1
```

Alles Weitere häng davon ab, was darin für Fehlermeldungen stehen.

niceguy0815, 2023-09-08:
Mich hat ma Mitwoch wohl auch das Update erreicht.
SENEC.Home V3 hybrid duo ist jetzt nur noch über https erreichbar.
Hab die genannten Änderungen gemacht und jetzt geht wieder alles DANKE!
[image]

closed 2023-09-18 by ledermann