solectrus/solectrus #2332: zwei Grafiken übereinander anzeigen

Zeppelin-PV, 2023-09-05:
wie schon in meinem Ideenkatalog #1874 beschrieben wäre es sinnvoll und hilfreich wenn man zwei Diagramme aufeinander legen könnte. So z.B. den Akkuladestrom und den Akku Ladestand. Seit gestern, nach dem WirrWar von Firmware Updates, habe ich festgestellt dass Senec massiv an den Ladekurven geschraubt hat. 
Auch andere Diagramme die im direkten Bezug zueinander stehen könnte man mit dieser Methode super vergleichen.
Sicher ein riesen Aufwand, aber ein "nice to have"

Zeppelin-PV, 2023-09-08:
unsere Anlage ist jetzt ein Jahr und eine Woche am produzieren.
Die Funktion mit zwei Grafiken aufeinander legen wäre auch super um z.B. den 7.9.22 und den 7.9.23 miteinander zu vergleichen.
Über die Jahre könnte man vielleicht sogar den nachlassenden Wirkungsgrad der Module erkennen
[image]
Bild 1 ist "nur" mit importierten Daten von Senec, damals noch kein Solectra
[image]

... um den 7.9.22 anzuzeigen braucht man viele Klicks
da kommen die Anfragen #2131 und #1931 ins Spiel

ledermann, 2023-09-08:
Zwei Tage miteinander zu vergleichen, macht meiner Ansicht nach nicht so viel Sinn. Auch wenn sie aus der gleichen Jahreszeit stammen, kann die Wolkenlage ja ganz anders sein. Wenn überhaupt, dann muss man längere Zeiträume miteinander vergleichen, also z.B. August 2022 mit August 2023. Also wie in #2131 beschrieben.

Aber auch das lässt vermutlich keinen Schluss auf die Degradation zu, sondern da spielt das Wetter eine viel zu große Rolle. Ein verregneter August drückt die Ausbeute nach unten, ohne dass die Module schlechter geworden sind.

Über Möglichkeiten, die Degradation darzustellen, habe ich schon mal länger nachgedacht. Herausgekommen ist dabei eine Erweiterung der Top 10, beschrieben hier:
https://solectrus.de/blog/2023-07-08-version-0-12/
Aber ob die Überlegung wirklich korrekt ist, wird sich erst nach einigen Jahren zeigen.

halmich, 2023-10-03:
> Über Möglichkeiten, die Degradation darzustellen, habe ich schon mal länger nachgedacht. Herausgekommen ist dabei eine Erweiterung der Top 10, beschrieben hier:

Zumindest in meinem Fall, lässt sich damit die Degradation nicht darstellen.
Ich habe 18 kWp Ost/West, dazu ein 15kW Wechselrichter, außerdem eine 70% Begrenzung (12,8 kW).
Deine Idee klappt m.E. nur, wenn der WR auch mind. die Peak-Leistung der Module hat und man nicht durch fehlenden Eigenverbrauch abgeriegelt wird.

Ich notiere mir immer an wolkenfreien (also quasi perfekten) Tagen den Ertrag in eine Excel Tabelle. So weiß ich, was ich in den Folgejahren ungefähr an Ertrag erwarten kann, wenn das Wetter wieder wolkenfrei ist. Natürlich hängt dies auch von der Temperatur ab...
[image]
Das sind die Daten aus 2 Jahren. Die Jahreszahlen in der Grafik haben keine Bewandnis.