solectrus/solectrus #2333: 500 after trying to login

flymann, 2023-09-05:
I wanted to enter the registration but unfortunately I'm not able to login anymore. 
A password was set in .env (`ADMIN_PASSWORD=***`)

Release 0.13.0

`I, [2023-09-05T14:20:26.274617 #12]  INFO -- : [192.168.178.24] method=POST path=/login format=turbo_stream controller=SessionsController action=create status=500 allocations=574 duration=1.89 view=0.00 db=0.00 user_agent=Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/115.0.0.0 Safari/537.36
F, [2023-09-05T14:20:26.276012 #12] FATAL -- : [192.168.178.24] `

ledermann, 2023-09-05:
Strange. Can you post more of the logs, please? It looks like the error message is in the lines **below** your extract.

Two general hints:

- Try to use a simple password without `#`, `$` etc. Or use quotes in the `.env` file, like `ADMIN_PASSWORD='s3cr#t$'`
- A change in the `.env` file requires the container to be restarted

flymann, 2023-09-10:
Sorry for the delay, please find below some more logs ...

`I, [2023-09-10T11:58:38.958411 #53]  INFO -- : [192.168.178.24] method=POST path=/login format=turbo_stream controller=SessionsController action=create status=500 allocations=574 duration=1.92 view=0.00 db=0.00 user_agent=Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/115.0.0.0 Safari/537.36
F, [2023-09-10T11:58:38.959581 #53] FATAL -- : [192.168.178.24]   
[192.168.178.24] ActionController::Redirecting::UnsafeRedirectError (Unsafe redirect to "http://100.98.245.70/inverter_power/2023-09-08", pass allow_other_host: true to redirect anyway.):
[192.168.178.24]   
[192.168.178.24] app/controllers/sessions_controller.rb:13:in `create'
[192.168.178.24] app/controllers/concerns/auto_locale.rb:14:in `switch_locale'
I, [2023-09-10T11:58:38.983907 #53]  INFO -- : [192.168.178.24] method=GET path=/500 format=turbo_stream controller=ErrorsController action=show status=500 allocations=6736 duration=23.34 view=22.84 db=0.00 user_agent=Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/115.0.0.0 Safari/537.36`

ledermann, 2023-09-10:
It seems there is an IP confusion. I can see two IP private addresses in your logs:

- `192.168.178.24`
- `100.98.245.70`

Can you identify them? Please ensure your `.env` includes the following line (insert the IP address you use in the browser to reach the device running SOLECTRUS):

```env
# Domain name or IP address of your host
APP_HOST=192.168.178.24
```

Don't forget to restart the container after editing the `.env` file.

closed 2023-09-18 by ledermann