solectrus/solectrus #2349: CSV Import Anzeige fehlender Daten

metzger304, 2023-09-10:
Nach dem Import von CSV Dateien, werden diese nicht in alle in Solectrus angezeigt.

Die letzten Werte, die anzeigt werden:
- ist der Monatsansicht der 01.01.2022
- in der Wochenansicht der 24.01.2022 
- in der Tagesansicht der 27.01.2022.

Es wurde, wie in https://github.com/solectrus/csv-importer beschrieben:
- die env.  Datei mit dem richtigen Installation-Datum angepasst
- und die Redis-Cache geflusht

Die Docker Container wurden auch schon neu gestartet.

Grüße
Thomas

ledermann, 2023-09-10:
Grundsätzlich ist es so, dass SOLECTRUS die Zeitauswahl per Vor-Zurück-Buttons einschränkt, damit man nicht zu Zeiträumen navigiert, in denen es gar keine Daten geben kann (also **vor** der Installation der Anlage oder in der Zukunft).

Relevant dafür ist der Eintrag `INSTALLATION_DATE` in der `.env`, der im Format `YYYY-MM-DD` angegeben werden muss. Wenn also deine PV-Anlage am 10. Dezember 2020 installiert wurde, muss der Eintrag lauten: `INSTALLATION_DATE=2020-12-10`.

In der `docker-compose.yml` muss unter `services.app.environment` der Eintrag `INSTALLATION_DATE` stehen, sonst bekommt der Container keinen Zugriff auf die Variable. Das ist in der per Hosting-Guide bereitgestellten Datei aber auch der Fall. Außerdem müssen nach einer Änderung in `docker-compose.yml` oder `.env` die betroffenen Container neu gestartet werden, damit sich die Änderung auch auswirkt.

Nach dem Import von Daten muss einmalig der Redis-Cache gelöscht werden, aber das hast du ja gemacht.

Wenn du das alles sichergestellt hast, habe ich noch keine Idee, was da falsch läuft. Prüfe mal bitte, ob die env-Variable wirklich im Container angekommen ist:

```
docker exec solectrus-app-1 env | grep INSTALLATION_DATE
```

Übrigens kannst du durch manuelle URL-Änderung auch Zeiträume ansteuern, deren Ansteuerung das UI verhindert, z.B. `http://[raspi-ip]/inverter_power/2020-12-31`. Werden da Messwerte angezeigt?

Ein weiterer Hinweis: Nach einem CSV-Import müssen sinnvollerweise auch die Preise angepasst werden, damit von Tag 1 an klar ist, was der Strombezug gekostet und die Einspeisung gebracht hat. Das Datum "Gültig ab" muss also richtig eingestellt sein.

Wenn das alles nicht weiter hilft, stelle bitte mal deine `.env` und `docker-compose.yml` bereit, vielleicht fällt mir dann etwas darin auf.

Zeppelin-PV, 2023-09-10:
... passende Frage hierzu
wenn bei mehreren Offlinezeiten von Solectrus (bedingt durch Senec Updates und NPU Fehler) die Senec csv Datei der betreffenden Woche eingelesen wird
... werden dann nur die Fehlzeiten importiert oder die komplette Woche von Senec im 5min Raster ?

metzger304, 2023-09-10:
> http://[raspi-ip]/inverter_power/2020-12-31

Hallo,

vielen Dank für die Ausführliche Antwort.
Was ich habe vergessen, zu erwähnen, dass alles  in einem Docker auf einer Synology NAS betrieben wird.

- Übrigens kannst du durch manuelle URL-Änderung auch Zeiträume ansteuern, deren Ansteuerung das UI verhindert, z.B. http://[raspi-ip]/inverter_power/2020-12-31. Werden da Messwerte angezeigt?
>  **### FUNKTIONIERT, es werden Daten angezeigt**

- docker exec solectrus-app-1 env | grep INSTALLATION_DATE
> es wird das Datum###  **INSTALLATION_DATE=2022-01-27 ausgegeben, was falsch wäre,**
denn in der .env steht drin:
># Date of commissioning of your photovoltaic system
**### INSTALLATION_DATE=2020-12-13**

wie finde ich die Docker-compose.yml?

Danke