solectrus/solectrus #2374: Unterschiedliche Zeiten Tag/Jetzt

GerhardPg, 2023-09-20:
In der Grafik zu "Jetzt" geht die X-Achse bis zur aktuellen Uhrzeit (z.B, 13:12) in der Grafik zu "Tag" ist bereits 1,5 Stunden weiter (z.B. 14:50). Habe ich irgendwo etwas falsch konfiguriert, z.B. im .env-File?

ledermann, 2023-09-20:
Klingt nach einem Zeitzonen-Problem. Hast Du vielleicht vorher Messwerte über den [CSV-Importer](https://github.com/solectrus/csv-importer/) importiert?

Eine Konfiguration, die ein solches Phänomen verursacht, kann ich mir momentan nicht vorstellen. Könntest Du vielleicht noch Screenshots ergänzen, an denen man das sehen kann?

ledermann, 2023-09-20:
Ich vermute, es handelt sich hier um ein Duplikat von solectrus/csv-importer#88

Falls ja, schließen wir dieses Issue und behandeln das Problem dort.

ledermann, 2023-09-23:
Ich gehe davon aus, dass es sich um einen Fehler handelt, der durch einen vorherigen CSV-Datenimport in Verbindung mit einem Zeitzonen-Problem ausgelöst wurde. Den CSV-Import habe ich zwischenzeitlich verbessert und aktualisiert. 

Ich schließe das Issue daher. Bitte Bescheid geben, falls das Problem dennoch weiterhin auftreten sollte.

closed 2023-09-23 by ledermann