solectrus/solectrus #2380: Swipen zum Datumswechsel

Robbe64, 2023-09-22:
Kleiner Verbesserungsvorschlag:
Wenn man Solectrus auf dem Handy offen hat wäre es toll in der Ansicht Tag/Woche/Monat/Jahr den eingestellten Zeitraum mit einer Wischgeste von rechts nach links bzw. links nach rechts ändern zu können.

Beispiel:
1. Tagesansicht steht auf 22. September
2. Mit dem Finger von links nach links nach rechts wischen
3. Tagesansicht wechselt afu den 21. September
Für die anderen Ansichten analog.

ledermann, 2023-09-23:
Ja, interessante Idee, dürfte auch leicht zu realisieren sein. Danke für den Vorschlag - und das Sponsoring :)

AlpenFlizzer, 2024-06-23:
Ich hätte in eine Branch eine Umsetzung mit Hammer.js.
 https://github.com/AlpenFlizzer/solectrus/commit/db5a98cb00737ef2183ba5aede5466ca081dcd85

Meinungen zum Code sind gerne willkommen.

ledermann, 2024-06-24:
Danke für den Code, @AlpenFlizzer! Ich habe das mal aufgegriffen und PR #3214 vorbereitet. Alles Weitere zur Implementierung besser dort.

PR #3214 (2024-06-24): Swipe control
Resolves #2380
closed 2024-06-24 by ledermann