solectrus/solectrus #2381: Darstellungsfehler vom Graph seit 0.13.0

max-horvath, 2023-09-22:
Moin,

Seit der letzten Version sieht der Graph (z.B. Erzeugung in kW) nach kurzer Zeit leider komisch aus, es sieht in etwa so aus, als wenn mehrere Graphen übereinander liegen würden.

Der Screenshot sollte es gut darstellen:
[image]

ledermann, 2023-09-22:
Das sieht natürlich wirklich falsch aus - aber ich kann das nicht reproduzieren. Kannst du genauer beschreiben, wann es zu dieser Fehldarstellung kommt? Also wie kann ich das provozieren?

ledermann, 2023-09-23:
@max-horvath Ich bin sehr daran interessiert, das Problem zu beheben, kann es aber nach wie vor nicht reproduzieren. Die Darstellung sieht danach aus, als würde beim Fortschreiben des Graphen etwas schieflaufen, vielleicht mit der Uhrzeit des nächsten Datenpunktes? Kannst du näher beschreiben, wie es zu dieser Darstellung kommt? Welchen Browser verwendest du?

Da die Version 0.13 seit einiger Zeit schon bei vielen Usern im Einsatz ist und sich bislang niemand mit einem solchen Problem gemeldet hat, vermute ich ein eher ungewöhnliches Setup.

max-horvath, 2023-09-25:
Hey @ledermann, ich werde schauen, dass ich Dir im Laufe der Woche hier weitere Infos zur Verfügung stellen kann. Was ich auf jeden Fall sagen kann ist, dass die Darstellung des Graphen beim Seitenaufruf korrekt ist und dann erst mit der Aktualisierung der Werte anfängt so komisch auszusehen.

Ansonsten habe ich ein ganz normales Setup mit RPi4 ...

Womit es vielleicht zusammenhängen könnte ist die Einstellung, dass ich den Senec-Speicher nur alle 10 Sekunden abfrage, da ich ihn auch per ioBroker abfrage und somit nicht überlasten möchte. Deshalb sieht man im UI auch regelmäßig "Fail", aber damit konnte ich bis jetzt gut leben. Hatte bis jetzt leider nicht die Zeit, Code für https://github.com/solectrus/solectrus/issues/1919 zu contributen.

max-horvath, 2023-09-25:
Die Darstellung des Graphen ist in jeglichem Browser so fehlerhaft ...

ledermann, 2023-09-25:
Der Hinweis auf ein 10-Sekunden-Intervall ist interessant, damit könnte es zu tun haben: Der SENEC-Collector hat ein konfigurierbares Intervall, das Dashboard von SOLECTRUS aber nicht - es aktualisiert alle 5 Sekunden die Anzeige, wenn man sich auf der Seite "JETZT" befindet. Eigentlich sollte das aber kein Problem ergeben, das Dashboard holt dann nur unnötig oft Werte aus der InflxuDB. Ein "Fail" wird angezeigt, wenn die letzte Messung mehr als 10 Sekunden her ist. Hier müsste #1919 mal erledigt werden, darum kümmere ich mich. 

Könntest du mal ausprobieren, ob das Problem verschwindet, wenn du das `SENEC_INTERVAL` auf 5 setzt? Eine Überlastung des SENEC_Speichers halte ich für sehr unwahrscheinlich. Ich frage meinen SENEC-Speicher mit zwei Geräten jeweils im 5-Sekunden-Takt ab, das klappt problemlos.

Eine ganz andere Vermutung: Dein Screenshot ist in Englisch, d.h. du hast im Browser als bevorzugte Sprache nicht Deutsch eingestellt. Möglicherweise sind noch andere (exotische) Lokaleinstellungen aktiv? Die Fortführung des Graphen erfolgt so, dass ein neuer Messpunkt mit der aktuellen Lokalzeit im Browser ergänzt wird. Kann es vielleicht sein, dass dein Browser eine falsche Zeit hat oder irgendein Zeitzonenproblem im Spiel ist? Ich könnte hier auch etwas verbessern, wüsste aber gerne, was genau bei dir die Ursache ist.

max-horvath, 2023-09-27:
So, Problem gelöst ... der Hinweis mit der Zeit war sehr hilfreich.

Da das Projekt ja auf einem Raspberry Pi läuft habe ich dann doch mal geschaut, was die Uhrzeit auf dem Gerät so sagt ... und die war ca. 20 Minuten in der Zukunft.

Habe dann nochmal sichergestellt, dass die Zeit von nun an regelmäßig per NTP abgeglichen wird.

Auf jeden Fall lag es nicht an einem speziellen Browser, Locale oder ähnlichem.

Jetzt könnte man das Issue einfach schließen oder wenn man das Problem ganz fancy lösen möchte noch einen Check einbauen, der die Zeit des Systems mit einem NTP-Server ab und zu vergleicht und auf eine zu große Abweichung hinweist. Aber ob das eingebaut werden sollte überlassen Dir.

Viele Grüße!

closed 2023-09-27 by ledermann

ledermann, 2023-09-27:
Bezüglich des 10 Sekunden-Intervalls: Ja, mir ist dieser Hinweis von ioBroker auch bekannt. Ich hatte da auch mal [nachgefragt](https://github.com/nobl/ioBroker.senec/issues/21#issuecomment-1552675884) und die Antwort erhalten, dass ich eben "Glück hätte", wenn es bei mir keine Probleme bei häufigen Abfragen gibt. 

Möglicherweise war das bei älteren Speichern so, aber mit meinem V3 habe ich überhaupt keine Probleme mit häufigen Abfragen. Außerdem: Öffnet man den integrierten Webserver mit `https://[ip]`, dann bombardiert dieser den SENEC-Speicher mit drei lala.cgi-Abfragen **pro Sekunde** (!)

Auch sind die Antwortzeiten des SENEC-Speichers minimal, ich messe da 8ms, auch bei dauerhafter Abfrage im 5-Sekunden-Takt. Das sieht nicht danach aus, als wäre der Webserver überfordert.