solectrus/solectrus #2384: Solectrus auf der Smartwatch, "wearable" Ansicht

StevieC121176, 2023-09-23:
Ich habe mal versucht Solectrus auf meiner Smartwatch zum laufen zu bringen und es funktioniert. Die Ansicht ist nat. nicht optimal. Man kann sie zwar vergrössern, aber das ist auch nicht das gelbe vom Ei.
Gerade wenn man unterwegs ist oder auch zu Hause mal schnell seine Anlage im Blick zu haben finde ich spannend.
Könnte man zusätzlich zur mobilen Ansicht eine Art "wearable" Ansicht machen wo nur die wichtigen Werte dargestellt werden. Vielleicht vorzugsweise nur Symbol und Wert oder als Kacheln?

Als Beispiel habe ich mal die Ansicht meiner zusätzlichen Balkonanlage angefügt.

[image]
[image]
[image]
[image]
[image]

ledermann, 2023-09-24:
Ein neuer Sponsor - vielen Dank :)

Ja, ich habe SOLECTRUS auf meiner AppleWatch auch mal ausprobiert und denke, dass zumindest der jetzige Zustand nicht ideal ist. Es funktioniert zwar prinzipiell, für das sehr kleine Display müsste die Darstellung aber deutlich simplifiziert werden.

ABER: Ich sehe (zumindest mit watchOS 9 oder 10) keine Möglichkeit, ein Icon für eine Web-Application auf den HomeScreen der Uhr zu legen, oder übersehe ich da etwas? Den Aufruf habe ich nur über einen Link geschafft, den ich mir selbst über die Nachrichten-App geschickt habe. Nicht ideal.

Die Darstellung erfolgt außerdem über den Webbrowser, der immer das Schließen-Icon und die Uhrzeit oben mit anzeigt - ich wüsste nicht, wie man das verschwinden lassen kann. Das Konzept "Progressive Web App", dass es für iOS gibt, scheint es für watchOS nicht zu geben.

Alternativ könnte man eine native App für die SmartWatch bauen, aber zum einen habe ich damit keine Erfahrung, zum anderen weiß ich nicht, ob der Aufwand gerechtfertigt ist. Für Android (Wear OS) und Apple (watchOS) müsste es auch separat entwickelt werden.

StevieC121176, 2023-09-24:
Ich hab das bei mir so gelöst, ein Kurzbefehl erstellt, der eine Internetseite öffnet. In diesem Fall die Solectrus Web Applikation. Diese Aktion dann auf den Action Button der Uhr gelegt. Ein Druck öffnet jetzt solectrus. 
Mit einem klick mehr geht es aber auch direkt über die Kurzbefehl App. Bei Android sollte ein Browser verfügbar sein.

Native App halte ich auch für übertrieben, besser wäre hier nur eine spezielle Web Ansicht für die Watch. Diese sollte auch nur die wichtigsten Infos anzeigen, wie Erzeugung, Hausverbrauch, Einspeisung usw. Evtl. noch einen Graphen für die letzte Stunde.

Als Hintergrund würde ich unbedingt schwarz nehmen, das streckt die Ansicht etwas und fügt sich besser den Gegebenheiten der Uhr ein (Ränder etc.)

Hier mal ein schnelles Beispiel:
Statt des Safari Icons könnte hier das entsprechende Icon für Erzeugung , Verbrauch usw. stehen.

[image]

StevieC121176, 2023-09-24:
Hier nochmal etwas besser angepasst

[image]

StevieC121176, 2023-09-24:
hier mal als Demo auf der Uhr. 

https://github.com/solectrus/solectrus/assets/113791299/c9a291cc-515a-40a8-9cfd-e8ae104b817b

ledermann, 2023-09-25:
Grandios, sieht sehr gut aus 👍 

Wie hast du diese Demo gebaut? Ich vermute, das ist einfach nur ein Dummy mit HTML/CSS, den du dann im Browser der Uhr anzeigen lässt. Das könnte ich im Prinzip nachbauen und in SOLECTRUS als Darstellung verwenden, wenn der Viewport sehr klein ist.

Was ich störend finde, ist die oberste Zeile mit dem X und der Uhrzeit. So sieht das bei mir auch aus und ich habe keinen Weg gefunden, das verschwinden zu lassen.

Außerdem: Der Aufruf über das Icon "Kurzbefehle" ist unschön. Auf der Apple Watch Ultra mit der Aktionstaste ist es sicher prima, aber ohne etwas fummelig. Gibt es wirklich keinen Weg, über die Kurzbefehle-App ein Icon zu bauen? Habe mich damit noch nicht viel beschäftigt.

StevieC121176, 2023-09-25:
Die Version von gestern war ein Dummy. Jetzt werden mir aber die realen Werte meiner Anlage angezeigt. praktisch die gleichen Werte die auch Solectrus nutzt (über den MQTT Collector) (sh. Video unten).
Ich habe dazu die Visualisierung 2.0 von iobroker genutzt. Im Prinzip ist es auch nur eine Webseite die man über den Browser der Uhr aufruft.
Das X und die Uhrzeit habe ich leider auch nicht wegbekommen. Aber durch den schwarzen Hintergrund ist das nicht weiter störend. 
Eigentlich ist es nur ein Hintergrundbild was ich mir mit Gimp gebastelt habe. Darauf werden dann die Werte und die Schrift über CSS und html angezeigt. Einziger Nachteil, die Anzeigereihenfolge kann man nicht beliebig ändern, man muss vorher wissen wie man es haben möchte und das Hintergrundbild dementsprechend aufbauen. 

Der Aufruf über den Action Button ist super, mit Siri müsste es auch gehen und eben die Kurzbefehle App. . Einen anderen Weg hab ich bis jetzt auch noch nicht gefunden

https://github.com/solectrus/solectrus/assets/113791299/7be69d60-6afa-4bc6-a92f-37c63eb6cfd3

StevieC121176, 2023-09-26:
Für die Watch ohne Action Taste gibt es noch die Möglichkeit unter WatchOS10 Solectrus über ein Widget zu starten, welches man sich oben Ansinnen kann. 

https://github.com/solectrus/solectrus/assets/113791299/5562db84-e5b7-435c-93ce-c0c904ed3e49

ledermann, 2023-09-29:
Vielen Dank für den umfangreichen und sehr nützlichen Input! Ich habe nun selbst damit experimentiert und deinen Layoutvorschlag nachbauen können. Damit das aber wirklich praxistauglich wird und sauber implementiert ist, müsste noch einiges an Arbeit investiert werden.

Abgesehen von der technischen Faszination: Ich bin mir nicht sicher, ob sich das wirklich lohnt. Ich wüsste nicht, ob ich selbst das überhaupt ernsthaft nutzen würde, da die Bedienung auf einer normalen AppleWatch (ohne ActionButton) auch mit WatchOS 10 doch etwas umständlich ist: Per Krone Widget auswählen, dann nochmal "Ausführen" anklicken, dann einige Sekunden warten. In der Zeit habe ich das iPhone längst aus der Tasche gezogen und sehe dort auch mehr auf einen Blick, ohne Scrollen zu müssen. 
Übrigens: Anstatt die Krone und ein Widget zu benutzen, kann man einen Kurzbefehl auch als Komplikation auf einem Ziffernblatt platzieren, dadurch wird der Zugriff noch etwas schneller.

Richtig cool wäre es, wenn man einzelne Werte von SOLECTRUS als "Komplikationen" in ein Ziffernblatt integrieren könnte. Aber das bräuchte natürlich eine native App.

Ich lasse das Thema mal etwas sacken ;-)

StevieC121176, 2023-09-30:
Ja ohne einen richtigen Browser auf der Watch ist das alles etwas schwierig. Das hat mich aber auf eine Idee gebracht. Evtl. kann man in Issue #1986 besprochene Kacheldarstellung und einen DarkMode kombinieren. 
Ich habe dazu mal etwas erstellt:

https://github.com/solectrus/solectrus/assets/113791299/f11287b0-f0ea-489c-940f-16457444c9f9

und hier mit einem Graphen für die letzten 30min

 
[image]

ledermann, 2023-10-17:
Danke, @StevieC121176 für die vielen Vorschläge und die äußerst schicken Entwürfe. Ich schätze das sehr! Aus zeitlichen Gründen kann ich die Implementierung leider nicht kurzfristig angehen.

Kachel-Modus (#1986) und Dark-Modus (#1859) sind dennoch weiter auf der Agenda :) Ich werde beizeiten darauf zurückkommen.

holiday-sunrise, 2024-09-01:
Wie kann man das einrichten ?

ledermann, 2024-09-01:
Die hier zu sehenden Screens sind Entwürfe bzw. Selbstbauten mit ioBroker des großartigen @StevieC121176. 

Mit iOS kann man zwar mit der App "Kurzbefehle" einen Safari-Aufruf hinterlegen, der dann auch auf der Apple-Watch aufgerufen werden kann (Krone drehen). Aber das ist wenig praktikabel und das Layout von SOLECTRUS ist auch noch nicht an das kleine Display angepasst.