solectrus/solectrus #2407: Verbrenner-Vergleich

ledermann, 2023-09-29:
Diskutiert in #2402:

Wer ein Elektroauto hat und an seiner PV-Anlage lädt, ist vermutlich auch daran interessiert, einen Kosten-Vergleich zu einem (fiktiven) Verbrennerauto zu ziehen.

Eine einfache Berechnung könnte so erfolgen:

"Du hast im gewählten Zeitraum 134 kWh ins E-Auto geladen. Damit konntest du 1.030 km weit fahren (wegen 13 kWh/100km). Ein Verbrenner hätte für die gleiche Strecke 72 L Diesel verbraucht (wegen 7L/100km) und dich 130 € gekostet (wegen 1,80€/L)."

Grafisch könnte das als simples Balkendiagramm dargestellt werden: Ein Balken zeigt das E-Auto, ein weiterer den Verbrenner. Die jeweiligen Kosten und der Unterschied würden in EUR dargestellt.

Benötigt werden dafür einige zusätzliche Angaben:

- Verbrauch des genutzten E-Autos in kWh/100 km
- Verbrauch eines fiktiven Verbrenners in L/100 km
- Spritpreisentwicklung im Zeitverlauf, ggfs. Ermittlung über eine API (z.B. https://creativecommons.tankerkoenig.de/), notfalls könnte man aber auch mit einem konstanten Wert rechnen.

Die Berechnung könnte noch verfeinert werden: Würde man einen Verbrenner fahren, würde man weniger Strom verbrauchen und somit weniger Strom einkaufen und mehr einspeisen. Beide Beträge (Reduzierung des Stromzukaufs sowie Erhöhung der Einspeisevergütung) könnten berechnet werden und würden die Verbrenner-Vergleichskosten reduzieren.

Damit die Berechnung erfolgen kann, müssen natürlich die einzelnen Ladevorgänge der Wallbox mit ihren Strommengen bekannt sein. Bei Verwendung des SENEC-Speichers in Verbindung mit der Original-SENEC-Wallbox liegen die Werte vor. Bei einer anderen Wallbox oder einem anderen Speicher müssen die Werte anderweitig ermittelt werden, das wäre aber ein separates Thema.

halmich, 2024-08-16:
Meine Idee für eine "schnelle" und einfache Umsetzung für Anwender, da nicht für jeden die entsprechenden Werte aus dem E-Auto abrufbar sind.

Der Anwender gibt folgende Daten in Solectrus ein (entweder in den Settings ähnlich dem Strompreis, oder über die .env):
- Verbrauch des E-Autos (z.B. 16 kWh / 100km), evtl. pro Quartal, da im Winter der Verbrauch höher ist.
- Verbrauch des (fiktiven) Verbrenners: z.B. 5 l / 100 km
- Preis für 1 Liter Diesel / Benzin (entweder über eine API (s.o.), oder vom User als Eingabe (z.B. 1,60 Euro)

Solectrus berechnet (im Hintergrund), wie weit man mit einer 1 kWh Strom elektrisch fahren kann (100 km / 16 = 6,25km) und was man dafür mit einem Verbrenner bezahlen müsste (5 x 1,60 / 100 * 6,25 = 0,50 Euro).
Man sieht die Ersparnis, 1 kWh Strom aus PV Energie spart 50 Cent Kosten an einer echten Tankstelle. 
Man sieht sogar: 1 kWh Strom aus dem Netz für z.B. 30 Cent spart ebenfalls Geld (bei den obigen Werten).

ledermann, 2024-08-18:
@halmich Ja, so wäre das machbar. Entschieden ist hier noch nichts, ich sammele noch Ideen. Gut finde ich den Vorschlag, auch bei fehlenden Auto-Messwerten rechnen zu können (durch manuelle Eingabe von Verbrauch und Fahrleistung).

Ein Detail nur: Anstelle von "wie weit kommt man mit 1 kWh Strom" würde ich eher vergleichen wollen, was 1 km Fahrt kostet. Das erscheint mir leichter verständlich zu sein.

Insgesamt sehe ich momentan drei Szenarien, die verglichen werden könnten:

1) Aktuelle, tatsächliche Situation: PV-Anlage mit E-Auto (PowerSplitter berechnet Anteil des grünen Stroms, siehe #2436)
2) Fiktive Situation: PV-Anlage mit Verbrenner (unter der Annahme einer identischen Fahrleistung und einem angenommenen Spritverbrauch)
3) Fiktive Situation: Keine PV-Anlage, aber E-Auto (wie 1, aber mit 100 % Netzbezug)

Für jedes Szenario würden die Kosten pro km berechnet werden und wären sehr leicht vergleichbar.

ledermann, 2026-10-18:
Nach einer intra-familiären Diskussion mit einem Verbrenner-Fahrer über die Energie-Kosten der Elektromobilität habe ich hier einen Prototypen gebaut, als zusätzliche Unterseite für neue Auto-Seite. Wann das in `develop` landet, ist noch offen. Dennoch möchte ich hier schon mal einen Screenshot zeigen:

[image]

Grundlage der Berechnung sind die in SOLECTRUS bereits vorhandenen Werte:

- Stromkosten der eigenen Wallbox 
- Stromkosten der Auswärtsladungen (manuell erfasst)
- Gefahrene Kilometer

Ergänzt habe ich noch die Erfassung von:

- Sonstigen Zahlungen (in meinem Fall genutzt für die erhaltene THG-Quote)
- Kraftstoff, Verbrauch (in L pro 100km), sowie Kfz-Steuer (pro Jahr) des fiktiven Verbrenners

Außerdem sind historische Preise für Benzin und Diesel hinterlegt - hierzu habe ich einen weiteren Collector in Arbeit, der solche Dinge aus öffentlichen Quellen einsammelt, auch historisch.

Erscheint euch das so korrekt? Oder mache ich da Denk- oder Rechenfehler? Das Fazit "der Verbrenner wäre 7x so teuer" klingt schon etwas wild, erscheint mir aber korrekt.

Und: Braucht man das in SOLECTRUS? ;-)

halmich, 2026-10-18:
Okay, das ist spannend. Muss ich mir mal in Ruhe anschauen und mit meinen Werten vergleichen. Das Fazit "7x so teuer" kann ich kaum glauben...
Zwei Fragen habe ich spontan:
Geht die entgangene Einspeisevergütung in die Betrachtung mit ein?
Sind Gestehungskosten der PV Anlage ein Thema?

ledermann, 2026-10-18:
@halmich Ja, die entgangene Einspeisevergütung ist berücksichtigt. PV-Strom, der zum Laden des E-Autos verwendet wird, ist also nicht kostenlos, sondern wird mit der Einspeisevergütung als Opportunitätskosten angesetzt. Netzstrom wird mit dem Bezugspreis berechnet. Es werden die historischen Strompreise und den historischen Sprit-Kosten verwendet.

Gestehungskosten der PV-Anlage berücksichtige ich dagegen bewusst nicht. Investition und Installation der PV-Anlage gehören für mich in die [Amortisationsrechnung](https://demo.solectrus.de/amortization) der PV, nicht in die laufenden Kosten des Autos. Würde das E-Auto nicht laden, könnte der Strom eingespeist werden – deshalb ist die entgangene Einspeisevergütung hier die relevante Größe.

Die etwa 7× beziehen sich außerdem nicht nur auf Strom vs. Kraftstoff. In der gezeigten Rechnung sind auch die tatsächlich erhaltene THG-Quote beim E-Auto (hier 950 €) und die Kfz-Steuer (hier 1.346 €) beim angenommenen Verbrenner berücksichtigt - und die machen sehr viel aus. Betrachtet man nur die Energiekosten, sind es in meinem konkreten Beispiel 2.439 € Strom gegenüber 9.460 € Kraftstoff, also knapp Faktor 4. Aber hey, die erhaltene THG-Quote und eingesparte Kfz-Steuer ist ja Realität :-)

ledermann, 2026-10-18:
Nach weiterem Nachdenken: Der Faktor 7 ist irreführend, das nehme ich wieder raus, denn:

Die Darstellung vergleicht nur die Kosten, die sich bei E-Auto und Verbrenner unterscheiden (Energiekosten, THG-Quote, Kfz-Steuer). Es gibt natürlich weitere Kosten wie Versicherung, Wartung etc., die hier nicht berücksichtigt werden. Würde man Kosten, die bei beiden Fahrzeugen in ähnlicher Höhe anfallen, auf beiden Seiten addieren, ändert sich die absolute Differenz kaum bzw. bei identischen Kosten gar nicht - der Quotient aber erheblich.

```
10805 / 1489 ≈ 7

aber (10805 + c) / (1489 + c) wird mit wachsendem c immer kleiner
```

Der Rest der Seite dürfte aber stimmen. Insbesondere die 9.316 € Ersparnis über die betrachteten sechs Jahre gegenüber dem angenommenen Verbrenner bleibt davon unberührt. Nur die Angabe eines Faktors ist bei dieser Art von Vergleich nicht wirklich aussagekräftig.

ledermann, 2026-10-20:
Hier der aktuelle Stand, verglichen werden nun **laufende Kosten insgesamt** (also inklusive Versicherung, Wartung usw.). Das ist ehrlicher, als nur die Energiekosten zu betrachten.

[image]

Aus meiner Sicht interessante Erkenntnisse daraus:

- Das E-Auto habe ich seit sechs Jahren, ein Verbrenner hätte mich bei gleicher Fahrtstrecke **über 9.000€ mehr** gekostet. Diesen E-Auto-Vorteil hätte ich geringer geschätzt. Wegfallende Kfz-Steuer und die THG-Quote machen viel aus.
- Eine PV-Anlage ist natürlich hilfreich beim E-Auto, bringt aber weniger, als gedacht. Auch ganz ohne PV hätte ich noch über 7.500€ gegenüber dem Verbrenner gespart. Ursache ist das Laden im Winter und auch die recht teuren Auswärtsladungen.

Um eine solche Darstellung in SOLECTRUS zu erhalten, benötigt man:

- Die geladene Energie über die eigene Wallbox für den gesamten Betrachtungszeitraum (der bekannte Sensor `WALLBOX_POWER`)
- Historisch erfasste Auswärtsladungen (Datum, Energie, Kosten) für den gesamten Betrachtungszeitraum
- Historische Kilometerstände, notfalls genügt ein Wert pro Monat (muss aber in InfluxDB stehen)
- Historische Strompreise und Einspeisevergütung in den Einstellungen von SOLECTRUS (hat man ja sowieso erfasst)
- Je E-Auto ein paar Angaben zum fiktiven Vergleichs-Verbrenner: Kraftstoffart (Benzin oder Diesel),  Verbrauch in Liter pro 100km und jährliche Kfz-Steuer.
- Historische Zahlungen für THG-Quote. Außerdem für Wartung, Versicherung, manuell in SOLECTRUS erfasst zu jedem E-Auto. Es wird angenommen, dass diese Kosten beim Verbrenner in gleicher Höhe angefallen wären.
- Historische SOC der E-Autos machen die Berechnung genauer, sind aber nicht zwingend nötig

Die historischen Spritpreise bringt SOLECTRUS selbst mit.

Das Feature ist noch nicht in `develop`, ich lasse das noch etwas bei mir lokal reifen.

deThommy, 2026-10-20:
Sieht schon ziemlich vielversprechend aus, auch wenn das für mich als gar nicht so wichtig ist (Dienstwagen), aber spannend allemal!

Du schreibst, dass Wartung etc. bei einem Verbrenner auch angenommen werden: die laufenden Kosten (Wartung, Verschleiß, sind aber ja bei Verbrenner deutlich höher als bei E-Auto. - mehr Teile die kaputt gehen können, Ölwechsel...) da gibt es doch bestimmt auch schon (nachweisliche) Quotienten zu, wie viel das bei einem e-Auto mehr ist, sodass sich das hier mittels Faktor berücksichtigen ließe?

halmich, 2026-10-20:
So, ich habe mir meine Zahlen angeschaut und komme auf grob ähnliche Werte. 
Bei deinem zweiten Diagramm fällt der Ausschlag durch die Verteuerung Benzinpreise gar nicht mal so stark aus. Scheint aber korrekt zu sein. Auch Beginn Ukraine-Krieg lässt sich erkennen. 
Ich fände glaub eine Trendlinie bei rot/grün noch sinnvoll, aber das ist Ansichtsache.
Bei mir macht PV zu nicht PV einen deutlicheren Unterschied aus, da ich noch nie auswärts geladen haben, außerdem eine Fahrleistung von ca, 17.000 km / Jahr habe.
Interessant wäre vielleicht noch so ein Wert: Bei einem Strompreis von 70 Cent wäre das Elektroauto gleich teuer wie der Verbrenner. 
Und auch wenn du Recht hast, dass THG-Prämie und KFZ-Steuer einberechnet werden sollen, so fände ich einen reinen Strom - Diesel/Benzin Vergleich auch spannend. Langfristig wird ja zumindest die KFz Steuer auch beim E-Auto anfallen.

@deThommy 
Bin mir nicht sicher, ob die Wartung bei Verbrennern viel höher ist. Bei VW hat der Kundendienst beim ID.3 600 Euro gekostet. Mit meinem Diesel war ich "früher" bei 300-500 Euro, inkl. Öl.  Bei einer langen Nutzungsdauer fällt bei einem Verbenner sicherlich einiges an (Zahnriehmen, Auspuff,...), bei dem Elektroautos fehlen da noch die Daten. unklar, ob hier Ausfälle der Elektronik / Sensoren teuer zu Buche schlagen.

ledermann, 2026-10-30:
Der Verbrenner-Vergleich ist jetzt in `develop` und lässt sich in der Live-Demo ausprobieren:
https://demo.solectrus.de/cars/savings

Gegenüber den Screenshots oben hat sich etwas getan:

- **Zeitraum:** Verglichen wird ein Monat, ein Jahr oder die gesamte Zeit. Das Balkendiagramm zeigt die Jahre bzw. die Monate, ein Klick darauf öffnet den jeweiligen Zeitraum.
- **Auto-Seite:** Auf der [Auto-Seite](https://demo.solectrus.de/cars/all) steht die Ersparnis jetzt direkt unter den Fahrwerten (für Monat, Jahr und Gesamt). Ein Klick darauf führt zum Vergleich desselben Zeitraums.
- **Rechenweg:** Jede der drei Spalten (E-Auto, E-Auto ohne PV, Verbrenner) zeigt ihre Rechnung Schritt für Schritt. Die Energiekosten stehen dabei getrennt von den übrigen Kosten, jeweils auch pro 100 km.
- **Reiner Energievergleich** (@halmich): Das Liniendiagramm zeigt Stromkosten gegenüber Spritkosten pro 100 km, also ohne Kfz-Steuer, THG-Quote und sonstigen Kosten.
- **Sonstige Kosten:** Zahlungen werden je Auto mit einer Kategorie erfasst. Versicherung, Wartung usw. bekommt der Verbrenner in gleicher Höhe. Sobald die Kfz-Steuer auch beim E-Auto anfällt, erfasst man diese ganz normal bei den Kosten. Kauf, Finanzierung, Förderung und Verkauf können erfasst werden, bleiben beim Vergleich aber außen vor. 
- **Kein Faktor:** Die Seite zeigt nur Differenzen, kein „x-mal so teuer“
- Ist man nicht als Admin eingeloggt, bleiben die Zahlen für Gestern+Heute unsichtbar.

Zu euren Vorschlägen:

- **Wartung beim Verbrenner** (@deThommy): Die nehme ich bewusst gleich hoch an. Belastbare Zahlen dafür kenne ich nicht, und die Beispiele von @halmich zeigen, dass es auch umgekehrt sein kann. Ein pauschaler Faktor wäre geraten.
- **Trendlinie:** Habe ich ausprobiert und wieder verworfen. Beim E-Auto würde eine Trendlinie vor allem die Jahreszeiten abbilden (Laden im Winter teurer) und erklärt eher wenig.
- **Break-even-Strompreis:** Interessante Idee, aber vorerst nicht umgesetzt.

Voraussetzungen, damit die Seite erscheint:

- Die historischen Spritpreise in InfluxDB. Die liefert ein neuer Collector, der noch nicht veröffentlicht ist. Bis dahin bleibt der Vergleich in einer normalen Installation ausgeblendet, auch mit der aktuellen `v1.4-beta-471` ist bei euch also noch nichts zu sehen.
- Die Währung muss EUR sein, weil die Spritpreise nur in Euro vorliegen
- Ein Kilometerstand des Autos in InfluxDB. Eine manuelle Eingabe ohne Sensor gibt es nicht.

Feedback zur Demo ist sehr willkommen, besonders wenn euch eine Zahl merkwürdig vorkommt.

ledermann, 2026-11-04:
Der Verbrenner-Vergleich lässt sich jetzt auch in der eigenen Installation nutzen. Die historischen Spritpreise liefert der neue [public-data-collector](https://github.com/solectrus/public-data-collector), und [HELIOS v1.6.0](https://github.com/solectrus/helios/releases/tag/v1.6.0) richtet diesen ein.

Voraussetzungen:

- Das Dashboard läuft auf dem Entwicklungskanal (`develop`, also v1.4-beta)
- Mindestens ein Auto mit Kilometerstand-Sensor (Akkustand alleine genügt also nicht)
- Land in der EU. In den Grundeinstellungen von HELIOS gibt es dafür eine neue Frage nach dem Land (siehe Einstellungen/Grundeinstellungen). Solange sie nicht beantwortet ist, leitet HELIOS das Land aus der Zeitzone ab.
- Währung EUR
 
Die Spritpreise stammen aus dem [Weekly Oil Bulletin](https://energy.ec.europa.eu/data-and-analysis/weekly-oil-bulletin_en) der EU-Kommission. Das wird leider nur im XLS-Format (!) angeboten, der Collector muss diese daher kreativ parsen. 

Nutzung:

1. HELIOS auf `v1.6.0` aktualisieren, was üblicherweise automatisch passiert. Sind die Voraussetzungen erfüllt, erscheint der neue Dienst "Public-Data-Collector", aber noch nicht gestartet.
2. Diesen Dienst in HELIOS bitte einmal starten. Beim ersten Start schreibt er die Spritpreise rückwirkend ab dem frühesten Zeitstempel, das in der InfluxDB gefunden wird. Danach werden einmal pro Woche (Freitags um 06:00) die neuen Preise geholt. Der Dienst läuft also permanent als Docker-Container mit, schläft aber die meiste Zeit. Der Collector legt die Kraftstoffpreise in einem festen Measurement namens `Fuel` ab. 
3. Im Dashboard (`v1.4-beta-587` ist aktuell) dann unter Einstellungen/E-Autos für jedes Auto den Vergleichs-Verbrenner eintragen: Kraftstoff (Benzin, Diesel oder Autogas), Verbrauch in l/100 km und die jährliche Kfz-Steuer. Ab jetzt lassen sich dort auch Zahlungen erfassen, etwa die THG-Quote, Versicherung oder Wartung. Es kann bis zu 5min dauern, bis diese Einstellmöglichkeiten zur Verfügung stehen, weil sie erst angeboten werden, wenn wirklich Kraftstoffpreise in InfluxDB gefunden werden.

Danach erscheint der Verbrenner-Vergleich unter "Ersparnis" auf der Auto-Seite, und sollte etwa so aussehen wie in der Live-Demo:
https://demo.solectrus.de/cars/all
https://demo.solectrus.de/cars/savings/all

Wer gerne YAML-Dateien bearbeitet: Auch ohne HELIOS lässt sich der neue Collector betreiben, die [Doku](https://solectrus.de/docs/referenz/public-data-collector/konfiguration/) beschreibt, wie das geht. HELIOS ist für SOLECTRUS aber die Zukunft, was Konfiguration angeht. Wer das noch nicht hat: https://solectrus.de/install/

Der neue Collector kann auch weitere öffentlich verfügbare Daten abholen: Strompreise der Strombörse (Day-Ahead), Anteil erneuerbarer Energien im Netz, CO2-Intensität des Netzstroms (nur für Deutschland) sowie Preise für Erdgas und Heizöl. SOLECTRUS nutzt diese Werte noch nicht, deshalb schaltet HELIOS vorerst nur die Spritpreise ein, die anderen Dinge werden später mal genutzt.

Feedback ist weiterhin sehr willkommen, besonders wenn euch eine Zahl merkwürdig vorkommt. Ich schließe das Issue hier. Bei konkreten Problemen oder Fehlern bitte separates Issue anlegen oder Beitrag im Forum erstellen.

closed 2026-11-04 by ledermann