solectrus/solectrus #2409: Einspeisebegrenzung wird mit 0% angezeigt

ledermann, 2023-10-01:
Version 0.13.1:

> Da ich keinen Sennen habe bekomme ich die Daten vom MQTT Collector. Mit der Version 13.1 bekomme ich nun ein "!" hinter dem Einspeisewert. Im Tooltip steht Begrenzung 0%. Ich habe generell keine 70% Regelung. Muss ich da noch irgendetwas in der `.env` ändern?
[image]

_Originally posted by @StevieC121176 in https://github.com/solectrus/solectrus/discussions/2408#discussioncomment-7157149_

ledermann, 2023-10-01:
Die Update-Beschreibung ist noch nicht fertig, da kommen schon die ersten Fehlermeldungen ;-)

@StevieC121176: Wenn keine Einspeisebegrenzung aktiv ist, sollte sich gar nichts verändern, es handelt sich also um einen Fehler. Ich kümmere mich drum.

ledermann, 2023-10-01:
Ich habe das gerade im Developer-Image korrigiert: Wenn das Feld `power_ratio` in der InfluxDB fehlt (z.B. bei Verwendung des MQTT-Collectors), wird dies nun korrekt behandelt. Das heißt, die Darstellung ist wie vorher, also ohne "!" und ohne "Begrenzung 0 %" im Tooltip.

@StevieC121176 Könntest du ausprobieren, ob das bei dir funktioniert? Ich mache dann ein weiteres Update.

StevieC121176, 2023-10-01:
@ledermann funktioniert wieder wie vorher mit der develop Version. Keine 0% mehr im Tooltip und kein "!"

ledermann, 2023-10-01:
Alles klar, danke für die Rückmeldung, dann erstelle ich nun ein weiteres Update (Version 0.13.2).

closed 2023-10-01 by ledermann