solectrus/solectrus #2423: Hohe GPU-Auslastung und Stromverbrauch in Live-Ansicht

Robbe64, 2023-10-04:
Habe mich gewundert, warum mein Lüfter am Notebook in letzter zeit so stark läuft.
Dabei ist mir aufgefallen, dass für den Browser im Taskmanager eine hohe GPU-Auslastung und "Stromverbrauch" anzeigt wird wenn Solectrus geöffnet und in der Live-Ansicht ist.
[image]
Getestet habe ich das im Chrome und Firefox.
Hätte höchtens erwartet, dass nur alle 20 Sekunden wenn neue Daten kommen die Auslastung kurz steigt.

ledermann, 2023-10-04:
Oh, danke für den Hinweis! Das soll nicht sein, ich werde das genauer untersuchen und abstellen.

Als Ursache vermute ich die Fluss-Animation auf dem Tab "JETZT". Die Auslastung in den anderen Tabs ist minimal. Lösung könnte sein, die Animation anders zu regeln.

ledermann, 2023-10-04:
Ich habe das nun behoben. Ursache waren tatsächlich CSS-Animationen auf der JETZT-Seite, und zwar die Fluss-Animation als auch das pulsierende Icon in der Status-Anzeige. Beides ist verzichtbar und wurde entfernt. 

In der [Live-Demo](https://solectrus.ledermann.dev/) ist es bereits zu sehen, die Änderung wird im nächsten Update enthalten sein.

closed 2023-10-04 by ledermann