solectrus/solectrus #2427: Zeit in Container - forecast-collector "verschoben"

q2doc, 2023-10-04:
Hi,

Da mein erster Post - tolles Projekt! Tolle App! --> VIELEN DANK

Meine Konfiguration die relativ problemlos einzurichten war (außer ein, zwei Laien-Fragen, die die wirklich sehr gute Beschreibung ergänzen würden und ich gerne an anderer Stelle teile): Raspberry Pi mit Docker (& Portainer), ioBroker mit MQTT und dahinter eine SMA mit SHM2 und BYD.

Nun zu meinem Problem (wobei ich mir nicht sicher bin ob es auf den forecast-collector beschränkt zutrifft) - alles läuft und stellt toll dar (außer der Temperatur welche warum auch immer nicht per MQTT collector kommt) - ABER die Zeitzonen in den Container sind nicht local time sondern 2 Stunden zurück. Die App selber scheint trotz Versatz die richtige Anzeige zu wählen aber der forecast-collector ist definitiv exakt die 2 Stunden off (screenshot war um kurz nach 18 Uhr):

[image]

Der Versuch die Zeit in den Container über environment: -TZ=Europe/Berlin und Volumen: /etc/localtime:/etc/localtime:ro hart zu setzen scheint zwar zu funktionieren, ändert aber leider nichts am Ergebnis. Auch der Versuche anstatt 3 Dachflächen nur eine zu konfigurieren:

[image]

selbes Ergebnis. Was mache ich falsch?

Gruß
Marc

q2doc, 2023-10-04:
Fehler gefunden... die lat/lon Angaben hatten mehr als 2 Ziffern nach dem Komma. Dann bin ich auf einmal nicht mehr in Deutschland sondern in Afrika 😳

closed 2023-10-04 by q2doc

ledermann, 2023-10-05:
Danke für deine präzise Beschreibung - und gut, dass du das Problem bereits selbst hast beheben können. Ich frage mich zwar, wie zusätzliche Ziffern **nach** dem Komma eine Ortsverschiebung nach Afrika bewirken können, aber grundsätzlich ist die Angabe korrekter Geodaten natürlich essenziell für die Solarprognose.

Die anderen erwähnten Probleme bzw. Verbesserungsvorschläge kannst du gerne in separaten Issues beschreiben, daran bin ich immer sehr interessiert.