solectrus/solectrus #2430: Temperatur über MQTT fehlt

q2doc, 2023-10-05:
Wie im anderen Beitrag (https://github.com/solectrus/solectrus/issues/2427) geschrieben läuft es soweit sehr gut nur wird leider die Temperatur nicht angezeigt. Alles andere holt er sich korrekt über MQTT - leider scheitere ich bisher an der Fehlersuche...

StevieC121176, 2023-10-06:
Funktioniert bei mir ohne Probleme. 

MQTT_TOPIC_CASE_TEMP=Solectrus/bydhvs/0/State/TempBatt

q2doc, 2023-10-06:
Danke - dann weiß ich zumindest das es theoretisch geht... (also mit "funktioniert ohne Probleme" gehe ich davon aus, dass Du es auch angezeigt bekommst wie in der Demo 😉).
Es scheint auch wohl nicht an den Daten im ioBroker zu liegen, seit gestern habe ich zwei Wallboxen und wollte vorhin (bis dann auch über MQTT mehr als eine geht https://github.com/solectrus/mqtt-collector/issues/46) den Verbrauch zum Laden einfach als Summe einspielen - Funktion in ioBroker über Javascript läuft, Publishing auch aber der MQTT_collector holt sich warum auch immer den Wert nicht. Interessanterweise funktioniert jedoch der Austausch der beide Loadpoints...

MQTT_TOPIC_WALLBOX_CHARGE_POWER=evcc/loadpoints/1/chargePower --> funktioniert 😀
MQTT_TOPIC_WALLBOX_CHARGE_POWER=evcc/loadpoints/2/chargePower --> funktioniert 😀
MQTT_TOPIC_WALLBOX_CHARGE_POWER=javascript/0/solarflow/chargePowerSum --> funktioniert nicht 😳

StevieC121176, 2023-10-06:
Ja, Anzeige funktioniert auch :-)

Den Wert der Temperatur aus dem BYD Adapter hole ich mir über den MQTT Client Adapter. 

[image]

ledermann, 2023-10-07:
@q2doc Ich habe das jetzt nochmal getestet, aus meiner Sicht funktioniert das Abgreifen und Weiterleiten der Temperatur korrekt. Vielleicht hilft dir das Folgende zum Debuggen.

Verwendet habe ich:

- SENEC Home V3
- ioBroker 6.10.1
- [ioBroker-SENEC-Adapter](https://github.com/nobl/ioBroker.senec) 1.6.5
- [ioBroker MQTT Broker/Client](https://github.com/ioBroker/ioBroker.mqtt/releases) 4.1.1
- [MQTT-Collector](https://github.com/solectrus/mqtt-collector) 0.1.1
- InfluxDB 2.7.1
- SOLECTRUS 0.13.2

In der `.env` für den MQTT-Collector steht u.a. diese Zeile:

```env
MQTT_TOPIC_CASE_TEMP=senec/0/TEMPMEASURE/CASE_TEMP
```

Der Start des MQTT-Collectors sieht so aus:

```
mqtt-collector-1  | MQTT collector for SOLECTRUS, Version 0.1.1, built at 2023-07-08T12:34:00.903Z
mqtt-collector-1  | https://github.com/solectrus/mqtt-collector
mqtt-collector-1  | Copyright (c) 2023 Georg Ledermann and contributors, released under the MIT License
mqtt-collector-1  |
mqtt-collector-1  | Using Ruby 3.2.2 on platform aarch64-linux-musl
mqtt-collector-1  | Subscribing from MQTT broker at mqtt://datenhalde:1883
mqtt-collector-1  | Pushing to InfluxDB at http://localhost:8086, bucket my-bucket
mqtt-collector-1  |
mqtt-collector-1  | 2023-10-07 09:01:06 +0000 {"bat_fuel_charge"=>63.6}
mqtt-collector-1  | 2023-10-07 09:01:06 +0000 {"bat_power_plus"=>131, "bat_power_minus"=>0}
mqtt-collector-1  | 2023-10-07 09:01:06 +0000 {"grid_power_plus"=>0, "grid_power_minus"=>1630}
mqtt-collector-1  | 2023-10-07 09:01:06 +0000 {"house_power"=>534}
mqtt-collector-1  | 2023-10-07 09:01:06 +0000 {"inverter_power"=>2296}
mqtt-collector-1  | 2023-10-07 09:01:06 +0000 {"current_state"=>"PEAK-SHAVING: WARTEN"}
mqtt-collector-1  | 2023-10-07 09:01:06 +0000 {"mpp1_power"=>1144}
mqtt-collector-1  | 2023-10-07 09:01:06 +0000 {"mpp2_power"=>0}
mqtt-collector-1  | 2023-10-07 09:01:06 +0000 {"mpp3_power"=>1152}
mqtt-collector-1  | 2023-10-07 09:01:06 +0000 {"case_temp"=>34.9}
mqtt-collector-1  | 2023-10-07 09:01:06 +0000 {"wallbox_charge_power"=>0}
```

Wie man sieht, ist eine Zeile mit `case_temp` im Log-Output enthalten, wird also per MQTT empfangen.

Dieser Wert landet dann auch in InfluxDB, wie ich mit dem Admin-Frontend prüfen kann:

[image]

Dort bedient sich dann SOLECTRUS und zeigt die Temperatur an:

[image]

q2doc, 2023-10-07:
Danke hat geholfen - manchmal sieht man den Wald vor Bäumen nicht 🙈 ... der Kaffee geht auf mich

Woran lag es, rückwärts betrachte, keine Daten in InfluxDB, MQTT_Collector fragt aber ab und überträgt. Im ioBroker hatte ich irgendwann mal noch den Client installiert und war im Glauben wenn der Haken zu dem Objekt auf aktiviert gesetzt ist reicht das auch für die mqtt Instanz. Tut es nicht, weil ist ja was anderes 😵.

Oh Wunder wenn ich also nun die States korrekt in der RICHTIGEN Instanz freigebe:

[image]

sehe ich auch die Temperatur:

[image]

blöd von mir aber ich schreib es mal, falls noch jemand darüber stolpert 🫣.

closed 2023-10-07 by q2doc

ledermann, 2023-10-07:
Stimmt, bei den MQTT-Einstellungen kann man auch noch einiges verkehrt machen. Es müssen die richtigen Werte in die Veröffentlichung mit aufgenommen werden.
Freut mich, dass es nun klappt!