solectrus/solectrus #2436: Erweiterung Tooltip Hausverbrauch

StevieC121176, 2023-10-08:
Wäre es möglich den Tooltip für den Hausverbrauch etwas zu erweitern und darzustellen womit der Hausverbrauch gedeckt wird (sh. Screenshot)

[image]

ledermann, 2023-10-17:
Deine Beispielrechnung stimmt meiner Ansicht nach nicht, da es keinen Netzbezug gab. Der Hausverbrauch (473 W) wird gedeckt aus Akku. (ca. 100 W) und Erzeugung (ca. 400 W). Genau das wird aber durch beiden Balken bereits dargestellt.

Die Deckung für einzelne Verbraucher (Haus + Auto) zu berechnen, ist knifflig. Ich versuche das ja beim Kreis "Kosten" und nenne das "abgeschätzte Kosten", um die Kosten den beiden Verursachern Haus und Auto zuzuordnen. Das ist aber nur eine Heuristik.

Allerdings, und hier greife ich deinen Vorschlag auf, könnte man diese Heuristik auch nutzen, um in den Segmenten E-Auto und Hausverbrauch die Deckung abzuschätzen. Also in etwa so:

- E-Auto 57 KW (gedeckt durch 29 KW Netzbezug und 57 - 29 = 28 KW Erzeugung)
- Hausverbrauch 74 KW (gedeckt durch 40 KW vom Akku und 74 - 40 = 34 KW Erzeugung)

Die Heuristik dabei:
- Annahme, dass Strombezug primär durch die Wallbox verursacht wird und erst dann vom Haus
- Annahme, dass Akku-Entladung nur ins Haus fließt, nicht ins E-Auto

Zahlenbeispiel von hier: https://solectrus.ledermann.dev/inverter_power/2023-W41

ledermann, 2024-04-14:
Hiermit habe ich jetzt mal beschäftigt (es gibt ja nicht genug Baustellen in SOLECTRUS 😄).

Mein Beitrag aus dem Oktober 2023 ist überholt. Die Anwendung von Heuristiken halte ich mittlerweile für **nicht** mehr sinnvoll - spätestens, seit hier bei uns mit der Wärmepumpe ein dritter großer Verbraucher eingezogen ist. Für eine sinnvolle Darstellung muss stattdessen **genau** berechnet werden, woher der Strom kommt, den ein Verbraucher benötigt. Das ist allerdings eine komplexe Sache und führt dazu, dass ein weiterer Docker-Container benötigt wird, der kontinuierlich im Hintergrund die Berechnungen durchführt. Dazu ggfs. später mehr.

Hier ein kleiner Screenshot von dem, was ich bezüglich der Darstellung bislang gemacht habe:

[image]
<small>(Noch nirgendwo auszuprobieren, nur lokale Entwicklung)</small>

Im Unterschied zum sehr guten Vorschlag von @StevieC121176 wird die Deckung aus dem Akku nicht berechnet. Ich halte das für vernachlässigbar. Ob der Strom direkt vom Dach in den Verbraucher fließt oder verzögert auf dem Weg über den Akku dorthin gelangt, ist IMHO egal. Wichtig ist nur, ob der Strom selbst erzeugt oder gekauft wurde. Als Ergebnis hat man nun quasi eine "Autarkie je Verbraucher".

Wichtig ist mir, darauf aufbauend die Kosten zu berechnen. Und dabei möchte ich sowohl die direkten Kosten (für den Netzbezug) als auch die kalkulatorischen Kosten (Opportunitätskosten, die durch die entgangene Einspeisevergütung entstehen, siehe #1891) berücksichtigen. Diese Kosten eignen sich dann für einen Vergleich mit einem Alternativ-Verbraucher (z.B. Gas/Öl-Heizung, Verbrenner-Auto).

Die Angaben im Screenshot sind folgendermaßen zu lesen:

Im laufenden Jahr 2024 wurden 46 % des von der Wärmepumpe benötigten Stroms selbst erzeugt, kam also vom Dach. Der Rest wurde aus dem Netz bezogen. Beides zusammen hat insgesamt 334 € gekostet. In die Berechnung der Kosten fließen die Einspeisevergütung sowie die erfassten Strompreise ein (noch ohne Berücksichtigung von Tibber & Co, das wird in #2150 separat behandelt). Die genaue Berechnung lässt sich hier nicht nachvollziehen, das würde die Einfachheit der Darstellung zunichtemachen.

StevieC121176, 2024-04-14:
Ich bin begeistert, gefällt mir sehr gut. Da bin ich echt schon gespannt darauf.

halmich, 2024-06-09:
Oh, ich habe eben erst diese Issue entdeckt.
Das kann dann evtl. mit meinem zusammengelegt werden, da dies im Endeffekt genau mein Vorschlag war :)
https://github.com/solectrus/solectrus/issues/3156

ledermann, 2024-06-09:
Ich verfolge für die Ermittlung der Werte derzeitig einen etwas anderen Ansatz, nämlich diesen:

Zu jedem Zeitpunkt wird in einem nachträglichen Berechnungsschritt ermittelt, wie sich der aus dem Netz bezogene Leistung (also W) auf die momentan aktiven Verbraucher verteilt. Es gibt also keine Rangordnung, sondern eine prozentuale Verteilung.

Beispiel:

- Auto 5 kW
- Wärmepumpe 4 kW
- Haus 1 kW

→ Gesamtverbrauch 10 kW

Angenommen, der Netzbezug ist zu diesem Zeitpunkt 8 kW, dann ergibt sich:
→ Auto aus dem Netz: 5/10 × 8 kW = 4 kW
→ Haus aus dem Netz 1/10 × 8 kW = 0,8 kW
→ Wärmepumpe aus dem Netz: 4/10 × 8 kW = 3,2 kW

So die Grundidee, mehr dazu demnächst.

halmich, 2024-06-09:
Deine Herangehensweise ist ebenfalls richtig und nachvollziehbar.
Ich vergleiche mein Haushalt gerne mit einem Haus, welches keine PV-Anlage, keine Wärmepumpe bzw. kein E-Auto hat.
Das machst du bisher mit Solectrus ja auch (bei den Kosten, bei der Ersparnis,...).

Beispiel:
Ich hatte bisher kein E-Auto. Dadurch hatte ich einen Gesamtstromverbrauch von 7000 kWh, 2000 kamen aus dem Netz, also ca, 28% aus dem Netz. Dieses Verhältnis würde sich bei deiner Betrachtungsweise ändern. Mein Haus würde aufgrund des E-Autos mehr Netzstrom beziehen, als es bisher tat.

Beispiel:
PV Erzeung: 6kW
Haus: 2kW
Wärmepume 4kW
Mein Haus und meine Wärmepumpe laufen "autark" (ich mag den Begriff nicht, da er sugerriert ihr brauche keine Netzversorgung mehr, was ja nicht stimmt...).
Wenn ich nun das E-Auto anstöpsel (4kW), würdest du dem Haus 0,8kW Netz, der Wärmepumpe 1,6kW Netz und dem E-Auto ebenfalls 1,6kW Netz zuordnen. Damit ginge die "Haus-WP-Autarkie" runter. Das Haus steht auf einmal schlechter da. 

Meine Rangfolge wäre deswegen:
Zuerst das Haus (Vergleich mit Haus ohne PV Anlage)
Dann die Wärmepumpe, dann das E-Auto.
Auch deswegen, da ich die Wärmepumpe wenig steuern kann. Die macht warm, wenn es draußen kalt ist. Das Auto kann ich zumindest teilweise bewusst laden oder nicht laden. Lade ich, wenn die Wärmepumpe aus ist, mache ich die Haus-Autarkie schlechter.

Ein Mittelweg wäre, dass man E-Auto und Wärmepumpe als große Verbraucher zusammen betrachtet, also Rangfolge
Haus  - WP/E-Auto.

Dennoch - dein Ansatz ist richtig und gangbar.

ledermann, 2024-06-09:
@halmich Ja, genau diese Überlegungen hatte ich auch angestellt. Eine Rangfolge zu beachten, ist zweifellos eine sinnvolle Alternative. Ich denke, für beide Sichtweisen gibt es stichhaltige Argumente. 

Wenn ich im Winter das E-Auto lade, während (zwangsläufig) gleichzeitig die Wärmepumpe läuft, wird die Wärmepumpe und das Haus dann weniger "autark" (und damit teurer)? Ich meine: ja. Stark hinkender Vergleich: Unerwarteter Gast auf der Geburtstagsparty soll auch ein Stück Kuchen bekommen. Schneidet man den Kuchen jetzt so, dass **alle** Stücke kleiner werden oder gibt es VIP-Gäste, die nichts abgeben müssen?

Denkbar wäre, dass man sich auswählen kann (bei Installation), welche Berechnung man gerne hätte. Die ganze Sache wird sowieso in einem separaten Docker-Container stattfinden, der nonstop Messwerte aus InfluxDB abholt, rechnet, und dann den Netzverbrauch pro Verbraucher als separate Werte nach InfluxDB schreibt.

Es gäbe also zwei verschiedene Berechnungsmethoden:

1) Anteilige Deckung (wie von mir oben beschreiben)
2) Deckung nach Rangfolge Haus-WP-Auto (wie von dir beschrieben)

Es würde mich sowieso interessieren, wie groß die Unterschiede der beiden Methoden tatsächlich sind.

halmich, 2024-06-10:
Wie gesagt, es gibt hier kein richtig oder falsch, sondern einfach verschiedene Herangehensweisen.
Dein Kuchenbeispiel passt auch zu mir. 
Normal hätten der Kuchen ja gut gereicht. Der Gast kommt übrigens nicht ganz unerwartet, den lade ich schon ein ;-)
Wenn nun der extra Gast kommt hilft nichts, dann muss ich los zum Bäcker und noch was extra dazu kaufen. Diese Kosten wurden dann aber durch den Extra-Gast verursacht. :)

Wenn ich das Auto laden, weiß ich in dem Moment, ob ich mit Überschuss lade oder ob ich mit Netzstrom laden werde. Und dann hätte ich gerne am Ende eine Anzeige, wie viele kWh nun aus der PV und wie viele aus dem Netz waren. Die Zappi Wallbox macht das von Haus aus, die Go-E (zumindest bisher noch) nicht.

Zwei Berechnungsansätze wären natürlich super!  :)

ledermann, 2024-07-18:
Hier gibt es nun etwas zu sehen - zumindest in der [Live-Demo](https://demo.solectrus.de/wallbox_power/2023):

Die Darstellung sieht wie folgt aus, die geänderten Bereiche sind hervorgehoben:

[image]

Im Einzelnen:

1) Bei jedem Verbraucher wird nun im MouseOver-Popup angezeigt, wie sich der Stromverbrauch auf Netz und Photovoltaik verteilt. Das sieht beim E-Auto beispielsweise so aus:

    Im Beispiel (vom Jahr 2023) hat das E-Auto also €387 an Stromkosten verursacht, wobei 70 % des Stroms vom Dach kam.

    Die Stromkosten berücksichtigen, wie oben ausführlicher erläutert, auch die kalkulatorischen Kosten: Durch das Beladen des E-Autos wurde weniger eingespeist, dieser entgangene Gewinn wird als Teil der Autokosten angesehen. Dem Ganzen liegt also die Sichtweise zugrunde, dass selbsterzeugter Strom **nicht kostenlos** ist, sondern ca. 8 Cent/kWh kostet.

    Diese Berechnung steht für die Zeiträume Tag/Woche/Monat/Jahr/Gesamt zur Verfügung, und für jeden Verbraucher (also Hausverbrauch, Wärmepumpe und Wallbox).

2) Im Diagramm eines Verbrauchers erfolgt jetzt bei Balkendiagrammen (also Woche/Monat/Jahr/Gesamt) eine gestapelte Darstellung. Jeder Balken setzt sich zusammen aus dem Photovoltaik-Strom (grün) und dem Netzstrom (rot).

3) Der Kreis "Kosten" hat jetzt ein anderes MouseOver-Popup und zeigt außerdem die Gesamtkosten an (also Netzbezugskosten **+ entgangene Einspeisevergütung**). Die bisherige, sehr ungenaue Heuristik wurde also ersetzt durch eine genaue Berechnung. 

Die Berechnung der Zahlen erfolgt durch einen neuen Docker-Container, der zusätzlich zu den bereits vorhandenen permanent läuft und aus dem vorhandenen Datenmaterial den Stromverbrauch der einzelnen Verbraucher aufteilt. Ich nenne das den **Power-Splitter**. Dieser ist noch nicht öffentlich, somit kann das Feature von Euch noch nicht ausprobiert werden. Das kommt aber noch. Nach Fertigstellung soll dies das erste _Sponsor-Only-Feature_ werden ;-)
Das Docker-Image mit dem Tag `pr-3160` steht zwar bereit, bringt aber ohne den Power-Splitter nichts.

Derzeitig erfolgt die Verteilung zu gleichen Anteilen. Der von @halmich vorgeschlagene alternative Berechnungsansatz (nach Rangfolge) ist noch **nicht** implementiert.

Bislang können die dargestellten Anteile und Kosten im UI kaum nachvollzogen werden, man muss also glauben, was da steht. Ich überlege, hier noch etwas nachzulegen und beispielsweise eine zusätzliche Tabellen-Ansicht zu bauen. Möglicherweise mache ich damit aber ein zu großes Fass auf, vielleicht kommt das also auch erst in einem zweiten Schritt.

StevieC121176, 2024-07-18:
Wow, das gefällt mir sehr gut .

ledermann, 2024-07-22:
Zwischenzeitlich habe ich mich mit den möglichen Berechnungsmethoden beschäftigt. Ich sehe mittlerweile drei Möglichkeiten (weitere sind denkbar), den bezogenen Netzstrom auf die Verbraucher zu verteilen:

1) Anteilige, proportionale Verteilung: Das ist die oben beschrieben Methode und war mein bisheriger Favorit: Der bezogene Netzstrom wird an die beteiligten Verbraucher im Verhältnis ihres Stromverbrauchs verteilt. 
2) Rangfolge Auto-WP-Haus: Der Netzstrom wird also zuerst dem Auto zugeordnet, ein etwaiger Rest dann der WP, ein weiterer etwaiger Rest dem Haus. Das ist der Vorschlag von @halmich - hauptsächlich aus dem Grund, dass ein Anstöpseln des Autos die "Autarkie" der anderen Verbraucher nicht beeinflussen sollte.
3) Gemischt: Auto erstrangig, Rest proportional auf WP und Haus. Das halte ich für einen guten Kompromiss, da man die Wärmepumpe ja nicht bewusst ein- oder ausschaltet.

Ich verteile übrigens **Netzstrom**, also nicht den **PV-Strom**. Das ist einfacher, weil ich dann nicht die Batterie berücksichtigen muss. Der PV-Strom für einen Verbraucher ergibt sich dann aus der Differenz `Verbrauch` - `Netzstrom`. Mit "Auto erstrangig" ist also gemeint, dass der teure und böse Netzstrom zuerst dem Auto zugerechnet wird.

Ich habe nun alle drei Methoden implementiert und dann einen Vergleich durchgeführt - anhand meines gesammelten Datenmaterials aus über 3,5 Jahren. Bei 2) und 3) wird das Auto logischerweise teurer als bei 1), weil es **mehr** Netzstrom zugeteilt bekommt. Aber der Unterschied ist (bei meinen Daten) recht gering und beträgt **nur ein paar Prozent**. Daher lege ich mich erst mal auf diese (dritte) Variante fest und plane keine Konfigurierbarkeit. Das kann ich aber je nach User-Feedback nochmal überdenken.

Als Nächstes sehe ich zu, dass ich das Feature zum Betatest bereitstellen kann.

halmich, 2024-07-22:
Finde ich super! Für mich die 3. Möglichkeit völlig ausreichend und ein super Kompromiss.

ledermann, 2024-08-07:
Wer möchte, kann das neue Feature jetzt selbst ausprobieren. Der Beta-Test ist eröffnet :-)

Details hier:
https://github.com/orgs/solectrus/discussions/3330

ledermann, 2024-08-11:
Hier habe ich noch eine wichtige Änderung vorgenommen: Ob bei der Kostenberechnung die Opportunitätskosten (= entgangene Einspeisevergütung) mit einfließen soll oder nicht, kann jetzt per Einstellung festgelegt werden. Der Default ist "NEIN", sodass sich SOLECTRUS wie `v0.15` oder früher verhält.

[image]

gereons, 2024-08-19:
Ich kann die Berechnung des Power-Splitters für heute Nacht (18. auf 19.8.) nicht nachvollziehen.

Gestern (18.8.) abend spät habe ich einen Ladevorhang gestartet, der bis heute (19.8.) morgen um ca 01:50 dauerte. Für den Strombedarf gestern hat der Splitter korrekt "0% aus PV" ermittelt. Für den Anteil der Ladung ab Mitternacht sagt er aber "42% aus PV", ich bin mir aber nahezu 100% sicher dass die Sonne zu der Zeit nicht geschienen hat...

[image]

In InfluxDB sieht der Zeitraum so aus:

[image]

ledermann, 2024-08-19:
@gereons Ja, die Anzeige stimmt nicht. Ich habe eine mögliche Ursache im Verdacht - es hat wieder mit Zeitzonen zu tun. Ein möglicher Fix ist drin. Bitte Docker-Image aktualisieren. Im Log muss dann das hier stehen:

```
Version develop, built at 2024-08-19 15:00:25 +0200
```

Wenn du das heute noch machst, ist nichts weiter zu tun (außer einer Stunde abzuwarten). Da nämlich der Power-Splitter den aktuellen Tag stündlich neu nach InfluxDB schreibt (von 0 Uhr an), wird der Fehler von heute automatisch korrigiert (sofern es das ist, was ich vermute).

Du kannst auch manuell das gesamte Measurement `power_splitter` aus InfluxDB löschen, um eine Neuberechnung zu erzwingen. Das geht so:

```
influx delete --bucket ${INFLUX_BUCKET} --start '1970-01-01T00:00:00Z' --stop $(date -u +"%Y-%m-%dT%H:%M:%SZ") --predicate '_measurement="power_splitter"' --org ${INFLUX_ORG} --token ${INFLUX_TOKEN}
```

Anschließend den Power-Splitter neu starten, der wird dann wieder von Anfang an alles durcharbeiten.

Wenn das alles nicht helfen sollte, liegt der Fehler an etwas ganz anderem. Dann würde es mir bei der Fehlersuche helfen, wenn du mir die InfluxDB von heute als CSV liefern könntest (am besten die beiden Measurements `SENEC` und `power_splitter`).

gereons, 2024-08-19:
Also der einfache Neustart hat leider nichts geändert, ebensowenig das Löschen von `power_splitter` mit anschliessender Neuberechnung.

CSV-Dump findest du unter https://www.dropbox.com/scl/fi/uh6cx85fx1354uhlodbme/solectrus-dump-08-19.csv.zip?rlkey=u6xp6ggi9e1gehpjtevnp6w9g&st=jg6le3kj&dl=0 - ich hoffe das enthält alle benötigten Daten (da ist noch ein kurzer Ladevorgang dazugekommen der auch teilweise PV-Strom benutzt hat, damit ist die Prozentangabe jetzt auf 56 gestiegen).

gereons, 2024-08-20:
Super, danke, das sieht jetzt für den heutigen Tag gut aus (ich hab extra gestern spät noch mal eine Ladung gestartet die über Mitternacht hinweg gedauert hat). 👍 

Ein Detail ist mit bei der Neuberechnung aufgefallen: der Splitter schreibt jetzt laut Log 289 Records pro Tag, aber ein 24-Stunden-Tag hat nur 288 5-Minuten-Intervalle. Off-by-one-Bug irgendwo? 😄

ledermann, 2024-08-20:
Ja, du hast völlig recht. Off-by-one, der Klassiker, ist jetzt ebenfalls korrigiert, jetzt sind es genau 288 Messpunkte pro Tag. Docker-Image ist aktualisiert.

PR #3160 (2024-08-30): Power Splitter
Resolves #2436
closed 2024-08-30 by ledermann