solectrus/solectrus #2497: Vergangene Werte

Deanw1975, 2023-11-03:
Hallo zusammen,

ich musste Solectrus neu aufsetzen.
Hatte in der .env zuerst ein falsches Datum eingetragen :( und dann den csv-importer gestartet.

Nachdem ich meinen Fehler gemerkt hatte, habe ich den Importer neu laufen lassen.
Dennoch werden mir unter den TOP 10 die Vergangenen Werte nicht dargestellt, nur die Werte seit der Installation.

[image]

Wenn ich mir unter "Verlauf" das Jahr ansehe, werden die Werte des Jahres 2023 dargestellt.
[image]

Ich komme aber nicht auf die Werte aus 2022, der Pfeil ist ausgeblendet um weiter zurück zu gehen.
[image]

Die Anlage wurde im April 2022 installiert.

Auch kann ich in der Wochen- oder Monatsansicht kann ich nur bis zur Installation von Solectrus zurück (30.10.2023):

[image]

[image]

[image]

In der letzten Ansicht sieht man, dass die Daten aus dem Oktober da sind.

Was kann ich tun?

VG
Dean

ledermann, 2023-11-03:
Bitte folgende zwei Dingen prüfen:

- Wenn die `.env`-Datei geändert wurde, müssen danach unbedingt die Container neu gestartet werden, um die Änderung zu übernehmen. Also `docker compose restart` ausführen.
- Nach einem Import von Daten muss der Redis-Cache geleert werden: `docker exec -it solectrus-redis-1 redis-cli FLUSHALL`. Näheres dazu hier: https://github.com/solectrus/csv-importer#beware-of-caching-issues

Ein falsches `INSTALLATION_DATE` in der `.env` hat übrigens keine Auswirkungen auf den Importer, sondern nur auf das Dashboard selbst.

Deanw1975, 2023-11-03:
Servus, beides schon gemacht.
Gerade nach deiner Anweisung wiederholt. Kein Erfolg

ledermann, 2023-11-03:
Was genau steht denn in `INSTALLATION_DATE` der `.env`? Da muss das Datum der Installation der PV-Anlage stehen (also in Deinem Fall aus April 2022), also nicht etwas das Datum der Installation von SOLECTRUS.

Wurden die Container wirklich neu gestartet? Es gab schon mal den Fall, dass ein Restart wegen irgendwelcher Fehler fehlgeschlagen ist und somit der alte Container weitergelaufen ist. Finden sich irgendwelche Fehlermeldungen in den Logs?

Falls auch das nicht hilft, poste bitte mal die `.env`, die `docker-compose.yml` und den Output von `docker ps`.

Deanw1975, 2023-11-03:
Nachdem ich deine Anweisung durch bin, hatte ich auch einen Reboot durchgeführt.

Hier die gewünschten Infos

docker ps
CONTAINER ID   IMAGE                                         COMMAND                  CREATED      STATUS                    PORTS                                                                                            NAMES
ee6b0a902555   ghcr.io/solectrus/solectrus:latest            "docker/startup.sh"      3 days ago   Up 11 minutes (healthy)   0.0.0.0:3000->3000/tcp, :::3000->3000/tcp                                                        solectrus-app-1
d2ce3ecd83d1   ghcr.io/solectrus/mqtt-collector:latest       "/bin/sh -c 'bundle …"   3 days ago   Up 11 minutes                                                                                                              solectrus-mqtt-collector-1
6071ea6d45fb   ghcr.io/solectrus/forecast-collector:latest   "/bin/sh -c 'bundle …"   3 days ago   Up 10 minutes                                                                                                              solectrus-forecast-collector-1
b6c0229bb3c9   influxdb:2.7-alpine                           "/entrypoint.sh infl…"   3 days ago   Up 11 minutes (healthy)   8086/tcp                                                                                         solectrus-influxdb-1
dd9f88cc51b6   redis:7-alpine                                "docker-entrypoint.s…"   3 days ago   Up 11 minutes (healthy)   6379/tcp                                                                                         solectrus-redis-1
1c86239974e4   postgres:16-alpine                            "docker-entrypoint.s…"   3 days ago   Up 11 minutes (healthy)   5432/tcp                                                                                         solectrus-db-1
8217712f2de3   containrrr/watchtower                         "/watchtower --scope…"   3 days ago   Up 11 minutes (healthy)   8080/tcp                                                                                         solectrus-watchtower-1
d94625da18b7   portainer/portainer-ce:latest                 "/portainer"             3 days ago   Up 3 days                 0.0.0.0:8000->8000/tcp, :::8000->8000/tcp, 0.0.0.0:9443->9443/tcp, :::9443->9443/tcp, 9000/tcp   portainer

############################################################

.env

#######################################
# Dashboard application (the main part)
#
# Domain name or IP address of your host
APP_HOST=solectrus.local
#
# SSL redirect: Use "true" (which is the default) if you want to auto-redirect to https
FORCE_SSL=false
#
# Secret token to secure cookies, 128 chars long hexadecimal encoded string (don't use this example, make some random changes!)
# Currently there are no cookies in Soletrus, but this may change in the future
xxxxxxxxxxxxxxxxxxxxxxxxx
#
# Date of commissioning of your photovoltaic system
INSTALLATION_DATE=2022-04-12
#
# Price you pay for 1kWH (in EUR)
# Can be changed later in the UI when logged in as administrator
ELECTRICITY_PRICE=0.31
#
# Price you get for 1kWH (in EUR)
# Can be changed later in the UI when logged in as administrator
FEED_IN_TARIFF=0.0777
#
# Password for the PostgreSQL database, used by the app to comunicate internally with the database
POSTGRES_PASSWORD=xxxxxxxxx
#
# Password to login as administrator, required to manage settings like historical prices
ADMIN_PASSWORD=xxxxxxxxx

##################################################################
# Influx database settings
#
# Influx host (to access from SOLECTRUS dashboard and collectors)
INFLUX_HOST=influxdb
INFLUX_SCHEMA=http
INFLUX_PORT=8086
#
# Credentials for the Influx database, don't change after the first run!
INFLUX_ORG=solectrus
INFLUX_USERNAME=xxxxxxxxx
INFLUX_PASSWORD=xxxxxxxxx
INFLUX_ADMIN_TOKEN=xxxxxxxxx
#
# Set these names before first run, they can't be changed later!
INFLUX_BUCKET=my-solectrus-bucket
INFLUX_MEASUREMENT_PV=xxxxxxxxx
INFLUX_MEASUREMENT_FORECAST=Forecast
#
# To keep things simple, we use ONE token (INFLUX_ADMIN_TOKEN) for both writing and reading.
# For better security, you can use two separate tokens, created via the InfluxDB frontend.
[... 125 lines cut ...]
      - FEED_IN_TARIFF
      - ADMIN_PASSWORD
      - DB_HOST=db
      - DB_PASSWORD=${POSTGRES_PASSWORD}
      - DB_USER=postgres
      - REDIS_URL=redis://redis:6379/1
      - INFLUX_HOST
      - INFLUX_TOKEN=${INFLUX_TOKEN_READ}
      - INFLUX_ORG
      - INFLUX_BUCKET
      - INFLUX_MEASUREMENT_PV
      - INFLUX_MEASUREMENT_FORECAST
      - INFLUX_POLL_INTERVAL
    healthcheck:
      test: ['CMD-SHELL', 'nc -z 127.0.0.1 3000 || exit 1']
    restart: always

  influxdb:
    image: influxdb:2.7-alpine
    labels:
      - 'com.centurylinklabs.watchtower.scope=solectrus'
    volumes:
      - ${INFLUX_VOLUME_PATH}:/var/lib/influxdb2
    environment:
      - DOCKER_INFLUXDB_INIT_MODE=setup

ledermann, 2023-11-03:
Hm, das sieht alles korrekt aus. 

Was mir noch einfällt: Check mal die Strompreise und die Einspeisevergütung. Decken beide wirklich den ganzen Zeitraum seit Installation der Anlage ab? Der älteste Preis muss ein Gültigkeitsdatum vom 2022-04-12 oder früher haben.

ledermann, 2023-11-03:
Nochmal zum Redis-Löschen: Hat das wirklich geklappt oder gab es dabei möglicherweise eine Fehlermeldung? Es müsste so aussehen:

```
pi@raspberrypi:~/solectrus $ docker exec -it solectrus-redis-1 redis-cli FLUSHALL
OK
```

Wichtig ist das `OK` als Rückmeldung.

Ansonsten: Ich sehe, dass Du Portainer verwendest. Hast Du da vielleicht einen Stack für SOLECTRUS angelegt, der mit ganz anderen Umgebungsvariablen arbeitet? Um das zu klären: Bitte prüfe, um folgendes Statement das korrekte `INSTALLATION_DATE` ausgibt:

```
docker exec -it solectrus-app-1 env
```

Wenn das alles auch nichts bringt: Stellt sich das Problem immer noch genauso dar wie oben beschrieben, also inklusive des Problems, dass man nicht zurück ins Jahr 2022 navigieren kann? 

Tut mir leid, dass es so mühsam ist. Ich denke, hier müsste langfristig eine Funktion in SOLECTRUS integriert werden, die die Korrektheit der Installation prüft und ggfs. Hinweise auf Korrekturansätze ausgibt.

Deanw1975, 2023-11-03:
Servus, 
mit dem folgenden Befehl wird das falsche Datum ausgegeben:

docker exec -it solectrus-app-1 env

Hier kommt: INSTALLATION_DATE=2023-10-30

Auch der falsche Preis bei 
ELECTRICITY_PRICE=0.31

in der .env steht es aber richtig
Wie behebe ich das nun?

VG
Dean

ledermann, 2023-11-03:
Wir scheinen der Sache auf die Spur zu kommen. Deine Änderungen in der .env werden also gar nicht berücksichtigt. Damit meine ich `INSTALLATION_DATE`, die Variable `ELICTRICITY_PRICE` hingegen wird nur beim ersten Start verwendet und kann dann in SOLECTRUS (bei den Strompreisen) geändert werden.

Bitte stelle sicher, dass sich `.env` und `docker-compose.yml` im gleichen Ordner befinden und du dich auch in diesem Ordner befindest, wenn du `docker-compose up` eingibst. Bitte stelle außerdem sicher, dass die Datei auch wirklich `.env` heißt, also inklusive Punkt.

Solltest du Portainer benutzen, kann die ganze Sache anders aussehen. Dort werden Umgebungsvariablen in der UI verwaltet, d.h. die `.env`-Datei wird gar nicht benutzt.

Deanw1975, 2023-11-03:
Das war es fast;-)

Nicht docker-compose up hat geholfen sondern
docker compose up

DANKE! für die Geduld

ledermann, 2023-11-04:
Ok, je nach Docker-Version heißt es `docker-compose up` oder `docker compose up`. Mal mit, mal ohne Bindestrich. Letzteres ist in neueren Versionen der Fall.

Weiter oben hatte ich aber selbst etwas Falsches erzählt: `docker coompose restart` hilft **nicht**, weil das nur die bestehenden Container neu startet, aber sie nicht neu erstellt. Somit werden auch keine geänderten Umgebungsvariablen übernommen. Es muss also zwingend `docker compose up` sein.

Freut mich, dass das Problem nun behoben ist.

closed 2023-11-04 by ledermann