solectrus/solectrus #2502: Datenbankeinträge korrigieren / Tage löschen

cg089, 2023-11-05:
Hallo zusammen,

Nachdem Solectrus monatelang zuverlässig lief, haben sich im Oktober gleich 3 Datenbankfehler eingeschlichen. Ich schiebe die Werte automatisiert über MQTT rüber, vielleicht gab es da mal einen Schluckauf im Home Assistant.

An 3 Tagen scheint der Gesamtzählerstand (ca. 4200 kWh) des Bezuges als Tagesbezug verwendet worden zu sein. Das verhagelt natürlich die ganze Statistik incl Kosten und Autarkie.

[image]

Nun bin ich kein Datenbankprofi und kann das definitiv nicht reparieren. Hier würde ich mich über eine einfache Möglichkeit freuen, den Tag zumindest herauszulöschen, damit die Statistik wieder grob hinhaut.

ledermann, 2023-11-07:
Die Ursache für die falschen Werte liegt wohl außerhalb von SOLECTRUS, da kann ich somit nichts tun. 

Für das Löschen einzelner Tage aus der InfluxDB könnte folgendes Statement hilfreich sein. Hier ein Beispiel, um den 22.10.2023 (von 0:00 Uhr bis 23:59 Uhr MESZ) zu löschen:

```bash
docker exec <influx-container-name> influx delete \
  --bucket 'your-bucket-name' \
  --org 'your-org-name' \
  --token 'your-token' \
  --start '2023-10-22T00:00:00+02:00' --stop '2023-10-22T23:59:59+02:00'
```

Die einzelnen Parameter (`bucket`, `org` und `token`) müssen natürlich angepasst werden, siehe Angaben in `.env`. Den Namen des Containers lässt sich mit `docker ps` ermitteln.

**Vorsicht, solche Operationen können schnell die ganze Datenbank unbrauchbar machen, wenn man versehentlich die falschen Dinge löscht!**

Nach dem Löschen aus InfluxDB bitte nicht vergessen, den Redis-Cache zu leeren, damit SOLECTRUS auch wirklich die veränderten Tage neu aus der Datenbank abholt. Das geht wie folgt:

```bash
docker exec -it <redis-container-name> redis-cli FLUSHALL
```

Auch hier bitte wieder den richtigen Container-Namen einsetzen.

cg089, 2023-11-13:
Wunderbar, danke für die Anleitung! Das hat geklappt :-)
Jetzt fehlen mir zwar 3 Tage, aber die verfälschen die Statistik nur marginal.
Danke!

[image]

ledermann, 2023-11-14:
Sehr gut! :)

closed 2023-11-14 by ledermann

bgriese0, 2025-03-02:
Ich habe ein ganz ähnliches Problem, heute Nacht um 2:30 hat sich der Inverter aufgehangen und hat immer wieder die gleichen Messwerte zurückgeliefert. Leider schaffe ich es mit diesem Befehl die fehlerhaften Daten nicht zu löschen:

`influx delete --bucket 'solectrus' --org 'solectrus' --token 'secret-admin-token' --start '2025-03-02T00:02:05+02:00' --stop '2025-03-02T00:08:30+02:00' `

Er macht zwar etwas, landet wieder in der Shell aber das war es dann auch. Der Flushall im redis wird mit OK quittiert. Was kann ich alternativ versuchen?

Viele Grüße
Björn

ledermann, 2025-03-02:
In deiner Beschreibung erscheint mir auf den ersten Blick alles korrekt. Wichtig ist nach manuellen Löschoperationen, dass man die Tageszusammenfassungen zurücksetzt. Diese gab es damals in 2023 (beim ursprünglichen Posting hier) noch gar nicht.

Ansonsten: Bitte bei neuen Problemen immer ein **neues** Issue anlegen, und nicht ein altes kommentieren - das führt zu unnötigem Durcheinander.