solectrus/solectrus #2511: Benutzerdefinierbare Kacheldarstellung

ledermann, 2023-11-07:
Fortsetzung von #1986:

Die derzeitige Kacheldarstellung (Menüpunkt: "Einzelwerte") zeigt sechs vordefinierte Werte an. Hier drängt sich ein Customizing auf, das aber noch genauer zu spezifizieren ist.

StevieC121176, 2023-11-07:
Die Kacheldarstellung funktioniert bei mir, gefällt mir auch schon ganz gut. 
Vielleicht könnte man hier einen kleinen Kompromiss finden. Einige Werte wie Ersparnis , Jahres und Gesamterzeugung sind zwar interessant, ändern sich aber nicht so schnell und signifikant das man sie immer im Blick haben muss.
Hausverbrauch, Batterie laden / entladen, SOC, Einspeisung / Netzbezug wären da schon interessanter. Einspeisung/ Netzbezug könnten auf eine Kachel da es ja nur entweder/oder gibt, genau wie Batterie laden/entladen.
Vielleicht erstmal nicht anpassbar, aber andere Werte?

StevieC121176, 2023-11-08:
Ja das ist ein sehr spannendes Thema.  Ich nutze Solectrus unter anderem auf unserem Kühlschrankdisplay und im Wohnzimmer auf dem Tabletwanddisplay. Die neue Kachelansicht ist dafür wie geschaffen, da man sie auch aus einiger Entfernung gut erkennen kann. 
Was sollte ich den sponsern für eine Ansicht (so wie die jetzige) mit den oben beschriebenen Anzeigewerten? ;-)

ledermann, 2023-11-09:
Die Idee finde ich gut. Man könnte zwei "Sets" von Kacheln zur Auswahl anbieten. Einmal die jetzige "Erzeugung/Ersparnis", und eine weitere mit "Haus/Akku/Netz". Das wäre noch kein Customizing, aber ein Kompromiss zur Unterstützung von zwei verschiedenen Anwendungsfällen. Und leicht zu implementieren.

Sponsoring: Entstanden ist die Idee für Features, die ich nicht selber benötige. Wenn dann jemand die Entwicklung nach Aufwand (!) bezahlen möchte, lege ich meinen halben Stundensatz zugrunde, da SOLECTRUS auf jeden Fall Open Source bleiben soll und der Sponsor keine Rechte erwirbt. Also eher teuer, insofern etwas für Unternehmen, weniger für Privatleute. Dafür gibt es dann auch eine Rechnung mit MwSt und Erwähnung im Blog ;-)

Mit der 2-Set-Idee werde ich mich bei Gelegenheit aber beschäftigen, das sieht nach Low-Hanging-Fruits aus.

PS: SOLECTRUS auf dem Kühlschrankdisplay klingt toll 👍

StevieC121176, 2023-11-09:
Das wäre super, vielen Dank.

BtW. Ja so wird das kühlschrankdisplay sinnvoll genutzt, denn digitale Einkaufslisten usw. nutzt man eher nicht :-)  Aber dank Chrome Browser läuft Solectrus und evcc darauf hervorragend.

Zeppelin-PV, 2023-12-05:
Könnte man in der Kacheldarstellung Uhrzeit und Status oben mit einblenden ?
vielleicht anstatt des Namens und dem Ort
[image]
Bei NPU Fehler oder Batteriediagnose ist man bei den Kacheln total ohne Info z.B. wenn diese Anzeige in der fully Kiosk App läuft und nicht so ohne Weiteres umgeschaltet werden kann

Zeppelin-PV, 2023-12-05:
... sorry hab das Menü oben rechts übersehen, dennoch eine Überlegung ?

ledermann, 2023-12-05:
@Zeppelin-PV Ich stimme dir zu, dass die Darstellung des Status auch auf der Seite "Einzelwerte" sinnvoll wäre. 

Da die Kacheldarstellung das Ziel verfolgt, auch aus der Ferne abgelesen zu werden, würde ich hier aber eher den Status im Großformat als einzelne Kachel anbieten wollen, also nicht irgendwo in kleiner Schrift in der Kopfzeile. Vielleicht ungefähr so:

[image]

Zeppelin-PV, 2023-12-08:
könnte man, rein theoretisch, den Kacheln einen Drop-/Pull down verpassen und dann einblenden was man gerne auf dieser Kachel sehen möchte
praktisch so wie im Grundbild beim Diagramm
[image]

ledermann, 2023-12-08:
Ja, @Zeppelin-PV, so in etwa denke ich mir das. Meine Beschreibung ganz oben ist noch sehr rudimentär. Was ich mir vorstelle, ist Folgendes:

- Die Darstellung beginnt mit einer Standardansicht, die der jetzigen entspricht und weiterhin genau sechs Kacheln umfasst.
- Jede Kachel erhält einen Edit-Button, über den man ein paar Dinge einstellen kann: zumindest Messwert und Zeitraum, möglicherweise noch mehr. 
- Die Reihenfolge der Kacheln ist anpassbar (per Drag & Drop)
- Konfiguration nur vom Admin (also nach Login)

Die Implementierung ist aber aufwendig und ich selbst benötige das nicht wirklich. Interessant ist es für Leute, die SOLECTRUS im Wohnzimmer/Küche/Geschäft/Büro auf dem Wand-Tablet/Kühlschrank/Fernseher/DigitalSignage installiert haben und aus der Ferne ohne Herumklicken die wichtigsten Werte präsentieren wollen.

Die derzeitige Kachel-Darstellung ist zustande gekommen, weil ein Sponsor mich explizit mit der Implementierung beauftragt und bezahlt hat.

Zeppelin-PV, 2023-12-11:
... es sind immer die Leute die am meisten wollen, aber nichts können  ;-)

wenn die benutzerdefinierbaren Kacheln kommen, wäre auch folgendes vorstellbar ?
Während des Tages ist alles auf Ertrag getrimmt, so wie die momentane Anzeige.
Nach Sonnenuntergang wäre es doch cool, wenn z.B. die oberen drei Kacheln umschalten auf
Speicherinhalt, Hausverbrauch und eine Kachel die anzeigt wie weit der Speicherinhalt noch reicht.
wenn leer dann Anzeige Netzbezug.
Bei Sonnenaufgang wieder die Ertragskacheln.

ledermann, 2023-12-11:
Ein Wechsel zwischen Tages- und Nachtansicht wäre denkbar und sinnvoll.

Auch die Anzeige, wie lange der Speicherinhalt noch reicht, ist eine schöne Idee. Einfach könnte man das so rechnen: `Aktueller Speicherinhalt in kWh / Verbrauch in den letzten 60min`. Dazu wäre aber auch die Gesamtkapazität erforderlich, siehe #2608

An Ideen mangelt es nicht :-)

StevieC121176, 2024-04-12:
Die Kachelansicht von SOLECTRUS hat es nun (nachdem unsere neue Küche nun fertig eingebaut ist :-) auf die Startseite unseres Kühlschranks geschafft. Einige kleine Sachen fallen dabei auf:
- die normale Kachelansicht wird korrekt dargestellt sh. Foto. Ich kann auch den vollbildmodus in SOLECTRUS aktivieren, nur da wird leider in das Querformat gewechselt und somit bleibt die untere Bildschirmhälfte leer. Das Display im Kühlschrank ist senkrecht . Kann man da Abhilfe schaffen?
-  Die anderen Punkte wurden hier schon genannt, wichtig für Privatanlagen wäre jegliche Art des Verbrauchs und das E-Auto
-  Farblich würde ich mir noch einen Darkmode wünschen 

[image]

ledermann, 2024-04-13:
@StevieC121176 Wow, das sieht wirklich toll aus, danke für das Foto!

Warum der Vollbildmodus ins Querformat wechselt, kann ich leider nicht erklären, das ist mir neu. Es wird einfach nur [requestFullscreen()](https://developer.mozilla.org/en-US/docs/Web/API/Element/requestFullscreen) aufgerufen. Was daraus wird, entscheidet der Browser. Grundsätzlich ist der FullScreen-Mode oft ein Problem. Das hängt total vom Browser ab. 

Ein Issue für den Darkmode ist in #1859 angelegt.

StevieC121176, 2024-04-14:
evtl. liegt es an der Auflösung. normal im Hochformat werden 982x1514 Pixel genutzt, beim Fullscreen dann 1080x1920px