solectrus/solectrus #2588: Numerical argument is out of domain

ledermann, 2023-12-01:
Gemeldet von @SGURLY in #2587:

```
[192.168.178.55] ActionView::Template::Error (Numerical argument is out of domain - log):
[192.168.178.55]      5: turbo-stream action="morph" target="stats" children-only= true
[192.168.178.55]      6:   template
[192.168.178.55]      7:     - if timeframe.now?
[192.168.178.55]      8:       = render StatsNow::Component.new calculator:, field:
[192.168.178.55]      9:     - else
[192.168.178.55]     10:       = render StatsRange::Component.new calculator:, timeframe:, field:
[192.168.178.55]  
```

> SOLECTRUS zeigt seitdem nur noch: `500 Internal Server Error`
>
> Es ist dabei egal, auf welche stats ich via URL zugreifen möchte, der Fehler kommt.

ledermann, 2023-12-01:
@SGURLY: Ursache scheint hier zu sein, dass das Attribut `inverter_power` negativ ist. Das sorgt dann bei der Berechnung des Logarithmus (benötigt für die passende Höhe der Flussanimation) für die Exception `Numerical argument is out of domain - log`. 

Warum der Wert bei dir negativ ist, müsstest du mal untersuchen. Was ich aber tun kann, ist die Verhinderung eines Fehler 500 in SOLECTRUS. Das habe ich gerade erledigt, die Korrektur wird ins nächste Update (in Kürze) mit einfließen. 

Wenn du möchtest, kannst du den Fix bei dir sofort ausprobieren, indem du das Docker-Image von `latest` auf `develop` wechselst.

SGURLY, 2023-12-01:
@ledermann 

Ich habe deine Änderungen schonmal lokal übernommen, das hat leider nicht geklappt - der Fehler war weiterhin der selbe, auch nachdem ich den container (solectrus-app) neu gebaut habe. 

Der Fehler tritt bei Sungrow leider auf, da dort die inverter power berechnet werden muss. Weiterhin runden sie leider sehr wage (Ergebnisse lediglich in kw verfügbar), wodurch ein paar watt unterschied entstehen können. 
Ich habe meinen Datenupload und die historischen werte in der Influx gefixed - jetzt funktioniert das Dashboard wieder.

Vielen dank für die sehr schnelle Hilfe!

SGURLY, 2023-12-01:
@ledermann
Vielen dank für die (sehr) schnelle Hilfe!

Zusätzlich habe ich den Fehler nochmal bei mir gefixed und setze negative Werte jetzt hart auf 0.
Sungrow scheint die Leistung welche der Wechselrichter selbst braucht mit einzubeziehen und die API gibt hier negative werte aus. Warum es ein paar Tage lief bleibt mir jedoch ein Rätsel, mit die ersten Werte in der Influx waren schon negativ.

[image]

ledermann, 2023-12-02:
Dass der Fehler anfangs nicht aufgetreten ist, könnte daran liegen, dass Du vielleicht nicht permanent den Tab "JETZT" offen hattest. Die Werte schwanken ja ständig und der Fehler tritt nur bei negativen Zahlen auf.

Wie ich zwischenzeitlich noch gesehen habe, könnte das Problem auch noch bei anderen Feldern auftreten. Die Größe der Icons in den Strombilanz-Segmenten wird nämlich auch logarithmisch angepasst. 

Aber wie dem auch sei, jetzt ist das Problem behoben und SOLECTRUS ist robuster gegenüber negativen Messwerten.