solectrus/solectrus #2595: Flussanimation optional wiederherstellen

ledermann, 2023-12-04:
Die Flussanimation ist aus Energiespargründen (siehe #2423) in v0.14 entfernt worden - was allerdings [bedauert](https://www.facebook.com/solectrus/posts/pfbid0NsJcW4Bz5j15jVJJJ43BaiSEm3hAjUXE9pPXHHwwNoJch2uXrRXqyqWGm9v37rcwl?comment_id=874432314076735) wurde.

Abhilfe:

- Wieder einführen, aber sparsamer gestalten
- Animation ausschalten, wenn der Browser dies explizit anfordert (mittels [prefers-reduced-motion](https://developer.mozilla.org/en-US/docs/Web/CSS/@media/prefers-reduced-motion))

StevieC121176, 2023-12-04:
Gefällt mir sehr gut die neue Animation der Flussrichtung.

ledermann, 2023-12-05:
@Zeppelin-PV Schöne Idee, das mit der Ferraris-Scheibe :-) 

Da ich aber kein CSS-Animations-Experte bin, möchte ich da keine weitere Zeit drin versenken. Ich lasse die Animation jetzt erst mal mit den Pfeilen, wie sie ist.

commit 49045e31 (2024-01-07): Re-add flow animation (unless user prefers reduced motion)

Resolves #2595