solectrus/solectrus #2625: nach update langsame Zugriffe

Deanw1975, 2023-12-11:
Hallo zusammen,

ich habe Solctrus auf einen RPI 3 am laufen. 

Bis zum Update auf 14.1 waren die Abrufe der einzelnen sichten recht zügig.
Seit dem Update verzögern sich die Aufrufe bis hin das ein Teil der Seiten nicht dargestellt werden
Beispiel: In der Tagesansicht wird die rechte Seite gar nicht mehr dargestellt.
Der Aufruf dauert teilweise bis zu einer halben Minute oder es wird abgebrochen.
Die drei kreise bleiben einfach stehen.

Was soll ich am sinnvollsten liefern?

Danke für die Mühe

LG
Dean

ledermann, 2023-12-11:
Das ist seltsam - von Problemen durch das Update auf v0.14.1 habe ich bislang nichts gehört. 

Ich würde eher vermuten, dass der Zugriff auf InfluxDB performance-kritisch ist. Je größer die Datenbank wird, desto langsamer wird der Zugriff. Mit den gesammelten Daten aus drei Jahren (im 5-Sekunden-Takt) ist bei mir aber noch alles im grünen Bereich - ich nutze allerdings keinen Raspi, sondern einen Cloud-Server von Hetzner.

Hast Du vielleicht sonst irgendwas verändert? Beispielsweise andere Dinge auf dem Raspi installiert, die Speicherplatz oder CPU benötigen? Oder große Datenmengen nach InfluxDB importiert? Besteht das Problem auch nach einem Neustart des Raspi weiter? Ist InfluxDB aktuell? Es sollte `v2.7.4` sein.

Ich schlage daher vor, zunächst sicherzustellen, ob es wirklich mit dem Update zusammenhängt.  Mach dazu bitte mal ein Rollback auf die Version 0.13.2. Das geht einfach wie folgt:

In der docker-compose.yml steht:

```yml
app:
  image: ghcr.io/solectrus/solectrus:latest
```

Dies bitte probeweise durch das hier ersetzen:

```yml
app:
  image: ghcr.io/solectrus/solectrus:0.13.2
```

Dann die Container neu starten und prüfen, ob wirklich die v0.13.2 läuft (unten im Footer). Ist dann wieder alles schnell? In jedem Fall anschließend bitte wieder auf `latest` wechseln, also auf Version `v0.14.2`.

Wenn es wirklich so ist, dass mit `v0.13.2` alles flott läuft, mit `v0.14.2` aber langsam - was ich bezweifle - dann müssen wir weiter forschen. 

Bitte dann prüfen, welche Ansichten genau betroffen sind. Grundsätzlich ist es so, dass je größer ein Zeitraum ist, desto länger dauert die Ermittlung der Daten. Am langsamsten ist somit der Zeitraum "Gesamt". Der Zeitraum "Tag" ist am schnellsten. Das ist normal, aber es sollte in allen Fällen erträglich sein. Übrigens ist es so, dass jede Abfrage in einem Cache auf Redis landet. Hat man also einmal das Jahr 2022 abgerufen, sind zukünftige Abfrage für dieses Jahr extrem schnell, weil gar nicht InfluxDB abgefragt wird, sondern das Resultat aus dem Cache genommen wird.

Noch ein Tipp zur Untersuchung: Starte eine Abfrage, die sehr langsam ist und führe genau währenddessen auf der Konsole folgendes Kommando aus:

```
docker stats --no-stream
```

Welcher Container verursacht die größte CPU-Last? Was ist mit dem Speicherverbrauch?

Ist überhaupt genug RAM vorhanden?

```
free -h
```

Es gibt vieles zu prüfen. Ich bin gerne bereit, Dich bei der Ursachenforschung zu unterstützen.

Deanw1975, 2023-12-12:
Heute morgen musste ich den RPI sogar neu starten. Da Telenet fast nicht mehr reagierte.
Auf dem PI läuft Solcetrus exclusiv.

> Ich würde eher vermuten, dass der Zugriff auf InfluxDB performance-kritisch ist. Je größer die Datenbank wird, desto langsamer wird der Zugriff. Mit den gesammelten Daten aus drei Jahren (im [...]

--------------------------------------------------------------------------------------------------------
Solecturs 0.14.1:

pi@solectrus:~ $ free -h
              gesamt       benutzt     frei      gemns.  Puffer/Cache verfügbar
Speicher:      909Mi       704Mi        37Mi       2,0Mi       167Mi       138Mi
Swap:           99Mi        99Mi       0,0Ki

pi@solectrus:~ $ docker stats --no-stream
CONTAINER ID   NAME                             CPU %     MEM USAGE / LIMIT   MEM %     NET I/O           BLOCK I/O        PIDS
f033623a8254   solectrus-app-1                  3.97%     0B / 0B             0.00%     193kB / 170kB     177MB / 0B       59
ecfaa87667a3   solectrus-forecast-collector-1   0.00%     0B / 0B             0.00%     14.3kB / 4.04kB   0B / 0B          2
7854e25e9388   solectrus-mqtt-collector-1       0.38%     0B / 0B             0.00%     114kB / 132kB     68MB / 0B        3
60bccf53ae49   solectrus-influxdb-1             3.10%     0B / 0B             0.00%     196kB / 163kB     577MB / 410kB    21
810dfbbeba6d   solectrus-db-1                   1.21%     0B / 0B             0.00%     54.8kB / 75.5kB   207MB / 57.3kB   12
99495730dc91   solectrus-redis-1                1.00%     0B / 0B             0.00%     56.9kB / 22.7kB   51.9MB / 0B      6
d94625da18b7   portainer                        0.32%     0B / 0B             0.00%     13.2kB / 1.6kB    123MB / 94.2kB   6
--------------------------------------------------------------------------------------
Solecturs 0.13.2:

pi@solectrus:~/solectrus $ free -h
              gesamt       benutzt     frei      gemns.  Puffer/Cache verfügbar
Speicher:      909Mi       448Mi        86Mi       5,0Mi       375Mi       392Mi
Swap:           99Mi        78Mi        21Mi

pi@solectrus:~/solectrus $ docker stats --no-stream
CONTAINER ID   NAME                             CPU %     MEM USAGE / LIMIT   MEM %     NET I/O           BLOCK I/O        PIDS
2c852464d46b   solectrus-app-1                  0.78%     0B / 0B             0.00%     317kB / 186kB     50.9MB / 0B      14
ecfaa87667a3   solectrus-forecast-collector-1   0.00%     0B / 0B             0.00%     55kB / 14.2kB     8.7MB / 0B       2
7854e25e9388   solectrus-mqtt-collector-1       0.04%     0B / 0B             0.00%     1.11MB / 1.49MB   97.8MB / 0B      3
60bccf53ae49   solectrus-influxdb-1             0.73%     0B / 0B             0.00%     1.55MB / 1.08MB   903MB / 631kB    21
810dfbbeba6d   solectrus-db-1                   10.90%    0B / 0B             0.00%     97.7kB / 125kB    390MB / 77.8kB   8
99495730dc91   solectrus-redis-1                8.40%     0B / 0B             0.00%     90kB / 34.6kB     96.1MB / 0B      6
9f712a731da2   solectrus-watchtower-1           0.00%     0B / 0B             0.00%     22.1kB / 1.35kB   12.5MB / 0B      9
d94625da18b7   portainer                        0.00%     0B / 0B             0.00%     27.6kB / 1.6kB    216MB / 176kB    6
pi@solectrus:~/solectrus $Ausser ein Update habe ich nichts geändert.

> 
> Ich schlage daher vor, zunächst sicherzustellen, ob es wirklich mit dem Update zusammenhängt. Mach dazu bitte mal ein Rollback auf die Version 0.13.2. Das geht einfach wie folgt:
> 
> In der docker-compose.yml steht:
> 
> ```yaml
> app:
>   image: ghcr.io/solectrus/solectrus:latest
> ```
> 
> Dies bitte probeweise durch das hier ersetzen:
> 
> ```yaml
> app:
>   image: ghcr.io/solectrus/solectrus:0.13.2
> ```
> 
> Dann die Container neu starten und prüfen, ob wirklich die v0.13.2 läuft (unten im Footer). Ist dann wieder alles schnell? In jedem Fall anschließend bitte wieder auf `latest` wechseln, also auf Version `v0.14.2`.
> 
> Wenn es wirklich so ist, dass mit `v0.13.2` alles flott läuft, mit `v0.14.2` aber langsam - was ich bezweifle - dann müssen wir weiter forschen.
------------------------------------------------------------------------
13.2 macht keine Probleme
 ------------------------------------------------------------------------
> Bitte dann prüfen, welche Ansichten genau betroffen sind. Grundsätzlich ist es so, dass je größer ein Zeitraum ist, desto länger dauert die Ermittlung der Daten. Am langsamsten ist somit der [...]
---------------------------------------------
Zurück auf Solectrus 0.14.1
Konnte ich zusehen wie der Cache runter ist, bis die Seite unter "Aktuell" abgeschemiert ist.
Seite neu aufgrufen
[... 87 lines cut ...]
pi@solectrus:~/solectrus $ free -h
              gesamt       benutzt     frei      gemns.  Puffer/Cache verfügbar
Speicher:      909Mi       760Mi        40Mi       4,0Mi       109Mi        83Mi
Swap:           99Mi        99Mi       0,0Ki
pi@solectrus:~/solectrus $ free -h
              gesamt       benutzt     frei      gemns.  Puffer/Cache verfügbar
Speicher:      909Mi       762Mi        38Mi       4,0Mi       108Mi        80Mi
Swap:           99Mi        99Mi       0,0Ki
pi@solectrus:~/solectrus $ free -h
              gesamt       benutzt     frei      gemns.  Puffer/Cache verfügbar
Speicher:      909Mi       768Mi        40Mi       4,0Mi       100Mi        74Mi
Swap:           99Mi        99Mi       0,0Ki
pi@solectrus:~/solectrus $ free -h
[... cut ...]
pi@solectrus:~/solectrus $ free -h
              gesamt       benutzt     frei      gemns.  Puffer/Cache verfügbar
Speicher:      909Mi       785Mi        35Mi       4,0Mi        87Mi        57Mi
Swap:           99Mi        99Mi       0,0Ki
pi@solectrus:~/solectrus $ free -h
              gesamt       benutzt     frei      gemns.  Puffer/Cache verfügbar
Speicher:      909Mi       786Mi        34Mi       4,0Mi        88Mi        56Mi
Swap:           99Mi        99Mi       0,0Ki
pi@solectrus:~/solectrus $

Deanw1975, 2023-12-12:
Heute morgen musste ich den RPI sogar neu starten. Da Telenet fast nicht mehr reagierte.
Auf dem PI läuft Solcetrus exclusiv.

Ich würde eher vermuten, dass der Zugriff auf InfluxDB performance-kritisch ist. Je größer die Datenbank wird, desto langsamer wird der Zugriff. Mit den gesammelten Daten aus drei Jahren (im 5-Sekunden-Takt) ist bei mir aber noch alles im grünen Bereich - ich nutze allerdings keinen Raspi, sondern einen Cloud-Server von Hetzner.

Hast Du vielleicht sonst irgendwas verändert? Beispielsweise andere Dinge auf dem Raspi installiert, die Speicherplatz oder CPU benötigen? Oder große Datenmengen nach InfluxDB importiert? Besteht das Problem auch nach einem Neustart des Raspi weiter? Ist InfluxDB aktuell? Es sollte `v2.7.4` sein.

--------------------------------------------------------------------------------------------------------
Solecturs 0.14.1:

pi@solectrus:~ $ free -h
              gesamt       benutzt     frei      gemns.  Puffer/Cache verfügbar
Speicher:      909Mi       704Mi        37Mi       2,0Mi       167Mi       138Mi
Swap:           99Mi        99Mi       0,0Ki

pi@solectrus:~ $ docker stats --no-stream
CONTAINER ID   NAME                             CPU %     MEM USAGE / LIMIT   MEM %     NET I/O           BLOCK I/O        PIDS
f033623a8254   solectrus-app-1                  3.97%     0B / 0B             0.00%     193kB / 170kB     177MB / 0B       59
ecfaa87667a3   solectrus-forecast-collector-1   0.00%     0B / 0B             0.00%     14.3kB / 4.04kB   0B / 0B          2
7854e25e9388   solectrus-mqtt-collector-1       0.38%     0B / 0B             0.00%     114kB / 132kB     68MB / 0B        3
60bccf53ae49   solectrus-influxdb-1             3.10%     0B / 0B             0.00%     196kB / 163kB     577MB / 410kB    21
810dfbbeba6d   solectrus-db-1                   1.21%     0B / 0B             0.00%     54.8kB / 75.5kB   207MB / 57.3kB   12
99495730dc91   solectrus-redis-1                1.00%     0B / 0B             0.00%     56.9kB / 22.7kB   51.9MB / 0B      6
d94625da18b7   portainer                        0.32%     0B / 0B             0.00%     13.2kB / 1.6kB    123MB / 94.2kB   6
--------------------------------------------------------------------------------------
Solecturs 0.13.2:

pi@solectrus:~/solectrus $ free -h
              gesamt       benutzt     frei      gemns.  Puffer/Cache verfügbar
Speicher:      909Mi       448Mi        86Mi       5,0Mi       375Mi       392Mi
Swap:           99Mi        78Mi        21Mi

pi@solectrus:~/solectrus $ docker stats --no-stream
CONTAINER ID   NAME                             CPU %     MEM USAGE / LIMIT   MEM %     NET I/O           BLOCK I/O        PIDS
2c852464d46b   solectrus-app-1                  0.78%     0B / 0B             0.00%     317kB / 186kB     50.9MB / 0B      14
ecfaa87667a3   solectrus-forecast-collector-1   0.00%     0B / 0B             0.00%     55kB / 14.2kB     8.7MB / 0B       2
7854e25e9388   solectrus-mqtt-collector-1       0.04%     0B / 0B             0.00%     1.11MB / 1.49MB   97.8MB / 0B      3
60bccf53ae49   solectrus-influxdb-1             0.73%     0B / 0B             0.00%     1.55MB / 1.08MB   903MB / 631kB    21
810dfbbeba6d   solectrus-db-1                   10.90%    0B / 0B             0.00%     97.7kB / 125kB    390MB / 77.8kB   8
99495730dc91   solectrus-redis-1                8.40%     0B / 0B             0.00%     90kB / 34.6kB     96.1MB / 0B      6
9f712a731da2   solectrus-watchtower-1           0.00%     0B / 0B             0.00%     22.1kB / 1.35kB   12.5MB / 0B      9
d94625da18b7   portainer                        0.00%     0B / 0B             0.00%     27.6kB / 1.6kB    216MB / 176kB    6
pi@solectrus:~/solectrus $Ausser ein Update habe ich nichts geändert.

Ich schlage daher vor, zunächst sicherzustellen, ob es wirklich mit dem Update zusammenhängt. Mach dazu bitte mal ein Rollback auf die Version 0.13.2. Das geht einfach wie folgt:

In der docker-compose.yml steht:

```yaml
app:
  image: ghcr.io/solectrus/solectrus:latest
```

Dies bitte probeweise durch das hier ersetzen:

```yaml
app:
  image: ghcr.io/solectrus/solectrus:0.13.2
```

Dann die Container neu starten und prüfen, ob wirklich die v0.13.2 läuft (unten im Footer). Ist dann wieder alles schnell? In jedem Fall anschließend bitte wieder auf `latest` wechseln, also auf Version `v0.14.2`.

Wenn es wirklich so ist, dass mit `v0.13.2` alles flott läuft, mit `v0.14.2` aber langsam - was ich bezweifle - dann müssen wir weiter forschen.
------------------------------------------------------------------------
13.2 macht keine Probleme
 ------------------------------------------------------------------------
Bitte dann prüfen, welche Ansichten genau betroffen sind. Grundsätzlich ist es so, dass je größer ein Zeitraum ist, desto länger dauert die Ermittlung der Daten. Am langsamsten ist somit der Zeitraum "Gesamt". Der Zeitraum "Tag" ist am schnellsten. Das ist normal, aber es sollte in allen Fällen erträglich sein. Übrigens ist es so, dass jede Abfrage in einem Cache auf Redis landet. Hat man also einmal das Jahr 2022 abgerufen, sind zukünftige Abfrage für dieses Jahr extrem schnell, weil gar nicht InfluxDB abgefragt wird, sondern das Resultat aus dem Cache genommen wird.

Noch ein Tipp zur Untersuchung: Starte eine Abfrage, die sehr langsam ist und führe genau währenddessen auf der Konsole folgendes Kommando aus:

[... 102 lines cut ...]
pi@solectrus:~/solectrus $ free -h
[... cut ...]
pi@solectrus:~/solectrus $ free -h
              gesamt       benutzt     frei      gemns.  Puffer/Cache verfügbar
Speicher:      909Mi       785Mi        35Mi       4,0Mi        87Mi        57Mi
Swap:           99Mi        99Mi       0,0Ki
pi@solectrus:~/solectrus $ free -h
              gesamt       benutzt     frei      gemns.  Puffer/Cache verfügbar
Speicher:      909Mi       786Mi        34Mi       4,0Mi        88Mi        56Mi
Swap:           99Mi        99Mi       0,0Ki
pi@solectrus:~/solectrus $

Deanw1975, 2023-12-12:
Danke für deine Geduld.
Irgendwie mag mich Solectrus nicht ;-)
Immer wieder mal komische Herausforderungen.

Heute morgen musste ich den RPI sogar neu starten. Da Telnet fast nicht mehr reagierte.
Auf dem PI läuft Solcetrus exclusiv.

Da meine ganzen Notizen leider vom Github Forum komisch fomatiert werden, füge ich diese als TXT an.

[pi.txt](https://github.com/solectrus/solectrus/files/13644894/pi.txt)

ledermann, 2023-12-12:
Ok, meine Vermutung, dass es an InfluxDB liegt, war offenbar **falsch**. Ursache ist vielmehr, dass SOLECTRUS selbst laut deinen Untersuchungen mehr RAM benötigt, als auf deinem 1GB-Raspi zur Verfügung steht.

Es ist tatsächlich so, dass `v0.14.1` standardmäßig mehr RAM benötigt als `v0.13.2`. Hintergrund ist der integrierte Web-Server, der seit v0.14.1 im Cluster-Mode startet und dabei so vielen Prozesse startet, wie Prozessorkerne vorhanden sind. Damit lassen sich dann zwar mehr Requests parallel verarbeiten, jeder Prozess benötigt aber etwa 100 MB RAM. Auf einem Raspi 3 mit 4 Kernen sind somit 4 Prozesse am Werk. Das sind über 400 MB RAM-Verbrauch, zu viel für den kleinen Raspi mit nur 1 GB RAM.

Das Problem kann aber leicht korrigiert werden. Die Anzahl der zu startenden Prozesse wird über die Umgebungsvariable `WEB_CONCURRENCY` festgelegt. Diese solltest du manuell auf `0` setzen, um den Single-Mode zu aktivieren. Das geht wie folgt:

Bearbeite die Datei `docker-compose.yml` und füge im Abschnitt `services.app.environment` folgende Zeile hinzu:

```
- WEB_CONCURRENCY=0
```

Anschließend wieder alles neu starten. Im Log der App sollte nun folgende Zeile stehen:

```
Puma starting in single mode...
```

Deanw1975, 2023-12-12:
Servus Georg,

sorry wenn ich es "nur" auf einem ollen RPI3 laufen lasse möchte:/

Jetzt "rennt" Solectrus wie noch nie zuvor!

Hier der LOG:
pi@solectrus:~/solectrus $ docker logs solectrus-app-1
Starting SOLECTRUS...
Version: v0.14.1 - 2023-12-03T14:36:01+01:00 - v0.14.1
----------------
influxdb (172.18.0.7:8086) open
InfluxDB is up and running!
db (172.18.0.3:5432) open
PostgreSQL is up and running!
Preparing database...
Database is ready!
Puma starting in single mode...
* Puma version: 6.4.0 (ruby 3.2.2-p53) ("The Eagle of Durango")
*  Min threads: 5
*  Max threads: 5
*  Environment: production
*          PID: 11
* Listening on http://0.0.0.0:3000

--------------------------------
Und hier free -h

free -h

gesamt       benutzt     frei      gemns.  Puffer/Cache verfügbar
Speicher:      909Mi       489Mi        91Mi       9,0Mi       328Mi       346Mi
Swap:           99Mi        48Mi        51Mi

DANKE dir
VG
Dean

ledermann, 2023-12-12:
Prima, freut mich sehr! Und ja, auch die kleinsten Raspis möchte ich gerne weiterhin unterstützen :-)

P.S.: Hier auf GitHub wird ein Beitrag per Markdown formatiert. Codeschnippsel z.B. mit Backticks. Das ist etwas gewöhnungsbedürftig, aber bietet viele [Möglichkeiten](https://docs.github.com/de/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax).

closed 2023-12-12 by ledermann