solectrus/solectrus #2647: House consumption is not displayed after Update v.0.14.1

holiday-sunrise, 2023-12-17:
After I updated to 0.14.1, the house consumption is only displayed briefly (a few minutes). In the daily overview and in the LOG you can see that the values are saved via MQTT.

Here below is a screenshot. 

[
[image]
](url)

ledermann, 2023-12-17:
Ich antworte mal auf Deutsch - meine ursprüngliche Idee, alles auf Englisch abzuwickeln und somit auch das internationale Publikum anzusprechen, hat zu einigen Irritationen hierzulande geführt ;-)

Welche Änderung in der Version `v0.14.1` zu diesem Phänomen geführt haben könnte, kann ich noch nicht sehen. Ein paar Ideen dazu:

- Bist du denn sicher, dass hinsichtlich MQTT alles richtig läuft, kommen die Werte für den Hausverbrauch zeitnah an? 
- Wie sieht das Diagramm für Hausverbrauch im Tab "Aktuell" aus? Ist das alles komplett?
- Verstehe ich dich richtig, dass auf den anderen Tabs der Hausverbrauch richtig dargestellt wird?
- Wird im Log des MQTT-Containers regelmäßig auch der Hausverbrauch angezeigt?
- Gibt es sonstige Auffälligkeiten mit Version `v0.14.1` oder läuft sonst alles?
- Verstehe ich dich richtig, dass mit der Version `v0.13.2` der Fehler nicht auftrat? Lässt sich das bestätigen, wenn du ein Rollback auf `v0.13.2` machst (in docker-compose.yml statt `latest` das Tag `0.13.2` eintragen)? Liegt es also wirklich an Version `v0.14.1` vs. `v0.13.2` ?

holiday-sunrise, 2023-12-17:
[image]

holiday-sunrise, 2023-12-17:
Hi,
ja die Daten werden auf allen anderen Tabs angezeigt. ImLog des MQTT stehen auch Werte für den Hausverbrauch. Ich mache mal einen Rollback auf 0.13.2 mal sehen was passiert.

Beste Grüße

Rainer

holiday-sunrise, 2023-12-17:
Logs MQTT 

- Es werden kontinuierlich Werte geschrieben.
- Ein Neustart der APP solectrus_app_1 hilft.

ledermann, 2023-12-18:
Danke für deine Tests. Das werde ich mir in Ruhe ansehen müssen. Noch habe ich überhaupt keine Vermutung, woran das liegen könnte. Auch dass es nur beim Hausverbrauch auftritt, finde ich merkwürdig. Naja, ich melde mich wieder.

holiday-sunrise, 2023-12-19:
Rollback auf 0.13.2 war erfolgreich. Die Werte werden wieder angezeigt. Hat es mit der Animation zu tun ?

Kann ich Logs zusenden ?

ledermann, 2023-12-20:
Schwierige Sache, noch habe ich keine Idee, was die Ursache sein könnte. Mit der Animation dürfte das nichts zu tun habe. Es sieht eher danach aus, dass der Hausverbrauch auf 0 wechselt. Ich weiß nur nicht, warum das passiert. Niemand sonst berichtet von einem solchen Problem. Da bei dir die Daten per MQTT reinkommen, würde ich hier ansetzen, aber dann müsste es bei 0.13.2 genauso passieren, was ja nicht der Fall ist.

In der Version 0.14.1 habe ich ein [größeres Refactoring im Bereich der Verarbeitung der Messwerte](https://github.com/solectrus/solectrus/commit/9cf49fa6b86790bfd7aa9b79eb41ac3e145d6e15) durchgeführt, eine Ursache für das Phänomen kann ich aber darin nicht entdecken.

Wenn du im Docker-Log der App irgendwelche Fehlermeldungen hast, würden mir die natürlich weiterhelfen. Ich vermute aber mal, dass da keine sind.

Ich fasse nochmal kurz zusammen, was ich verstanden habe:

- Wechselt man auf das Tab `AKTUELL`, erscheint zunächst korrekt die Strombilanz und das Diagramm der letzten Stunde. Nach einigen Minuten verschwindet der Hausverbrauch aus der Strombilanz und der Balken `VERWENDUNG` ist dann komplett leer. 
- Das Problem tritt nur bei 0.14.1 auf, nicht aber bei 0.13.2

Fragen:
- Was ist mit den Kreisen für Autarkie und Verbrauch? Verschwinden die ebenfalls zur gleichen Zeit?
- Wie sieht es zu anderen Tageszeiten aus, wenn z.B. der Akku beladen wird oder eine Einspeisung stattfindet. Ist `VERWENDUNG` dann auch leer oder fehlt nur der Hausverbrauch?
- Wie lange dauert es genau, bis die Anzeige verschwindet? Schwankt die Zeit?
- Die gesamte Anzeige wird ja standardmäßig alle 5 Sekunden aktualisiert, linkst passt sich die Strombilanz an, rechts im Diagramm kommt ein Datenpunkt im Diagramm hinzu. Was genau passiert mit dem Diagramm, wenn der Hausverbrauch verschwindet? Bleibt das Diagramm dann stehen? 
- Hast du eine Wallbox? Gibt es das Feld `wallbox_charge_power` in deiner InfluxDB? 
- Funktioniert die 0.14.1 ansonsten problemlos? Oder gibt es andere seltsame Phänomene an anderen Stellen?

Es gibt Hinweise, dass es möglicherweise mit der Wallbox zu tun haben könnte. Hier ist es tatsächlich so, dass ein fehlender Wert (also NULL statt 0) zu einem Problem führen könnte. Allerdings wäre mir weiterhin unklar, warum das erst nach ein paar Minuten auftritt.

Ich hoffe, wir bekommen das Problem gemeinsam in den Griff. Ich brauche da deine Mithilfe, weil bei mir sowohl mit 0.13.2 als auch mit 0.14.1 alles korrekt erscheint.

ledermann, 2024-01-03:
Anmerkung von @q2doc, der in #2654 ebenfalls mit MQTT zu kämpfen hatte und folgendes berichtet:

> Das Issue (https://github.com/solectrus/solectrus/issues/2647), das nur auf 0.14.1 auftritt habe ich übrigens auch reproduzierbar. Der Neustart von MQTT Collector Container löst es für kurze Zeit. Hatte ich übrigens bei den vorherigen Versionen auch jedoch nur bei der Batterieanzeige die auch. nur sehr sporadische Änderungen schickt und dann nur noch nach Neustart des Containers auftauchte. Jedoch nicht bei der Hausanzeige... in der Hoffnung das es bei der Ursachensuche hilft.

Da ich selbst MQTT gar nicht im Einsatz habe, sind meine praktischen Erfahrungen damit gering. Der obige Hinweis bringt mich aber auf einen Gedanken:

Beim Start des MQTT-Collectors werden ja meist die aktuellen Messwerte vom Broker an den Client geschickt, im ioBroker heißt das z.B. "Publish states on subscribe". Wenn jetzt im weiteren Verlauf nicht permanent neue Messwerte reinkommen, dann könnte es zum fehlerhaften Darstellungen kommen.

ABER: Der Screenshot von oben (19:30) widerspricht dem. Denn er zeigt klar, dass lückenlos Messwerte zum Hausverbrauch vorliegen, nur bei AKTUELL fehlt der Wert in der Strombilanz. Für mich ist das weiterhin ein Rätsel. Alles deutet darauf hin, dass es an SOLECTRUS liegt, weil es in der Vorgängerversion ja funktionieren. Es kann also eher kein Problem in InfluxDB sein.

Ich bin sehr daran interessiert, der Sache weiter auf den Grund zu gehen. @holiday-sunrise: Falls du mir Messwerte von einem ausgewählten Tag als CSV zur Verfügung stellen kannst (am besten [exportiert direkt aus InfluxDB-Admin-Oberfläche](https://docs.influxdata.com/influxdb/v2/query-data/execute-queries/data-explorer/#save-as-csv)), dann würde ich versuchen, damit die Sache lokal nachzustellen.

ledermann, 2024-01-04:
Ich bin der Ursache auf der Spur. Kann es sein, dass Du keine Wallbox hast? Und es gar keine Werte für `wallbox_charge_power` in Deiner InfluxDB gibt?

Das weiter oben von mir erwähnte Refactoring führt zu einer unbeabsichtigten Verhaltensänderung, `NULL`-Werte werden nicht mehr als `0` behandelt. Das wäre leicht zu korrigieren.

@holiday-sunrise Bitte kurze Rückmeldung, ob meine Vermutung stimmt.

ledermann, 2024-01-04:
Ich habe jetzt einen Fix implementiert, der fehlende Werte als `0` interpretiert, was dem Verhalten der Version `v0.13.2` entspricht.

@holiday-sunrise: Du kannst das ausprobieren, indem du das [develop-Image](https://github.com/solectrus/solectrus/pkgs/container/solectrus?tag=develop) installierst. Ich vermute, dass damit das Problem bei nicht mehr auftritt.

Was mir allerdings weiterhin seltsam vorkommt ist, dass in deiner ursprünglichen Beschreibung davon die Rede war, dass das Problem erst nach ein paar Minuten auftritt, also nicht sofort. Das lässt sich mit dem diagnostizierten und behobenen Fehler eher nicht erklären.

ledermann, 2024-01-04:
Ich habe jetzt doch eine Erklärung: Beim Start des MQTT-Collectors holt dieser die aktuellen Messwerte vom Broker ab und schreibt sie nach InfluxDB - fehlende Werte werden dabei vom Collector (!) als 0 interpretiert und geschrieben. Hat man aber keine Wallbox, kommen keine Werte mehr hinzu. Da an einer Stelle im Code der `v0.14.1` die Berechnung bei einem NULL-Wert abgebrochen wird, entsteht die fehlerhafte Darstellung in der Strombilanz der JETZT-Seite. Die anderen Zeiträume werden intern anders berechnet, da gibt es den Fehler nicht.

@q2doc: Wieso auch die Batterietemperatur davon betroffen ist, habe ich jetzt nicht geprüft, aber das dürfte sich ähnlich verhalten. Das `NULL` → `0` wird jetzt für alle Datenfelder angewandt.

Bin gespannt, ob der Fehler auch bei @holiday-sunrise behoben ist.

ledermann, 2024-01-05:
@holiday-sunrise Den Export brauche ich vermutlich nicht mehr, da der Fehler aller Wahrscheinlichkeit nach behoben ist. Gib mir einfach Bescheid, wenn Du es getestet hast. Ich mache dann ein Bugfix-Update.

holiday-sunrise, 2024-01-05:
Develop funktioniert recht gut. Danke für den fix.

zu deinen Fragen:

>> Was ist mit den Kreisen für Autarkie und Verbrauch? Verschwinden die ebenfalls zur gleichen Zeit?
sind nicht vorhanden wie auf dem Screenshot zu sehen. Batterie, Temp werden auch nicht angzeigt.
>> Wie sieht es zu anderen Tageszeiten aus, wenn z.B. der Akku beladen wird oder eine Einspeisung stattfindet. Ist VERWENDUNG dann auch leer oder fehlt nur der Hausverbrauch?
auch fehlerhaft.
>> Wie lange dauert es genau, bis die Anzeige verschwindet? Schwankt die Zeit?
nein, es geht recht schnell das es nicht funktioniert. Die Annahme, das es ein paar Minuten dauert war nicht richtig.
>> Die gesamte Anzeige wird ja standardmäßig alle 5 Sekunden aktualisiert, linkst passt sich die Strombilanz an, rechts im Diagramm kommt ein Datenpunkt im Diagramm hinzu. Was genau passiert mit dem Diagramm, wenn der Hausverbrauch verschwindet? Bleibt das Diagramm dann stehen?
>>  Nee es bleibt nicht stehen
>> Hast du eine Wallbox? Gibt es das Feld wallbox_charge_power in deiner InfluxDB?
keine Werte in der InfluxDB
>> Funktioniert die 0.14.1 ansonsten problemlos? Oder gibt es andere seltsame Phänomene an anderen Stellen?
keine anderen Probleme gesichtet.

Sobald ich was finde melde ich mich.

ledermann, 2024-01-05:
Danke für deine Rückmeldung. Wenn ich dich recht verstehe, ist mit dem `develop`-Image jetzt alles korrekt, oder? - das "recht gut" klingt so, als würde es noch Macken geben. Deine Antworten beziehen sich vermutlich auf Version `0.14.1`, nicht auf `develop`.

Grundsätzlich steht `develop` immer zum Testen bereit (das ist quasi eine Beta-Version), ich verwende es auch für meine Installation (aka "Live-Demo"). Das Image wird aber mehrmals täglich aktualisiert, auch durch automatische Dependency-Updates, eine manuelle Installation wäre also etwas mühsam. Wenn du aber Watchtower am Start hast, hättest du automatisch immer die aktuelle Fassung. Über Beta-Feedback freue ich mich immer :)

ledermann, 2024-01-05:
Sehr schön, danke für deine Geduld und die vielen Tests und Rückmeldungen. Ich werde kurzfristig das develop-Image als v0.14.2 veröffentlichen.

closed 2024-01-05 by ledermann

holiday-sunrise, 2024-01-06:
Alles gut. Ist ja nicht lebenswichtig. Sondern unser Hobby 😉 Helfe immer wieder gern.