solectrus/solectrus #2654: Fehlerhafte Berechnung der Lademenge E-Auto

q2doc, 2023-12-20:
Hi Georg,

ich könnte schwören, dass es bisher funktioniert hat. Aktuell bekomme ich aber eine fehlerhafte Berechnung des Lademenge pro Tag/Monat. Im Livebild sieht es so aus:

[image]

also kurzer Peak (zu Demo Zweck). Die Rohdaten in der Influx sehen so aus:
[image]

&

[image]

auffällig ist beim der grafischen Darstellung, dass wohl der 0 Startwert falsch kommt?

Die Tagesansicht zeigt dann zwar den Peak korrekt aber eine viel zu Hohe kWh:

[image]

Die Daten kommen unverändert vom iObroker/Pi über MQTT und der Rest scheint zu stimmen. Sehe ich mal wieder den Wald vor Bäumen nicht?

ledermann, 2023-12-23:
Das ist tatsächlich etwas merkwürdig. Damit meine ich konkret die 6,9 kWh in der Strombilanz vom 20. Dezember. Besteht das Problem denn weiterhin? Wird an diesem Tag immer noch ein zu hoher Wert angezeigt? Also auch, nachdem der Tag bereits vorbei ist?

Falls ja, schlage ich Folgendes vor:

- Redis-Cache löschen (`docker exec -it solectrus-redis-1 redis-cli FLUSHALL`, muss mit OK antworten). Ändert das etwas?
- Du bist auf Version 0.14.1. Kannst Du probeweise auf 0.13.2 zurückgehen (in docker-compose.yml anderes Docker-Tag eintragen)? Aktuell gibt es ein anderes Issue (#2647), das nur auf 0.14.1 auftritt und bislang ungeklärt ist. Ich möchte ausschließen, ob es hier einen Zusammenhang gibt.

Wenn das keine Erkenntnisse bringt: Bitte erstelle mit InfluxDB ein CSV-Dump für diesen einen Tag (oder einen Zeitraum, das den Tag enthält), am besten mit allen Feldern und stelle mir den zur Verfügung. Das geht ganz leicht über das Admin-UI von InfluxDB. Diese CSV-Datei kann ich dann hier lokal importieren und nachstellen. Ein Fehler, den ich reproduzieren kann, ist fast schon behoben ;-)

q2doc, 2023-12-30:
Sorry hat etwas gedauert, also:

  - Redis-Cache löschen --> keine Änderung (OK kam)
  - auf Version 0.13.2 --> keine Änderung

der CSV-Dump 
[2023-12-30_17_52_influxdb_data.csv](https://github.com/solectrus/solectrus/files/13798719/2023-12-30_17_52_influxdb_data.csv)
 (musste dafür von InfluxDB2 2.7 auf 2.6 - was aber auch nichts geändert hat)

Ich vermute, das der kurze Peak um 18:57 zu dem Fehler führt. Sobald ich länger lade scheint es ok aber kurze Ladepeaks (die durch EVCC kommen) führen scheinbar zu einer fehlerhaften Lademenge.

q2doc, 2023-12-30:
gerade noch einmal nachgestellt, Ladevorgang mit korrekter Berechnung:

[image]

und dann ca. 1 Monte geladen:

[image]

leider schafft das weder meine Wallbox noch das Auto 😉 - scheint also wirklich an kurzen Ladepeaks zu liegen.

ledermann, 2023-12-30:
Danke für die Rückmeldung und insbesondere die CSV-Daten. Warum dazu InfluxDB 2.6 (statt 2.7) notwendig war, habe ich zwar nicht verstanden, aber mit den Daten kann ich das Problem reproduzieren.

Um zunächst klar zu formulieren, was ich als Problem überhaupt sehe: Eine Wallbox-Ladung über einen Zeitraum von ungefähr zwei Stunden (20.12.2023, abends etwa 20:30 bis 22:30) mit etwa 7kW plus in kleiner Peak von 3KW gegen 19:00 Uhr wird von SOLECTRUS links in der Strombilanz als 21 kWh dargestellt. 7 KW für zwei Stunden müssten aber etwa 14 kWh ergeben. SOLECTRUS zeigt also **zu viel** an.

Das Problem lässt sich bereits in InfluxDB nachstellen, die Ursache ist also nicht in SOLECTRUS selbst zu suchen. Folgende vereinfachte Query, auszuführen über das Admin-Frontend von InfluxDB, führt zur gleichen (falschen) Summe von 21 kWh:

```
from(bucket: "my-bucket")
|> range(start: 2023-12-19T23:00:00+01:00, stop: 2023-12-20T23:59:59+01:00)
|> filter(fn: (r) => r["_field"] == "wallbox_charge_power" )
|> integral(unit: 1h)
```

Bei Betrachtung der Rohdaten fällt mir auf, dass diese Lücken enthalten und am Schluss auch nicht auf 0 zurückgeht. Als Graph sieht das so in InfluxDB aus:

[image]

Schätzt man von dieser Kurve das Integral ab, erscheinen mir 21 kWh gar nicht mehr so falsch. Ursache ist, dass der Graph zwischen 13:45 und 18:49 von 0 bis auf 2000 linear steigt, obgleich gar keine Zwischenwerte vorhanden sind. Und am Schluss endet er "in der Luft".

Ursache sind meiner Ansicht nach nicht die kurzen Peaks, sondern fehlende Messpunkte. Ich habe ein paar Dinge ausprobiert, aber noch keine Lösung gefunden. Gesucht ist eine Flux-Query, die das Integral auch dann korrekt berechnet, wenn die Messreihe lückenhaft sind.

Oder, anderer Ansatz: Kannst Du dafür sorgen, dass Lücken gar nicht erst entstehen? Also indem kontinuierlich eine 0 geschrieben wird, wenn nicht geladen wird. So läuft es nämlich mit dem SENEC-Collector, sodass da das Problem gar nicht auftritt.

q2doc, 2023-12-30:
Danke, ja fehlende Werte hatte ich auch in verdacht (die lineare Steigung ist Quatsch...) und versucht über den ioBroker zumindest 0 Werte regelmäßig zu übertragen. Problem ist aber das die 0 irgendwie nicht immer (trotz korrekter Einstellung) übermittelt wird (versuche mal zu tricksen mit +- 0,0001 und teste morgen mal die Lücken zu füllen). EVCC zeigt den realen (= Zählerstand) Wert an:

[image]

also 16 kWh.

Ach so influx 2.7 hat bei mir keinen CSV export button und mit dem Query scheitere ich immer... 😒

q2doc, 2024-01-02:
Also der "Trick" funktioniert. Beim Verbinden des Autos schreibt ein Javascript in ioBroker 10 mal in 10 ms Abstand 0 und 1e-10. In Influx sieht das dann korrekt aus:

[image]

und auch Solectrus zeigt richtig an:

[image]

&

[image]

Es scheint also an fehlenden/fehlerhaften Nullwerten zu liegen.

Das Issue (https://github.com/solectrus/solectrus/issues/2647), das nur auf 0.14.1 auftritt habe ich übrigens auch reproduzierbar. Der Neustart von MQTT Collector Container löst es für kurze Zeit. Hatte ich übrigens bei den vorherigen Versionen auch jedoch nur bei der Batterieanzeige die auch. nur sehr sporadische Änderungen schickt und dann nur noch nach Neustart des Containers auftauchte. Jedoch nicht bei der Hausanzeige... in der Hoffnung das es bei der Ursachensuche hilft.

closed 2024-01-03 by ledermann