solectrus/solectrus #2671: 500 internal server error

StevieC121176, 2023-12-27:
seit der Version von heute morgen bekomme ich beim Aufruf immer einen einen 500 internal server Fehler.
geändert an meiner config habe ich nichts, nur die neue Version installiert. Die Version von gestern lief ohne Probleme. Hier mal das Log:

```solectrus-app-1                 | I, [2023-12-27T07:18:19.358938 #1]  INFO -- : [192.168.178.22] method=GET path=/tiles/inverter_power/now format=html controller=TilesController action=show status=200 allocations=8482 duration=91.44 view=90.53 db=0.00
solectrus-app-1                 | I, [2023-12-27T07:18:19.479541 #1]  INFO -- : [192.168.178.22] method=GET path=/tiles/inverter_power/year format=html controller=TilesController action=show status=200 allocations=31924 duration=280.58 view=194.95 db=83.95
solectrus-app-1                 | I, [2023-12-27T07:18:19.536700 #1]  INFO -- : [192.168.178.22] method=GET path=/tiles/savings/year format=html controller=TilesController action=show status=200 allocations=41104 duration=366.34 view=307.87 db=41.25
solectrus-app-1                 | I, [2023-12-27T07:18:19.546849 #1]  INFO -- : [192.168.178.22] method=GET path=/tiles/co2_savings/year format=html controller=TilesController action=show status=200 allocations=39731 duration=335.85 view=281.23 db=52.83
solectrus-app-1                 | I, [2023-12-27T07:18:19.552334 #1]  INFO -- : [192.168.178.22] method=GET path=/tiles/inverter_power/month format=html controller=TilesController action=show status=200 allocations=36390 duration=296.09 view=256.29 db=38.96
solectrus-app-1                 | I, [2023-12-27T07:18:19.633636 #1]  INFO -- : [192.168.178.22] method=GET path=/tiles/inverter_power/day format=html controller=TilesController action=show status=200 allocations=3279 duration=62.82 view=61.09 db=0.91
solectrus-mqtt-collector-1      | 2023-12-27 07:18:19 +0000 {"inverter_power"=>115}
solectrus-mqtt-collector-1      | 2023-12-27 07:18:19 +0000 {"mpp1_power"=>102}
solectrus-mqtt-collector-1      | 2023-12-27 07:18:19 +0000 {"mpp2_power"=>13}
solectrus-mqtt-collector-1      | 2023-12-27 07:18:19 +0000 {"mpp3_power"=>0}
solectrus-mqtt-collector-1      | 2023-12-27 07:18:21 +0000 {"bat_fuel_charge"=>46.0}
solectrus-mqtt-collector-1      | 2023-12-27 07:18:21 +0000 {"bat_power_plus"=>0, "bat_power_minus"=>18}
solectrus-mqtt-collector-1      | 2023-12-27 07:18:21 +0000 {"grid_power_plus"=>3300, "grid_power_minus"=>0}
solectrus-mqtt-collector-1      | 2023-12-27 07:18:21 +0000 {"house_power"=>3433}
solectrus-mqtt-collector-1      | 2023-12-27 07:18:21 +0000 {"wallbox_charge_power"=>0}
solectrus-app-1                 | I, [2023-12-27T07:18:22.341494 #1]  INFO -- : [192.168.178.22] method=GET path=/essentials format=html controller=EssentialsController action=index status=500 allocations=84680 duration=254.04 view=0.00 db=0.00
solectrus-app-1                 | E, [2023-12-27T07:18:22.345342 #1] ERROR -- : [192.168.178.22]   
solectrus-app-1                 | [192.168.178.22] ActionView::Template::Error (Vite Ruby can't find entrypoints/application.ts in the manifests.
solectrus-app-1                 | 
solectrus-app-1                 | Possible causes:
solectrus-app-1                 |   - The file path is incorrect.
solectrus-app-1                 |   - The file is not in the entrypoints directory.
solectrus-app-1                 |   - Some files are outside the sourceCodeDir, and have not been added to watchAdditionalPaths.
solectrus-app-1                 |   - You have not run `bin/vite dev` to start Vite, or the dev server is not reachable.
solectrus-app-1                 |   - "autoBuild" is set to `false` in your config/vite.json for this environment.
solectrus-app-1                 | 
solectrus-app-1                 | Visit the Troubleshooting guide for more information:
solectrus-app-1                 |   https://vite-ruby.netlify.app/guide/troubleshooting.html#troubleshooting
solectrus-app-1                 | 
solectrus-app-1                 | 
solectrus-app-1                 | 
solectrus-app-1                 | 
solectrus-app-1                 | ):
solectrus-app-1                 | [192.168.178.22]     51:     = csrf_meta_tags
solectrus-app-1                 | [192.168.178.22]     52: 
solectrus-app-1                 | [192.168.178.22]     53:     = vite_client_tag
solectrus-app-1                 | [192.168.178.22]     54:     = vite_typescript_tag 'application', 'data-turbo-track': 'reload'
solectrus-app-1                 | [192.168.178.22]     55:     = vite_stylesheet_tag 'application.css', 'data-turbo-track': 'reload', media: 'screen'
solectrus-app-1                 | [192.168.178.22]     56: 
solectrus-app-1                 | [192.168.178.22]     57:     - if Rails.env.test?
solectrus-app-1                 | [192.168.178.22]   
solectrus-app-1                 | [192.168.178.22] app/views/layouts/application.html.slim:54
solectrus-app-1                 | [192.168.178.22] app/controllers/concerns/auto_locale.rb:14:in `switch_locale'
solectrus-app-1                 | I, [2023-12-27T07:18:22.369592 #1]  INFO -- : [192.168.178.22] method=GET path=/500 format=html controller=ErrorsController action=show status=500 allocations=8420 duration=21.91 view=0.00 db=0.00
solectrus-app-1                 | Error during failsafe response: Vite Ruby can't find entrypoints/application.ts in the manifests.
solectrus-app-1                 | 
solectrus-app-1                 | Possible causes:
solectrus-app-1                 |   - The file path is incorrect.
solectrus-app-1                 |   - The file is not in the entrypoints directory.
solectrus-app-1                 |   - Some files are outside the sourceCodeDir, and have not been added to watchAdditionalPaths.
solectrus-app-1                 |   - You have not run `bin/vite dev` to start Vite, or the dev server is not reachable.
solectrus-app-1                 |   - "autoBuild" is set to `false` in your config/vite.json for this environment.
solectrus-app-1                 | 
solectrus-app-1                 | Visit the Troubleshooting guide for more information:
solectrus-app-1                 |   https://vite-ruby.netlify.app/guide/troubleshooting.html#troubleshooting
solectrus-app-1                 | 
solectrus-app-1                 | 
solectrus-app-1                 | 
solectrus-app-1                 | 
solectrus-app-1                 | 
solectrus-app-1                 |   /usr/local/bundle/gems/vite_ruby-3.5.0/lib/vite_ruby/manifest.rb:215:in `missing_entry_error'
solectrus-app-1                 |   /usr/local/bundle/gems/vite_ruby-3.5.0/lib/vite_ruby/manifest.rb:79:in `lookup!'
solectrus-app-1                 |   /usr/local/bundle/gems/vite_ruby-3.5.0/lib/vite_ruby/manifest.rb:28:in `block in resolve_entries'
solectrus-app-1                 |   /usr/local/bundle/gems/vite_ruby-3.5.0/lib/vite_ruby/manifest.rb:28:in `map'
solectrus-app-1                 |   /usr/local/bundle/gems/vite_ruby-3.5.0/lib/vite_ruby/manifest.rb:28:in `resolve_entries'
solectrus-app-1                 |   /usr/local/bundle/gems/vite_rails-3.0.17/lib/vite_rails/tag_helpers.rb:46:in `vite_javascript_tag'
solectrus-app-1                 |   /usr/local/bundle/gems/vite_rails-3.0.17/lib/vite_rails/tag_helpers.rb:59:in `vite_typescript_tag'
[... 286 lines cut ...]
solectrus-app-1                 |   /usr/local/bundle/gems/railties-7.1.2/lib/rails/rack/logger.rb:24:in `block in call'
solectrus-app-1                 |   /usr/local/bundle/gems/activesupport-7.1.2/lib/active_support/tagged_logging.rb:135:in `block in tagged'
solectrus-app-1                 |   /usr/local/bundle/gems/activesupport-7.1.2/lib/active_support/tagged_logging.rb:39:in `tagged'
solectrus-app-1                 |   /usr/local/bundle/gems/activesupport-7.1.2/lib/active_support/tagged_logging.rb:135:in `tagged'
solectrus-app-1                 |   /usr/local/bundle/gems/activesupport-7.1.2/lib/active_support/broadcast_logger.rb:240:in `method_missing'
solectrus-app-1                 |   /usr/local/bundle/gems/railties-7.1.2/lib/rails/rack/logger.rb:24:in `call'
solectrus-app-1                 |   /usr/local/bundle/gems/actionpack-7.1.2/lib/action_dispatch/middleware/remote_ip.rb:92:in `call'
solectrus-app-1                 |   /usr/local/bundle/gems/request_store-1.5.1/lib/request_store/middleware.rb:19:in `call'
solectrus-app-1                 |   /usr/local/bundle/gems/actionpack-7.1.2/lib/action_dispatch/middleware/request_id.rb:28:in `call'
[... cut ...]
solectrus-app-1                 |   /usr/local/bundle/gems/puma-6.4.0/lib/puma/request.rb:99:in `handle_request'
solectrus-app-1                 |   /usr/local/bundle/gems/puma-6.4.0/lib/puma/server.rb:443:in `process_client'
solectrus-app-1                 |   /usr/local/bundle/gems/puma-6.4.0/lib/puma/server.rb:241:in `block in run'
solectrus-app-1                 |   /usr/local/bundle/gems/puma-6.4.0/lib/puma/thread_pool.rb:155:in `block in spawn_thread'
solectrus-app-1                 | I, [2023-12-27T07:18:22.709209 #1]  INFO```

ledermann, 2023-12-27:
Auf develop habe ich heute auf Ruby 3.3 aktualisiert, vermutlich liegt es daran. 

Überhaupt kann auf develop alles Mögliche schiefgehen. Bei Problemen besser auf latest wechseln.

ledermann, 2023-12-28:
Die relevante Zeile im Log scheint diese zu sein:

```
[192.168.178.22] ActionView::Template::Error (Vite Ruby can't find entrypoints/application.ts in the manifests.
```

Der Fehler tritt (im `develop`-Branch) seit dem [Update auf Ruby 3.3](https://github.com/solectrus/solectrus/commit/d935bd80ce0c7debe481b82c2a8463070ce7a782) auf, aber scheinbar nur auf der `aarch64`, also z.B. einem Raspi. Auf `x86_64` läuft es problemlos. Ursache noch unklar. 

Die aktuelle Version 0.14.1 ist **nicht** betroffen, da diese noch auf Ruby 3.2 läuft.

StevieC121176, 2023-12-28:
Ja, ich habe jetzt erstmal auf `latest` umgestellt und die 14.1 installiert.

ledermann, 2023-12-28:
Es scheint sich wohl um diesen Fehler zu handeln, der im Zusammenhang mit Ruby 3.3 auf `aarch64-linux` auftritt und voraussichtlich kurzfristig behoben wird:

https://bugs.ruby-lang.org/issues/20085
https://github.com/ruby/ruby/pull/9371

closed 2023-12-28 by ledermann

ledermann, 2023-12-28:
Die Lösung sieht nun so aus, dass der `develop`-Branch wieder Ruby 3.2 verwendet, somit funktioniert es auch wieder auf dem Raspi.

Mit dem [Upgrade auf Ruby 3.3](https://github.com/solectrus/solectrus/pull/2674) warte ich, bis der oben erwähnte Bug behoben ist.

commit d9a8edc4 (2023-12-28): Temporary revert to Ruby 3.2

Fixes #2671