solectrus/solectrus #2715: Influx DB won't start

Timokirch, 2024-01-08:
Hi @ledermann, 

Unfortunately my influx-db container decided to be unhealthy as of today. Do you know a good medicine to get it back to a healthy state? 

```

See 'influxd -h' for help
ts=2024-01-08T20:42:20.792494Z lvl=error msg="Failed opening bolt" log_id=0mbhp~E0000 error="unable to open boltdb: timeout"
2024-01-08T20:42:27.    info    found existing boltdb file, skipping setup wrapper      {"system": "docker", "bolt_path": "/var/lib/influxdb2/influxd.bolt"}
2024-01-08T20:42:27.    info    found existing boltdb file, skipping setup wrapper      {"system": "docker", "bolt_path": "/var/lib/influxdb2/influxd.bolt"}
ts=2024-01-08T20:42:27.815629Z lvl=info msg="Welcome to InfluxDB" log_id=0mbhqUPl000 version=v2.7.4 commit=19e5c0e1b7 build_date=2023-11-08T17:07:39Z log_level=info
ts=2024-01-08T20:42:27.815980Z lvl=warn msg="nats-port argument is deprecated and unused" log_id=0mbhqUPl000
Error: unable to open boltdb: timeout
See 'influxd -h' for help
ts=2024-01-08T20:42:28.775371Z lvl=error msg="Failed opening bolt" log_id=0mbhqUPl000 error="unable to open boltdb: timeout"
``` 

Kind regards and many thanks in advance 
Timo

ledermann, 2024-01-08:
Never seen this before, but just found this:
https://github.com/influxdata/influxdb/issues/24320

Does this help?

Timokirch, 2024-01-09:
Hi @ledermann, 

Thank you very much for the fast reply. I can't say what it actually was but somehow the file was not there anymore. Maybe i did something horrible wrong (by accident of course). I've restored a backup of my container and everything is up again. 

Case closed 

Kind Regards

closed 2024-01-09 by Timokirch