solectrus/solectrus #2795: Werte unter 1 KW als Watt darstellen

ledermann, 2024-02-07:
Vorschlag per E-Mail:

Umschaltung zwischen W und kWh erst ab der 4. Stelle, also damit z.B. 236 Watt und nicht 0.2 kWh dargestellt werden. Platz hätte es ja überall. Für kleine Anlagen wie ich sie habe sehr wünschenswert.

StevieC121176, 2024-02-09:
Sehr guter Vorschlag.

Zeppelin-PV, 2024-02-21:
Auch ich befürworte diesen Vorschlag

ledermann, 2024-02-21:
Ok, ist gemacht und fließt ins nächste Update :)

closed 2024-02-21 by ledermann

GrimmiMeloni, 2024-03-23:
Können wir hierüber nochmal verhandeln? 😄 
Ich persönlich empfinde die Vermengung von Watt und Kilowatt die jetzt in den Grafiken entsteht eher suboptimal. Wäre hier ggf. denkbar zumindest die Darstellung einheitlich zu halten? Also, wenn in Watt, dann alle Werte in Watt? Oder alternativ, eine Konfiguration für das alte Verhalten?

ledermann, 2024-03-23:
Klar können wir verhandeln :) Ich bin auch nicht ganz glücklich mit der jetzigen Darstellung, aber aus noch einem anderen Grund: Kleinste Änderungen an den Messwerten werden zu sehr hervorgehoben. Ob im Haus gerade `510 W` oder `520 W` verbraucht werden - die sich ändernde Zahl sorgt für Unruhe in der Darstellung. Mir persönlich würde in diesem Fall die Angabe von `0,5 kW` genügen.

Einen zusätzlichen Button zur Wahl zwischen `W` und `kW` möchte ich vermeiden. Außerdem wird ja bei sehr großen Zahlen, z.B. in der Jahresansicht, auch auf `MW` umgeschaltet. Diese Abhängigkeit der Einheit von der Zahlengröße finde ich eigentlich sehr sinnvoll. Derzeitig gibt es in der Strombilanz folgende Regel:

| Grenzwert         | Anzeige in ... |
|-------------------|----------------|
| Ab 1.000.000 Watt | MW             |
| Ab 1.000 Watt     | kW             |
| darunter          | W              |

In Diagrammen wird hingegen **immer** kW verwendet. Ja, das ist inkonsistent. Aber in den Diagrammen gibt es fast immer größere Ausschläge, die eine Darstellung in `W` nicht sinnvoll erscheinen lassen.

Wie kann denn eine ideale Darstellung aussehen? Ich tendiere dazu, es wieder in den vorherigen Stand zu setzen, also `kW` bereits ab ´100 Watt`. Aber der ursprüngliche Wunsch war ja gerade, diese Details anzuzeigen.

StevieC121176, 2024-03-23:
Oh nein bitte nicht, evtl. wäre eine Config per .env möglich?

Zeppelin-PV, 2024-03-23:
Ich bin sehr zufrieden mit der unterschiedlichen Darstellung W, kW, MW wie sie seit ver14.4. ist.
ein DIN A4 Blatt ist 21cm breit, und nicht 0,21m
kleine Dinge misst man eben mit kleinen Einheiten
... bitte so lassen, oder Benutzer einstellbar.

ledermann, 2024-03-23:
Ok, einfach zurückdrehen ist wohl keine Option ;-) Es läuft wohl auf eine Einstellmöglichkeit hinaus. Aber wie genau? Eine Möglichkeit:

**Darstellung der elektrischen Leistung**
A: Möglichst genau
B: Gerundet 

Bei A: kW erst ab 1000 Watt (also wie seit Version 0.14.3)
Bei B: kW ab 100 Watt (also wie bis Version 0.14.2)

Die Option käme dann ins UI.

StevieC121176, 2024-03-23:
Ja so könnte man es machen.

Zeppelin-PV, 2024-03-23:
... sollte irgendwann mal mein Wunsch #2621 in Erfüllung gehen wäre zweimal das "A" die vernünftigste Anzeigeform
wenn A mit B oder B mit A einstellbar wäre hätte doch jeder was er möchte 
betrifft das die Bilanz- als auch die Kachelansicht ?

ledermann, 2024-03-23:
Die Einstellung würde grundsätzlich gelten, also für sämtliche Leistungsmesswerte und auch in der Kachelansicht.