solectrus/solectrus #2796: Mobile: Untere Navigation weniger eng gestalten

ledermann, 2024-02-07:
Per E-Mail erhalten:

Mit dem iPhone kollidiert die Navigation unten mit der systemweiten Funktion für das Fenster (swipen der offenen Fenster, verkleinern, Einhandbedienung). Könnte man hier etwas mehr margin-bottom geben?

closed 2024-02-11 by ledermann

StevieC121176, 2024-02-12:
Wäre es möglich die Beschriftung der Buttons mittig zu gestalten? 

[image]

ledermann, 2024-02-12:
@StevieC121176 Von welchem iOS/iPhone stammt der Screenshot? Bei mir (mit iOS 17.3.1 auf iPhone 15 Pro Max) sieht es anders (und besser) aus, die Homebar integriert sich da ins UI der App, steht also nicht darunter. Daher habe ich nur den unteren Rand vergrößert.

[image]

StevieC121176, 2024-02-12:
Den Screenshot hatte ich vorhin mit meinem iPhone 15pro erstellt mit ios17.4 . Ich fürchte das das kein Bug von Apple ist, sondern so gewollt (keine Progressive Darstellung mehr). Ich denke dadurch wirkt das so bescheiden.
Leider wird man irgendwann gezwungen sein auf ios17.4 oder höher zu wechseln.

commit 6357781f (2024-02-11): UI: Increase bottom margin on mobile

This prevents the bottom navigation from being too close
to the iOS home bar

Resolves #2796