solectrus/solectrus #2814: Vollbilddarstellung unter iOS (Safari)

StevieC121176, 2024-02-09:
Ich weiß jetzt nicht ob es durch die aktualisierte Solectrus oder iOS Version auftritt, aber bei mir wird Solectrus durch öffnen auf dem homescreen nicht mehr als vollbild angezeigt. Zur Verdeutlichung hier mal 2 Screenshots:

[image]
[image]

ledermann, 2024-02-09:
Welches iOS? Ich habe genau von diesem Phänomen gehört, das unter 17.4-beta auftritt. Auf meinem 17.3.1 ist alles ok.

StevieC121176, 2024-02-09:
Ja ich nutze 17.4 Beta.

_In der EU sind sogenannte „progressive Web-Apps“ unter iOS 17.4 nicht mehr progressiv.

ledermann, 2024-02-09:
Mal abwarten, wie sich das so entwickelt. Derzeitig herrscht [große Aufregung](https://stadt-bremerhaven.de/ios-17-4-unterstuetzung-fuer-web-apps-pwa-anscheinend-eingestellt/) über die anstehenden Änderungen Apples für den EU-Raum. Möglicherweise sind aber auch noch Bugs im Spiel.

Das Problem beschränkt sich nicht auf SOLECTRUS, sondern betrifft grundsätzlich alle Websites, die als App "installiert" werden. Ich verfolge das mit Sorge, da ich PWAs für eine ziemlich feine Sache halte.

ledermann, 2024-02-14:
Nix Neues zum Thema, aber gute Zusammenfassung der Situation:
https://www.heise.de/news/Web-Apps-ausser-Funktion-iOS-17-4-Beta-bereitet-Entwicklern-grosse-Sorge-9627974.html

StevieC121176, 2024-02-14:
echt grauenvoll was die da veranstalten...
zumindest gibt es eine Möglichkeit wenigstens den Look&Feel wieder herzustellen. Nach dem öffnen in der Adresszeile auf das"aA" klicken und dann Symbolleiste ausblenden wählen. 
Leider muss man das nach jedem öffnen erneut machen.

[image]
[image]

Timokirch, 2024-02-15:
https://developer.apple.com/support/dma-and-apps-in-the-eu/

spirit0607, 2024-02-18:
Betrifft das alle Browser in iOS 17.4? Oder nur Safari? Mit Edge z.B. funktioniert Vollbild ja auch.

StevieC121176, 2024-03-05:
Apple ist zurück gerudert. Mit dem heutigen iOS 17.4 funktionieren PWAs wieder.

closed 2024-03-05 by StevieC121176