solectrus/solectrus #2841: Anzeige Einspeisung immer mit Icon „!“

StevieC121176, 2024-02-21:
Bei mir wird neuerdings die Einspeisung immer mit einem Icon „!“ versehen. Im Tooltip steht Begrenzung 0%. Bei mir ist aber keinerlei Begrenzung eingestellt da es bei der Inbetriebnahme meiner Anlage die 70% Regel nicht mehr gab.

[image]

[image]

hambue, 2024-02-21:
Schau mal in den Logs der letzten Tage, ob es da infolge eines NPU-Fehlers zu einer externen Steuerung kam. Manchmal bleiben da komische Werte wie eine Einspeisebegrenzung von 1W stehen und verschwinden erst wieder nach einem Neustart des Speichers. Der genaue Eintrag in den Logs lautet dann `External Charge Control`.

ledermann, 2024-02-21:
Ich denke, bei @StevieC121176 ist gar kein SENEC im Einsatz, daher muss er sich auch nicht mit NPU-Fehlern herumplagen ;-)

Der SENEC-Collector schreibt ins Feld `power_ratio` der InfluxDB die mögliche Einspeisebegrenzung rein. Typische Werte sind 70 oder 100. Der MQTT-Collector kann das in der aktuellen Version noch nicht, d.h. bei Verwendung des MQTT-Collectors gibt es diesen Wert nicht in InfluxDB.

Und jetzt entsteht der Fehler: SOLECTRUS interpretiert seit Version `0.14.2` diesen fehlenden Wert als 0 und stellt ihn dar. Ist natürlich Blödsinn. Ich korrigiere das.

ledermann, 2024-02-21:
Fixed in `develop`. @StevieC121176 Passt es so?

closed 2024-02-21 by ledermann

StevieC121176, 2024-02-22:
Scheint zu funktionieren, das ! Icon ist bis jetzt nicht wieder aufgetaucht.

[image]

ledermann, 2024-02-22:
@StevieC121176 Prima, danke für die Rückmeldung!