solectrus/solectrus #2863: Wechsel Aktuell/Tag/Woche... schaltet immer auf "Erzeugung" um

gereons, 2024-02-27:
Mit der aktuellen develop-Version v0.14.2-186-g30443d6b ist mit aufgefallen, dass beim Wechsel des Zeitraums auch immer auf die Erzeugungs-Darstellung gewechselt wird. Früher war es so, dass bei Betrachtung von zB. Ladestand/Tag und dann Klick auf "Woche" ich den Ladestand im Wochenverlauf sehen konnte. Das kostet jetzt extra-Klicks.

Technisch ausgedrückt: ich bin z.B. auf `https://<mein solectrus>/house_power/2024-W09` und klicke auf "Jahr", lande damit auf 
`https://<mein solectrus>/inverter_power/2024` statt wie bisher auf `https://<mein solectrus>/house_power/2024`

Bug oder Feature?

gereons, 2024-02-27:
Ist womöglich mit 5d0be6ec7503c9d34cace460c280f9da029778d4 passiert?

closed 2024-02-27 by ledermann

ledermann, 2024-02-27:
Gut beobachtet, genau so ist es. Das war ein unbedachter Seiteneffekt der gestrigen "Optimierung". Ich habe das korrigiert. Danke für den Hinweis!

commit 31fca01c (2024-02-27): Revert "Fix double-rendering nav"

This reverts commit 5d0be6ec7503c9d34cace460c280f9da029778d4.

Fixes #2863