solectrus/solectrus #2884: Größe der Icons bei geringer Höhe der Anzeigefelder

StevieC121176, 2024-03-04:
Die Icons werden bei abnehmender Höhe der Anzeigefelder teilweise zu groß dargestellt bzw. sogar verdeckt 

[image]
[image]
[image]

ledermann, 2024-03-04:
Hm, offensichtlich ein Fehler. Welche Version? Ich nehme an, die aktuelle Betaversion https://github.com/solectrus/solectrus/pull/2836.

Ist dir das vorher schon mal begegnet? Tritt das auch mit `develop` auf?

Hintergrundinfo: SOLECTRUS ermittelt einen Peak der Werte aus den letzten 30 Tagen und benutzt das für die Skalierung von Icons. Möglicherweise stimmt etwas mit der Peak-Ermittlung nicht.

In dem Zusammenhang noch folgender Hinweis: Du hast ja in den letzten Tage die Datenbank mehrfach neu aufgesetzt. Es ist ja so, dass InfluxDB-Abfragen in einem Redis-Cache liegen. Möglicherweise passt der Cache nicht zum Stand der InfluxDB. Daher einfach mal den Cache löschen und gucken, ob das Auswirkungen hat:

```
docker exec -it solectrus-redis-1 redis-cli FLUSHALL
```

StevieC121176, 2024-03-04:
Ja die aktuelle Beta #2836 
Tritt aber auch mit der develop Version auf.
Aber es ist mir tatsächlich erst in den letzten Tagen aufgefallen. Ich dachte erst an ein Feature bis ich heute gesehen hatte das das Icon abgeschnitten wird.

`docker exec -it solectrus-redis-1 redis-cli FLUSHALL`

hat geholfen. Die Icon sind wieder normal groß

[image]

ledermann, 2024-03-04:
Ok, dann waren also im Cache noch Daten, die nicht zur neu aufgesetzten InfluxDB passten. Somit alles geklärt, nichts weiter zu tun.

closed 2024-03-04 by ledermann