solectrus/solectrus #2905: Display 'over Size' bei Auswahl Diagramme (ver.14.3)

Zeppelin-PV, 2024-03-10:
hallo Georg
die Darstellung auf dem Smartphone (Galaxy A54-6,4Zoll) ist beim start normal.
Nachdem aber eines der Diagramme angewählt wird (DropDown Menü) wird die Anzeige größer und man muss das Display scrollen um entweder nach oben zu kommen bzw. nach unten auf aktuell, Tag, etc. zu kommen.
Der Bereich des Diagramm wird bei Aufruf etwas größer dargestellt, als bei beim Start, setzt sich aber nicht mehr auf diese Größe zurück.
Vielleicht liese sich dies mit einem Klick außerhalb des Diagramm bewerkstelligen.
Auf dem 10Zoll Tablet tritt dieser Effekt nicht auf.
LG Ernst

ledermann, 2024-03-10:
Ich kann das derzeitig noch nicht reproduzieren, auf diversen iOS-Geräten scheint alles korrekt zu sein. Daher erst mal ein paar Rückfragen:

- Ich nehme an, das beschriebene Phänomen tritt erst mit Version `0.14.3` auf, nicht aber zuvor mit `0.14.2`. Richtig?
- Das Dropdown-Menü habe ich in `0.14.3` wegen eines anderes Problem verändert. Ist es richtig, dass das Problem wirklich nur beim Dropdown auftritt, nicht aber beim direkten Klick in ein Strombilanz-Segment?
- Kannst du einen Screenshot erstellen, auf dem man die falsche Darstellung sehen kann?
- Kann jemand anderes das Problem reproduzieren? Wenn ja, welches Gerät?

Zeppelin-PV, 2024-03-10:
- ja, erst seit 14.3
- ja, nur beim DropDown Menü, nicht über die Segmente (da bleibt die Größe)

Anzeige direkt nach Start
[image]

Anzeige unten aktuell, Tag etc. fehlt
[image]

Anzeige oben Strombilanz Menü fehlt
[image]

Zeppelin-PV, 2024-03-10:
mit Lineal gemessen wird das Diagramm über die DropDown Auswahl 5mm größer in der Höhe
mit dem Android Menü kommt man aber stressfrei zurück 
[image]

Zeppelin-PV, 2024-03-10:
... und noch eine Feststellung - es ist nur im Firefox Browser nicht im Google Chrome

ledermann, 2024-03-10:
Danke für den Input. Ich verstehe nun, was passiert, aber ich kenne den Grund nicht. Dass es nur in Firefox auftritt, ist interessant. Ich selbst teste hier zwar mit den verschiedenen Browsern, auch Firefox, habe aber momentan kein Android zur Verfügung.

Dennoch habe ich eine Änderung versucht, die sich am vorherigen Zustand (aus 0.14.2) orientiert. Möglicherweise löst es das Problem. Wenn du magst, kannst du von "latest" auf "develop" wechseln, um das auszuprobieren.

Zeppelin-PV, 2024-03-10:
... würde ich gerne, auf dem zweiten Raspi
mit einem kurzen 'how to...'   :)

ledermann, 2024-03-10:
Howto: `docker-compose.yml` editieren, dort unter `services.app.image` von `:latest` auf `:develop` wechseln. Speichern. Dann `docker compose pull`, dann `docker compose up -d`. Fertig :)

Zeppelin-PV, 2024-03-10:
bei zwei Raspi' und zwei Browsern kann man schnell mal durcheinander geraten.
Google Chrome hatte das Problem, nicht der Firefox, aber - ein Raspi läuft jetzt auf 'develop' und das Problem ist beseitigt.
Großartig, DANKE Georg

ledermann, 2024-03-11:
Prima, somit ist der Bug erledigt. Mal sehen, wann ich ein weiteres Update mache, vielleicht kommt ja noch was hinzu.

closed 2024-03-11 by ledermann