solectrus/solectrus #2956: Strompreise und Tibber

hj-hefer, 2024-03-23:
Wäre schön für die Eerechnung der Ersparnis auch den Tbberpreis einzugeben statt einen festen Wert.
Ist das iregendwie möglich oder steht soetwas in den Startlöchern?

ledermann, 2024-03-23:
Ja, das wäre auf jeden Fall wünschenswert. Für die Ermittlung der individuellen Tibber-Preise gibt es ja bereits den [Tibber-Collector](https://github.com/solectrus/tibber-collector/). Der schreibt die stündlich sich ändernden Preise in die InfluxDB. Verwendet werden die Preise bislang aber ausschließlich vom [SENEC-Charger](https://github.com/solectrus/senec-charger), um den Speicher kostenoptimiert aus dem Netz zu beladen. 

In SOLECTRUS selbst (also dem Dashboard) werden die Tibber-Preise noch nicht verwendet. Das wäre möglich, ist aber keineswegs trivial. Der bisher verwendete Algorithmus zur Kosten- und Ersparnisberechnung funktioniert gut für Preise, die sich selten (d.h. wenige Male pro Jahr) ändern, aber nicht für `365 * 24` Preisänderungen pro Jahr. Da muss ich mir einen ganz anderen Ansatz für ausdenken.

Da ich grundsätzlich noch so einiges bezüglich Kosten/Ersparnis-Berechnung vorhabe, werde ich das Thema später nochmal aufgreifen.

ledermann, 2024-04-04:
Fortführung in #2150

closed 2024-04-04 by ledermann