solectrus/solectrus #3007: Auswahl beliebiger Zeiträume

ledermann, 2024-04-11:
Neben der bereits möglichen Auswahl **einzelner** Jahre, Monate und Wochen sollen auch **beliebige** Zeiträume ausgewertet werden können - definiert durch Angabe eines Start- und eines Ende-Datums.

Dafür braucht es mehrere Dinge:

1) Eine Zeitraumauswahl, sinnvollerweise mit einem Datepicker (siehe auch #1931).

2) Beim Diagramm müsste eine Automatik ergänzt werden, die eine sinnvolle Gruppierung bestimmt. Ein großer Zeitraum sollte mit Balken für jeden Monat erfolgen, bei kleineren vermutlich besser Tage oder Wochen. Denkbar wäre auch, die Gruppierung vom Benutzer festlegen zu lassen. 

3) Hat man einen größeren Zeitraum gewählt, der nicht nur vollständige Monate oder Wochen umfasst (z.B. 10.09.2023 bis 27.03.2024), dann wäre der erste und/und der letzte Balken kein vollständiger Monat bzw, Woche. Das müsste dann entsprechend visualisiert werden, z.b. durch schraffierte Darstellung o.ä.

(Vorgeschlagen in #3004)

gereons, 2024-04-12:
Ich vermisse so ein Feature auch sehr. In erster Näherung wäre für mich aber schon viel gewonnen, wenn man aus dem starren Kalender-Schema ausbrechen könnte und zB eine Woche wahlweise nicht als Kalenderwoche sondern "heute und die letzten 6 Tage davor" definiert wäre - dann würde so eine Wochenansicht am Montag auch wieder sinnvolles zeigen. Monat und Jahr könnten analog funktionieren.

ledermann, 2024-04-12:
@gereons Ja, absolut verständlich. Der Hauptgrund, der mich bislang von einer Implementierung abhält, ist kurioserweise die Wahl eines geeigneten Datepickers. 

Da du ja selbst auch Entwickler bist, frage ich mal so: Hast du eine Empfehlung? Meine Anforderungen wären:

- Vanilla JS, also kein jQuery, React etc.
- Auswahl von Tag, Woche, Monat, Jahr und Zeiträumen
- Maintained, letzter Commit also nicht vor etlichen Jahren

[Flatpickr](https://flatpickr.js.org/) kommt dem schon recht nah, hat aber keine Jahresauswahl und scheint seit 2 Jahren tot zu sein.

gereons, 2024-04-12:
Leider kann ich nicht mit einer Empfehlung dienen, ich mach fast ausschliesslich native iOS-Entwicklung und ein bisschen Serverkram und bin bei den ganzen Webdev-Sachen komplett 'raus. Ich frag aber bei Gelegenheit mal die Kollegen die unsere Webseite bauen.

Für meinen Vorschlag wäre aber auch gar kein Picker nötig :-)

ledermann, 2024-04-12:
Stimmt, eine Auswahl von "Letzte 7 Tage", "Letzte 30 Tage", oder "Letzte 12 Monate" würde auch ohne Datepicker funktioniert. Da es den Datepicker aber für andere Zeiträume bräuchte, möchte ich das gerne integriert lösen.

gereons, 2024-04-18:
Mir wurde https://github.com/t1m0n/air-datepicker als brauchbar empfohlen, das scheint auf den ersten Blick deine Checkliste zu erfüllen.

ledermann, 2024-04-18:
Danke für den Tipp, @gereons, diesen Datepicker hatte ich nicht auf dem Schirm. Sieht gut aus, den werde ich bei nächster Gelegenheit mal austesten.

bennysweb, 2024-08-23:
@ledermann wie schaut es hier aus? Konntest du schon testen? Bin immer noch stark an dem Feature interessiert.

ledermann, 2024-08-23:
@bennysweb Nein, bislang ist hier noch nichts passiert.

closed 2025-03-04 by ledermann